OpenEtherCATsociety/SOEM #201: CiA402 support for Nidec Unidrive M700

jakelstr, 2018-08-24:
Hey Everyone,
I'm having a little trouble setting up my Unidrive M700 to work with CiA402 support with soem. 

Everything appears to be normal, I can read/write SDOs, change Syncmanager configurations, read/write PDOs without any issue. 

The issue is that the drive will not step through the CiA402 State Machine. It is stuck at "Not Ready To Switch On", which according to the book should be an automatic transition to "Switch on disabled".

I have spent quite a while on the phone with Nidec support, and the only thing they can point to is my EtherCAT master not initializing the drive profile properly. 

I have pasted the ESI file here: https://pastebin.com/V3p6U7FP

The portion i believe is relevant is the tags "InitCmd" under "CoE". Specifically, the first 2 "initcmd" tags tell me to write "FA" into an 8 bit register (makes no sense if it is supposed to be 0xFA, I suspect that's not what it means).

Something else I would like to know is what it means that the "CoE" tag appears under "Mailbox". Does that mean I need to do something with the mailboxes? I am not, unless soem is doing it automatically.

ArthurKetels, 2018-08-24:
Could you also please give me the output of slaveinfo -map? And perhaps also a zipped wireshark trace of simple_test?

And indeed for this slave you would need to use the ex_slave[n].PO2SOconfig hook setup to program the CoE indexes that you found in the ESI file. The rest you could just ignore but I am a bit worried if SOEM will set SM4 and SM5. It would only do so if they are correctly defined in the slave SII eeprom.

The datatype for <Data> is hexBinary meaning FA is 0xFA. So CoE Index 0x60c2 subindex 0x02 should be 0xfa. There is the modifier AdaptAutomatically="1". This means the default value 0xfa should be replaced by the "real" value. 

Looking into DS402:
60C2h: Interpolation Time Period   
The interpolation time period defines the rate in which target commands are sent by the host to the drive. 

Interpolation Time Period = [interpolation time period value] x10^(interpolation time index) seconds

The conclusion is that Interpolation time period = 250*10^-6s = 250us.

jakelstr, 2018-08-25:
Here is slaveinfo -map, you will note I am using a driver for a Wiznet 5500 chip written by the gentleman indicated in the output. I do not believe this driver is the culprit as I am able to successfully exchange data in both directions, but anything could be a problem at this point. 
[map.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2320635/map.txt)

I tried setting the interpolation values to 0xFA, I also tried 250, -6, which is the default value in the drive, to no avail. The drive indicates it is receiving PDO data 2000 times a second, which I increased to 8000 times a second from the master, which made no difference. 

Something interesting to note is the drive does flash "PDOs Lost" periodically, though I have no reason to believe that is the truth. The only thing I can attribute that to is the fact that my master essentially sends out the same PDO data cyclically (it writes decimal 6 into the DS402 control word on each pass), as I am presently trying to diagnose the drive sticking in the initial mode. 

I unfortunately do not have a way to get a Wireshark trace on this as it is running on a raspberry pi, through an alternate interface, unless it is sufficient to plug my windows laptop into the network as the last link in the chain and record that way. I can try that if it sounds reasonable.

ArthurKetels, 2018-08-25:
Previous experiences with servo drives let me to become careful with the DS402 control word. My most successful strategy is still:
- enter safe-op
- start with CW 0
- make PDO's go as synchronous as possible with low jitter (at first stick to 1KHz update rate)
- wait 100ms or so
- enter op
- wait for 1s (to let the drive catch you PDO period and sync on it)
- Check status word
- Only when the status word is ok enter the next step in CW
- repeat above two steps until the SW shows operational

Shand2011, 2021-11-07:
hi ,have you got a solution for Nidec Unidrive M700 CiA402 ESI ?,please if you have a solution please help me

Shand2011, 2021-11-07:
Where did you get the ESI file ?

ArthurKetels, 2021-11-07:
Try the Unidrive website? Seems to be just there for download.

Shand2011, 2021-11-10:
thanks ,so did you find anything relating CiA402 ?