OpenEtherCATsociety/SOEM #203: Fail ecx_FPWR due to frame length ?

alessiomargan, 2018-08-29:
Hi all,

I am using soem 1.3.1 since years with no big issues, but I am going crazy now.
The slaves are based on ET1100 with a modified version of soes.
  
I am running the same simple_test on 3 different x86 machines :
- laptop ubuntu 16.04 ==> ok
- embedded com-express debian with xenomai ==> ok
- desktopp ubuntu 16.04 ==> KO

the test fail on ecx_mbxsend : Fail ecx_FPWR -1

using wireshark the only difference seems the frames length : 
- 170 bytes (with pad bytes ?!?) when fail
- 156 bytes works fine 
[soem_simple_test_mbx.pcap.gz](https://github.com/OpenEtherCATsociety/SOEM/files/2333148/soem_simple_test_mbx.pcap.gz)

Any hints ?

Regads
Alessio

ArthurKetels, 2018-08-30:
Thanks for the report and wireshark capture. With these I can almost certainly say that this is a bug in af_packet.c in the Linux kernel. In the Linux tree this bug is patched but upstream kernels from distributions (not only Ubuntu) recently (ubuntu 4.15.0-33) included the buggy patch without the fix patch.

Buggy patch 11-05-2018 : https://github.com/torvalds/linux/commit/b84bbaf7a6c8cca24f8acf25a2c8e46913a947ba#diff-e936270d5d8fefaf66cb5ded38a638fc
Fix for above patch 25-05-2018: https://github.com/torvalds/linux/commit/9aad13b087ab0a588cd68259de618f100053360e#diff-e936270d5d8fefaf66cb5ded38a638fc
[image]

You hint with the difference in frame length explains the behaviour. The bug causes the skb (where the packet resides in the kernel) to be extended at the end with eth_header length (14 bytes). The 14 bytes show up as padding bytes after the normal packet.

EtherCAT must allow padding bytes because the minimum Ethernet packet is defined as 64 bytes including FCS (frame check sequence / CRC). Short EtherCAT packets get padded by the network stack to 64bytes. But EtherCAT packets >64bytes are not allowed to have padding. The ESC checks this and will invalidate those packets.

The result is that at first there seems to be nothing wrong when SOEM starts up. The first frames are all shorter than 64bytes. But when mailbox transfer is started or PDO data is transferred the packets are larger than 64 bytes. SOEM (or better the master NIC) will not see any return packets. Configuration will therefore fail. 

The solution is to apply the fix patch to your kernel and recompile. Also it would be nice if anyone out there would file a bug report to the distributions involved. The fix is already there.

It also explains why mainstream Linux kernels do not show this bug, it got patched rather quickly. My 4.16-preempt-rt kernel from stock Linux just works.

alessiomargan, 2018-08-30:
Thanks to you Arthur !!
just downgrade the kernel (4.11.0-14-lowlatency) and work as expected.
Thanks again
Alessio

closed 2018-08-30 by alessiomargan