OpenEtherCATsociety/SOEM #204: slave_info: Not all slaves reached safe operational state: Invalid output configuration

matwey, 2018-08-31:
Hello,

I am running `slave_info` example from SOEM with the following configuration. I have three Nanotec N5 in row and fourth ECAT-2052.

I see randomly appearing message
```
Not all slaves reached safe operational state.
Slave 2 State=12 StatusCode=  1d : Invalid output configuration
```
And then the application hung for a few seconds.
The issue is reproduced randomly. Sometimes application works as expected without errors. When the issue appears the slave number is always different. The status code may be either `1d` or `1e`.

Bisecting SOEM I found that the issue not reproduced at current master. The first fixing commit is 59a87a36c16a125062a06e3d2394a8ccb3c83381

However 59a87a36c16a125062a06e3d2394a8ccb3c83381 only masks the issue, it is still there until 2e165bee41dbb9983d2c46677844e7adbd8c446f where the issue disappears even with `#define EC_MAX_MAPT 8`

This note is just for info purposes. If you think that the information is consistent and #194 is really supposed to fix this issue, please close the issue.

Reference: #112 #194

mvukov, 2018-08-31:
There is an older thread here about N5. Boils down that you need recent firmware for N5, I think its 1650.

Before switching from PREOP to SAFEOP make sure to configure the drive and make at least 1 ms pause before the transition. That should solve the problems. The pause in particular should solve random transition failures.

matwey, 2018-08-31:
I do have v1650 on this N5-s.

nakarlsson, 2018-08-31:
Can you please provide a wireshark log when it fails.

matwey, 2018-08-31:
pcap dump is here: https://transfer.sh/bLVKp/ecat.pcap

nakarlsson, 2018-08-31:
the pcap is missing the interesting part, please capture the startup eg. config_init and config_map

matwey, 2018-08-31:
Hm, sorry. Could you please try this one: https://transfer.sh/foVcw/ecat.pcap

nakarlsson, 2018-08-31:
So to clarify, after #194 was commited the issue you mention "Time: 3952743.254 PACKET slave:2 index:1600.03 error:1" have disappeared even with #define EC_MAX_MAPT 8?

matwey, 2018-08-31:
Yes, I cannot reproduce the issue with `#define EC_MAX_MAPT 8` and #194.
It is not trivial for me why does #194 fix the issue, or it is just a coincidence.

nakarlsson, 2018-08-31:
Could you provide a new pcap with matching packet ERROR print from the console

matwey, 2018-08-31:
- https://transfer.sh/ekFbC/slaveinfo.txt
- https://transfer.sh/qXzvW/ecat.pcap

nakarlsson, 2018-08-31:
Yes, you can see where the error comes from in the pcap, you can also see that the system is "overrun", lots of re-transmissions due to timeouts. The PAKET error log I believe is a slave bug, it actually seems to respond with the request packet from the master on a repeat request, framno 890 in the last pcap.

But, I can't see the relation to the fix in #194 fixing it, assuming all spawned tasks get the same prio, if they do they should not be affected by prio inversion, but my knowledge of the scheduling in Linux is very limited. If the prio inversion flag actually helps smoothing the system, perhaps an external low prio task is blocking the EtherCAT threads and get boosted by the prio inversion, making the system run smoother?
Can you post a pcap with the fix #194?
Perhaps someone else on here knows more on Linux and the scheduling, or can make a trace to give an 100% answer, would be nice to know.

matwey, 2018-08-31:
- https://transfer.sh/fr6Wk/slaveinfo.txt
- https://transfer.sh/G5Qea/ecat.pcap

nakarlsson, 2018-09-02:
Yes, from the pcap it looks like #194 make the system run smoother, I can just assume it suffer from prio inversion issues without #194. But I think #194 only masks the "issue" in some sense, say you would have N slaves and having N threads configuring it, sooner or later you would see the symptoms again I believe. So as I said in #112, it is an advanced feature and need system tuning.

But the issues of "Time:   26586.814 PACKET slave:2 index:1c00.03 error:1" is a slave issue I believe, see https://github.com/OpenEtherCATsociety/SOEM/issues/204#issuecomment-417735918, if you would have a second CoE slave type in the network I think you would see only "Drive" slaves causing the issue.
Does it pass CTT?

In the end I think this issue can be closed, #112 is still valid and #194 greatly improves the threading capability during configuration.

nakarlsson, 2018-09-21:
Ok If I close it?

nakarlsson, 2018-09-28:
We'öö close it

closed 2018-09-28 by nakarlsson