OpenEtherCATsociety/SOEM #206: Empty info from slaveinfo

deysumitkr, 2018-09-02:
On running [slaveinfo](https://github.com/OpenEtherCATsociety/SOEM/tree/0398b6ba72f909f300ada01e92bc490b64f427cd/test/linux/slaveinfo) for PR2 robot's slaves, very little information about the slaves is found. For other slaves, like EBox, a rich info about the slave is obtained.

What might be the possible reason for almost empty output from slaveinfo?

Below is the stdout of slaveinfo on PR2 for the first slave (output is similar for all slaves): 
`$ sudo ./slaveinfo enp10s0 -map -sdo`

```
Slave:1
 Name:? M:0000eeee I:0067d60d
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 0000eeee ID: 0067d60d Rev: 06020116
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to SII :
```

nakarlsson, 2018-09-02:
Probably communication issues or a faulty slave. The data presented are incomplete, since it is EEPROM data it is harder to track in wireshark but please provide a wireshark log.

ArthurKetels, 2018-09-02:
Yes a writeshark log would be most helpful. But if the data above is accurate it is not the only slave. It has ports 0 and 1 active. So it must mean there are other slaves. What is their output? Also this slave just looks like a bridge slave with no PDI. It has no FMMU's defined, no mailbox and no CoE.

ArthurKetels, 2018-09-04:
Thanks for the wireshark traces. The reason for the slave behaviour is clear. They do not conform to the EtherCAT standard. The standard eeprom data is very minimal and only enough to recognise the slave manufacturer and product code. The rest is vendor specific. Therefore SOEM can not configure the slave in the standardised way.

The reason it works with eml is that there are special drivers programmed. They will configure the slaves directly programming the slave registers. Actually eml also uses SOEM internally.

So your only option is to replicate the WillowGarage drivers for your application.

nakarlsson, 2018-09-28:
@deysumitkr , can we close this?

deysumitkr, 2018-10-02:
Yes, we can close this.

closed 2018-10-02 by deysumitkr