OpenEtherCATsociety/SOEM #208: Building error on RT kernel - undeclared ‘PTHREAD_PRIO_INHERIT’

deysumitkr, 2018-09-04:
On building SOEM on (ubuntu) RT kernel, the following error occurs.
```
$ make
Scanning dependencies of target soem
[  7%] Building C object CMakeFiles/soem.dir/soem/ethercatconfig.c.o
[ 14%] Building C object CMakeFiles/soem.dir/soem/ethercatsoe.c.o
[ 21%] Building C object CMakeFiles/soem.dir/soem/ethercatdc.c.o
[ 28%] Building C object CMakeFiles/soem.dir/soem/ethercatcoe.c.o
[ 35%] Building C object CMakeFiles/soem.dir/soem/ethercatfoe.c.o
[ 42%] Building C object CMakeFiles/soem.dir/soem/ethercatprint.c.o
[ 50%] Building C object CMakeFiles/soem.dir/soem/ethercatbase.c.o
[ 57%] Building C object CMakeFiles/soem.dir/soem/ethercatmain.c.o
[ 64%] Building C object CMakeFiles/soem.dir/osal/linux/osal.c.o
[ 71%] Building C object CMakeFiles/soem.dir/oshw/linux/nicdrv.c.o
SOEM/oshw/linux/nicdrv.c: In function ‘ecx_setupnic’:
SOEM/oshw/linux/nicdrv.c:130:51: error: ‘PTHREAD_PRIO_INHERIT’ undeclared (first use in this function)
SOEM/oshw/linux/nicdrv.c:130:51: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [CMakeFiles/soem.dir/oshw/linux/nicdrv.c.o] Error 1
make[1]: *** [CMakeFiles/soem.dir/all] Error 2
make: *** [all] Error 2

```

OS/Kernel details:
```
$ cat /proc/version
Linux version 3.4.28.rt40-1-pr2-rt (willow@ian) (gcc version 4.6.3 (Ubuntu/Linaro 4.6.3-1ubuntu5) ) #1 SMP PREEMPT RT Mon Feb 4 15:07:27 PST 2013
```

It appears to compile and run fine, if required changes are made in `/usr/include/pthread.h` - which is not a good idea. Using a generic kernel also solves the problem.

Is there a way to build on rt kernel without messing with the existing files?

ArthurKetels, 2018-09-04:
I have build the latest SOEM on my box with 4.16-rt and all is fine. 

Also pthread.h is not part of the kernel but gnu clib. I checked clib back to 2014 and there pthread.h is fine and has PTHREAD_PRIO_INHERIT defined.

Perhaps you use a very old clib?

deysumitkr, 2018-09-04:
On the concerned system, `PTHREAD_PRIO_INHERIT` is defined under a condition in pthread.h
```
#ifdef __USE_UNIX98
/* Mutex protocols.  */
enum
{
  PTHREAD_PRIO_NONE,
  PTHREAD_PRIO_INHERIT,
  PTHREAD_PRIO_PROTECT
};
#endif
```
The condition `__USE_UNIX98` gets undefined by features.h
As pthread.h includes features.h, the condition fails and `PTHREAD_PRIO_INHERIT` remains undefined

Below is the gnu libc version details. It is indeed older than 2014.
```
$ /lib/x86_64-linux-gnu/libc.so.6
GNU C Library (Ubuntu EGLIBC 2.15-0ubuntu10.7) stable release version 2.15, by Roland McGrath et al.
Copyright (C) 2012 Free Software Foundation, Inc. [...]
Compiled by GNU CC version 4.6.3.
Compiled on a Linux 3.2.60 system on 2014-08-28.
[...]
```

deysumitkr, 2018-09-04:
Updating glibc might affect the existing build system.

I suppose the easiest and possibly clean way to solve it is, by defining `_XOPEN_SOURCE` to a value greater than 500 within the source which will set `__USE_UNIX98` and consequently `PTHREAD_PRIO_INHERIT`.
```
// /usr/include/features.h

# if (_XOPEN_SOURCE - 0) >= 500
#    define __USE_XOPEN_EXTENDED  1
#    define __USE_UNIX98  1
[...]
```

ArthurKetels, 2018-09-04:
I would recommend to update your linux box to something more standard and not something hacked by willow garage.

Timercmp is a really standard function and should just work.
https://linux.die.net/man/3/timercmp

deysumitkr, 2018-09-05:
Right, it might be best to use a new linux box. 
Thanks!

closed 2018-09-05 by deysumitkr