OpenEtherCATsociety/SOEM #209: Possibility of master missing EtherCAT frames with ec_receive_processdata?

mohse-n, 2018-09-17:
Hi. I want to know whether there is the possibility of master missing frames when using 
```
   ec_send_processdata();
   wkc = ec_receive_processdata(EC_TIMEOUTRET);
```
1- If, after `ec_send_processdata`, there is a context switch for some reason, will I still receive the frame after returning to this thread of execution (maybe it's stored in a low-level buffer?), or is it lost?
2- In `simple_test.c`, there is a 5 ms pause at the end of the for loop. What are some factors (in hardware or software) affecting the maximum possible rate of execution of this loop? 
For example, in the loop in [my code](https://github.com/messiah1376/L7N_EtherCAT/blob/master/simple_test.c#L313), I send target position to servo drives. When I reduced this to `osal_usleep(200)`, I noticed one of them (there are 2) is moving in a jerky fashion and is seemingly missing frames.

nakarlsson, 2018-09-18:
Yes, you will still receive the frame, it will be stored in lower levels, where exactly depends on your software.

nakarlsson, 2018-09-21:
Should be OK

closed 2018-09-21 by nakarlsson