OpenEtherCATsociety/SOEM #210: Question about the loop in simple_test.c

mohse-n, 2018-09-18:
In `simple_test.c`, there is a 5 ms pause at the end of the for loop. What are some factors (in hardware or software) affecting the maximum possible rate of execution of this loop? 
For example, in the loop in [my code](https://github.com/messiah1376/L7N_EtherCAT/blob/master/simple_test.c#L313), I send target position to two servo drives. When I reduced this to `osal_usleep(200)`, I noticed one of motors (there are 2) is moving in a jerky fashion and is seemingly missing frames.

nakarlsson, 2018-09-18:
I'd say CPU speed, RAM access, threading, ethernet controller usage, software overhead, EtherCAT frame size. I'd think you need to know all of these to achive realy low cycle times.

Examples, 
1.  I've a XMC4800 cortex M4F running at 144Mhz and it run SOEM on our RTOS rt-kernel, memory and peripherals are internal, I can run it at an EtherCAT cycle of 100(us) in a DC control application. It works well even if it runs at a low CPU speed compared to Intel CPUs.

2. A customer of us run SOEM at 100(us) on RTOS InTime on an 2GHz Intel i7 with external PCIe ethernet controllers.

3. You would probably NOT be able to run SOEM at 100 (us) on vanilla Linux or Windows regardless of HW speed.

So, the conclusion is that you need deterministic software that can access Ethernet HW in an efficient way to be able to reach low cycle times.

**OBS!** It is strange that only one motor is moving in a jerky fashion, shouldn't make a difference, both get the same frame all the time if the network HW with PHYs and cables work properly.

mohse-n, 2018-09-18:
Thanks. What are some methods for measuring the rate at which the master sends out EtherCAT frames? 
Although I can put a delay at the end of my loop, I want to get a sense of other latencies (linux networking stack, network card, scheduler, etc.) at work.
Is using the "Time" value in Wireshark a good method?

nakarlsson, 2018-09-21:
No not Time in wireshark.

Perhaps ec_DCtime , the diff between DC time would be the frames actual cycltime since it a timestamp at the first DC slave.