OpenEtherCATsociety/SOEM #212: Slave Missing Object Dictionary,How to config slave from a ESI file?

Bingyu1413, 2018-09-18:
I use soem to control some ethercat module well,But here I'v got a IO Device that can not mapping config correct, the device does not have a default configuration, at least does not have enough CoE the Object Dictionary.This question may be same with question #94 .

I'v read the #94,but I can't solve this problem.
Can soem like twincat to config a slave from a ESI file?
And how?

nakarlsson, 2018-09-18:
Try posting the ESI file

Bingyu1413, 2018-09-19:
[LCT EtherCAT Module V2.2 ForADlink.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2394897/LCT.EtherCAT.Module.V2.2.ForADlink.txt)

Bingyu1413, 2018-09-19:
here is the LCT_Setup Code:
int l = sizeof(uint32);
	uint32 u32val = 0;
	ec_SDOread(slave, 0x1000, 0, FALSE, &l, &u32val, EC_TIMEOUTRXM);
	if (u32val == 0x12d)
	{
		//It is LCT IO module
		l = sizeof(uint32);
		uint32 module_num = 0;

		ec_SDOread(slave, 0xF050, 0, FALSE, &l, &module_num, EC_TIMEOUTRXM);

		uint16 ic12_data[32] = { 0 };
		uint8 ic12_num = 0;

		uint16 ic13_data[33] = { 0 };
		uint8 ic13_num = 0;

		uint8 u8val = 0;
		ec_SDOwrite(slave, 0x1c12, 0, FALSE, sizeof(uint8), &u8val, EC_TIMEOUTEEP);
		ec_SDOwrite(slave, 0x1c13, 0, FALSE, sizeof(uint8), &u8val, EC_TIMEOUTEEP);

		for (int i = 1; i < module_num; i++)
		{
			ec_SDOread(slave, 0xF050, 0, FALSE, &l, &module_num, EC_TIMEOUTRXM);

			uint32 module_type = 0;
			l = sizeof(uint32);
			ec_SDOread(slave, 0xF050, i, FALSE, &l, &module_type, EC_TIMEOUTRXM);

			if (module_type == 1)
			{
				ic13_data[ic13_num] = 0x1a01 + i * 0x10;
				ic13_num++;

				uint16 u16val = 0x1a01 + i * 0x10;
				uint8 u8val = 0x00;
				ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTEEP);

				u32val = 0x60000108 + i * 0x10000;
				ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTEEP);

				u32val = 0x60000208 + i * 0x10000;
				ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTEEP);

				u16val = 0x1a01 + i * 0x10;
				u8val = 0x02;
				ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTEEP);
			}
			else
			{
				ic12_data[ic12_num] = 0x1603 + i * 0x10;
				ic12_num++;

				uint16 u16val = 0x1603 + i * 0x10;;
				uint8 u8val = 0x00;

				ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTEEP);

				u32val = 0x70000108 + i * 0x10000;
				ec_SDOwrite(slave, u16val, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTEEP);

				u32val = 0x70000208 + i * 0x10000;
				ec_SDOwrite(slave, u16val, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTEEP);

				u16val = 0x1603 + i * 0x10;;
				u8val = 0x02;

				ec_SDOwrite(slave, u16val, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTEEP);
			}

		}

		
		for (int j = 0; j < ic12_num; j++)
		{
			ec_SDOwrite(slave, 0x1c12, j + 1, FALSE, sizeof(uint16), &ic12_data[j], EC_TIMEOUTEEP);
		}

		for (int j = 0; j < ic13_num; j++)
		{
			ec_SDOwrite(slave, 0x1c13, j + 1, FALSE, sizeof(uint16), &ic13_data[j], EC_TIMEOUTEEP);
		}

		ec_SDOwrite(slave, 0x1c12, 0, FALSE, sizeof(uint8), &ic12_num, EC_TIMEOUTEEP);
		ec_SDOwrite(slave, 0x1c13, 0, FALSE, sizeof(uint8), &ic13_num, EC_TIMEOUTEEP);
	}//*/

	ec_slave[slave].SM[2].SMflags = 0x10024;
	ec_slave[slave].SM[4].StartAddr = 0;
	ec_slave[slave].SM[5].StartAddr = 0;

nakarlsson, 2018-09-21:
According to the ESI ,
the vendor id/eep_man is  0x186 
the product code/eep_id/SDO 0x1018:02  is 0x24

If TwinCAT can sort out the configuartion from the ESI, you can re-use the configuration settings from TwinCAT and make SOEM do the same, all tools are available SDOread/write and with Complete Access if needed.

Bingyu1413, 2018-09-25:
if I do not use the LCT_Setup(int slave) function,I use simple_test,CoE object Description found,23 entries.
and here is one of the entrie:
Index: 7000 Datatype: 002a Objectcode: 09 Name: DO Outputs
 Sub: 00 Datatype: 0005 Bitlength: 0008 Obj.access: 0007 Name: SubIndex 000
         Value :0x02 2
 Sub: 01 Datatype: 0005 Bitlength: 0008 Obj.access: 0047 Name: Byte Output
         Value :0x00 0
 Sub: 02 Datatype: 0005 Bitlength: 0008 Obj.access: 0047 Name: \377
         Value :0x00 0

But if I use the LCT_Setup function,I can only find 17 Coe entries.the entry like index:7000 has been missing.Why?
How can I add Coe entries ?

nakarlsson, 2018-09-25:
By looping from int i = 1 you’ll skip those entries since you add the offet * i?

Should you loop from i = 0 or adjust the offset * (i - 1)?

nakarlsson, 2018-12-18:
will close since no update since sep

closed 2018-12-18 by nakarlsson