OpenEtherCATsociety/SOEM #214: Slave doesn't switch to OPERATIONAL STATE under Linux

1maennel, 2018-09-26:
Hi,

I am currently working on a project using Infineon XMC4800 with the SSC-Tool v.5.12 and SOEM on several operating systems. I have my system running perfectly fine under Windows 10 and Windows 7. When I now port me setup to a Linux system with a RT-patch I get the same results as described in Issue https://github.com/OpenEtherCATsociety/SOEM/issues/138.
I already checked if the kernel allows packed structs and run out of ideas where to look.

I really would prefer a solution to this problem, other switching to the old SSC-Tool Version.

Please Find attached the Wireshark captures on both systems:
for Linux: [wireshark_linux.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2421661/wireshark_linux.pcapng.txt)
For Windows: [wireshark_win.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2421662/wireshark_win.pcapng.txt)

nakarlsson, 2018-09-27:
In #138 the data in the slave was not aligned, that would give the same issue on both Win & Linux

Also, #138 was fixed by modifying the salve code.

ArthurKetels, 2018-09-28:
The issue is with your slave implementation.

With windows SOEM reads some of the eeprom wrong and can not do CoE configuration. By luck it then retries with eeprom mapping that actually succeeds. 
With linux the eeprom read is ok but you have something wrong in the CoE mapping of the slave. SOEM configures the FMMU wrong and then the slave does not go to OP mode (should have given an error with SafeOP+error).

My suggestion is to sort out your slave implementation issues first and do a slave conformance test. Then retry with SOEM.

When I have some more time I dig deeper in the wireshark traces to figure out the exact fail reasons.

nakarlsson, 2018-10-09:
We’ll close for now, open a new issue if issue still appear after following Arthurs recommendation. #138 contain information about misaligned slave data.

closed 2018-10-09 by nakarlsson

arne48, 2018-10-17:
I just also stumbled over this issue and read this issue and #138 thoroughly.
But I couldn't find any hint for the final solution, which was found in modifying the slave code according to @nakarlsson.
I also don't want to post code as it would be a snippet of undisclosed code. But I want to tell you where you can find it.
@1maennel and @mikekaram because you have access to the slave stack you should also have access to the Dev-Forum  on ethercat.org. There under "Evaluation Kit Software" you find an announcement to the SSC 5.12. This post contains recommendations for fixes. The recommendation of the 27th of June 2018 seems to be the modification @nakarlsson mentioned.

arne48, 2018-10-23:
A patch-file including this change is now part of the newest ECAT_SSC DAVE-App (4.0.22).