OpenEtherCATsociety/SOEM #216: Configure new EtherCAT-Device with ROS

schwarzp93, 2018-10-11:
Hello all :)

first of all, I'm sorry, if my question might be silly or trivial, but I'm pretty new to the world EtherCAT.

I'm trying to controll a Festo CPX-FB37 bus node (https://www.festo.com/net/SupportPortal/Files/344605/CPX-FB37_2014-06_8029675g1.pdf) via ros, using SOEM. 

I'm already having problems, when using the simple_test or slaveinfo. In both cases, the following error gets displayed:
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Is there a kind of tutorial on how to start with SOEM? How can I connect new devices, which steps are done automatically, which have to be done manually?

Thanks a lot in advance, 

Philipp

Attached you find the outputs of simple_test and slaveinfo.
[simpleTest_Output.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2468070/simpleTest_Output.txt)
[slaveinfo_Output.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2468071/slaveinfo_Output.txt)

carlosjoserg, 2018-10-11:
I don't know about your question, but since you mention via ROS and you are new to the world of EtherCAT (like me too ;) ), be aware that the soem library distributed as  ros-*distro*-soem is not the one coming from this repository.

The one distributed within the ROS repositories is: https://github.com/smits/soem (see also the ros wiki: http://wiki.ros.org/soem for more info about it)

Though the APIs are similar (I have code compiling with both libraries), they are not the same maintainers, I guess both have part of the "original soem c code provided by the Technische Universiteit Eindhoven" as said in the ROS wiki.

schwarzp93, 2018-10-11:
Thanks Carlos!

yes, that's also the one I tried to use! I just thought, I should ask the question here too, since it's pretty basic...

schwarzp93, 2018-10-11:
Thanks nakarlsson!

I will try to submit a wireshark tomorrow...
Unfortunately, I don't have another slave. I'm running Ubuntu 16.04 and ROS Kinetic.
I downloaded the soem package from https://github.com/smits/soem, since it was the first one, I found. Are there differences between the packages?

nakarlsson, 2018-10-12:
I can't say, but for basic functionality such as detect and configure slaves I doubt there are any differences.

schwarzp93, 2018-10-12:
Hi :)

here are some wireshark files, I hope, those are the ones you wanted to see and that they are helpful...

[wiresharks.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2472282/wiresharks.zip)

nakarlsson, 2018-10-12:
You can try to increase the "smits/SOEM" default mailbox size, the CPX-FB37 got bigger mailboxes than 
"
/** max. mailbox size */
#define EC_MAXMBX         0x3ff
"

schwarzp93, 2018-10-13:
Thanks again :)
Unfortunately, I don't have access to the device until probably wednesday.
I will keep you updated

schwarzp93, 2018-10-17:
It worked! At least kind of :)

I now get the folliwing output: 

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp3s0 succeeded.
wkc = 1
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 46 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
Processdata cycle    1, WKC 3 , O: 00 00 00 00 00 00 00 00 I: 00 00 00 00 00 00
(The last line repeats itself until Processdata cycle 10000, rest ist always the same)
Processdata cycle 10000, WKC 3 , O: 00 00 00 00 00 00 00 00 I: 00 00 00 00 00 00 00 00 T:52476062680
Request init state for all slaves
End simple test, close socket
End program

Is this a normal behavior?

nakarlsson, 2018-10-18:
Looks OK, what does slave info say?
O: & I: is IO data, so that depends on what you have mapped. The O you need to set from SOEM code, I is read from the inputs connected, digital IO? Apply some signal and verfiy input.

schwarzp93, 2018-10-18:
Hi again, and thank again :)

Here are the results from slavinfo:
[slaveinfo.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2491134/slaveinfo.txt)
[slaveinfoMAP.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2491135/slaveinfoMAP.txt)

Currently, I'm trying to apply signals. is there a prepared "application" in the package, or do I have to write this on my own?

Sorry again for these maybe dumb questions...

schwarzp93, 2018-10-18:
I'm using digital and analog IOs, both connected to the same slave

nakarlsson, 2018-10-18:
You can check test examples and tutorial, basically how to work with the IOmap. Every slave instance have their ref to the IOmap via ->inputs,->outputs.

nakarlsson, 2018-10-24:
I think the main issue is solved

closed 2018-10-24 by nakarlsson