OpenEtherCATsociety/SOEM #217: Device with EoE support causes issue when using CoE SDO functions

antol85, 2018-10-16:
Hi, 

first of all thank you for your great work. I have found the library easy to use and it works well in most of the cases. 

I wanted to report an issue I had with a device which supports both CoE and EoE.
When using ethercatcoe.c's functions to read and write SDOs (such us ec_SDOread), I receive an "unexpected frames", which causes the SDO reading operation to fail. 

The reason is that, when for example the ecx_mbxreceive function is called in ecx_SDORead, the system receives a message (wrk >0) but a EoE one (instead of CoE). 
For now I have adopted the workaround of looping until I don't get a message (wrk <= 0) or  I get a CoE message.

      do{
        /* clean mailboxbuffer */
        ec_clearmbx(&MbxIn);
        /* read slave response */
        wkc = ecx_mbxreceive(context, slave, (ec_mbxbuft *)&MbxIn, timeout);
      } while (wkc > 0 && ((aSDOp->MbxHeader.mbxtype & 0x0f) != ECT_MBXT_COE)); 

In my understanding the architecture is not supporting any kind of multiplexing on mailboxes, so it is not ensured that the right message will be dispatched to the right caller.

My question is: have I misinterpreted something on how the ethercatcoe.c's functions are supposed to be used in this context? Are you planning a kind of multiplexing functionality to support devices with both EoE and CoE support?

ArthurKetels, 2018-10-17:
Can you please post a wireshark capture that shows this EoE mailboxes coming in? It is not so difficult to adapt SOEM to handle these unrequested mailmoxes.

nakarlsson, 2018-11-06:
@antol85  , can you provide a wirshark?  thanks.

antol85, 2019-01-11:
Hi, sorry if I reply so late. Unfortunately I didn't do a wirshark capture at the time and since I don't have the device anymore.

closed 2019-01-11 by antol85