OpenEtherCATsociety/SOEM #218: Servo cannot set to OP state, without error

fred-lee-sv, 2018-10-24:
Hello Guys,

I'm trying to using SOEM 1.3.1 for driving a servo, but after run simple test, the servo (ethercat slave) cannot go into op state, it stay in safe op state without error.

Following are the simple_test console output:
```
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp0s25 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 14 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Not all slaves reached operational state.
Slave 1 State=0x04 StatusCode=0x0000 : No error

Request init state for all slaves
End simple test, close socket
End program
```

And following are the slaveinfo -map output:

```
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s25 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Drive
 Output size: 56bits
 Input size: 56bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000016 ID: 0002000c Rev: 00010000
 SM0 A:1000 L: 512 F:00010016 Type:1
 SM1 A:1200 L: 512 F:00010012 Type:2
 SM2 A:1400 L:   7 F:00030064 Type:3
 SM3 A:1568 L:   7 F:00030020 Type:4
 SM4 A:16d0 L:  12 F:00030024 Type:3
 FMMU0 Ls:00000000 Ll:   7 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000007 Ll:   7 Lsb:0 Leb:7 Ps:1568 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:3 2:3 3:1
 MBX length wr: 512 rd: 512 MBX protocols : 06
 CoE details: 2f FoE details: 00 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6060:0x00 0x08 INTEGER8     ModesOfOperation
  [0x0001.0] 0x6040:0x00 0x10 UNSIGNED16   Controlword
  [0x0003.0] 0x60FF:0x00 0x20 INTEGER32    TargetVelocity
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0007.0] 0x6061:0x00 0x08 INTEGER8     ModesOfOperationDisplay
  [0x0008.0] 0x6041:0x00 0x10 UNSIGNED16   Statusword
  [0x000A.0] 0x606C:0x00 0x20 INTEGER32    VelocityActualValue
  SM4 outputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket
End program
```

Attached are the wireshark data 
[simple_test.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2510145/simple_test.zip)

and esi xml file
[LTi_22_SOCM.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2510147/LTi_22_SOCM.zip)

Can you give me  a hint what's wrong with my system?

Thanks in advance,

With best regards,
Fred

ArthurKetels, 2018-10-24:
From the LTi ESI file :
```xml
<Dc>
  <OpMode>
    <Name>DC-Synchron</Name>
    <Desc>DC-Synchron-SE0</Desc>
    <AssignActivate>#x300</AssignActivate>
    <CycleTimeSync0 Factor="1">0</CycleTimeSync0>
    <ShiftTimeSync0>0</ShiftTimeSync0>
  </OpMode>
</Dc>
```
This means that your application code has to program the Sync0 in the slave and run the process data synchronous with the Sync0 at safe-OP and OP. This is outside the scope of simpletest and slaveinfo.

Often this DC synchronous mode is mandatory for servo drives. Look for older post here on the forum to get hints how to solve that with SOEM.

On the other hand the slave in question should bail out after a while with Safe-OP+Error indicating a sync fault. Perhaps the timeout in simpletest is too short to show this (slaves assume 10s timeout).

fred-lee-sv, 2018-10-25:
@ArthurKetels, Thanks very much for your valuable information. I programmed sync0 as method mentioned int #142, and now the servo is set to OP state successfully. This issue can be closed.

closed 2018-10-25 by fred-lee-sv

nakarlsson, 2018-10-25:
@ArthurKetels , we have an issue on timeouts, not highest prio but should be fixed

https://github.com/OpenEtherCATsociety/SOEM/issues/141