OpenEtherCATsociety/SOEM #219: Unable to get sample working in XMC4300 Relax

jamescarter-le, 2018-10-25:
I am using the instructions in this PDF: https://rt-labs.se/content/uploads/EtherCAT_SDK_DAVE.pdf

To get up and running with a very simple Slave based on the XMC4300. I originally constructed this in Rowley Crossworks, however it did not work so I followed the instructions in the document using their DAVE IDE.

In both cases (My Crossworks application, and the copied to the letter DAVE application) the device does appear to start the ESC, and TwinCAT is reporting frames (The RT Lab EtherCat explorer didnt work, maybe it doesnt like sharing the network adapter with the Twincat driver attached? It just says Frame lost).

TwinCAT3 is reporting approx 500 frames/sec, with zero Lost Frames, it does recognise there is a Box, and has addressed it 1001, however it does not seem to read the EEPROM as it has no name, and no Object Dictionary returned.

I do get an error in TwinCat of Severity	'Box 1 ()' (1001): abnormal state change (from 'INIT' to 'Invalid (0)'). Go to 'INIT' and try to go back to 'PREOP'...
But I think regardless I should have gotten details about the same of the Box and the dictionary?

Checking the XMC4300 emulated EEPROM, it does have the appropriate data set in the relevant memory sections.

I would appreciate a couple of pointers from someone who has used an XMC4300 Relax kit to get started with SOES.

nakarlsson, 2018-10-26:
Export the DAVE project and send it to support[at]rt-labs.com

No EtherCAT Explorer don’t work in parallel with TwinCAT

jamescarter-le, 2018-10-26:
Hi @nakarlsson,

Thanks for your response, I've sent over the soes and my test project to your email.

It is understandable that TwinCAT and EtherCAT Explorer would conflict on the same network adapter, not a problem, I should try and find another network adapter for testing as the other network adapter is carrying these packets.

I would be interested in hearing what your findings are, perhaps XMCLib has changed since that PDF was written?

nakarlsson, 2018-10-31:
@jamescarter-le , OK to close this?

nakarlsson, 2018-10-31:
endian issue due to missing LITTLE_ENDIAN define

closed 2018-10-31 by nakarlsson