OpenEtherCATsociety/SOEM #220: ec_SDOread failed

lo11212001, 2018-10-26:
Hello, everyone:

My slave devices are two yaskawa Sigma-5 servo amplifiers.
First, I used TwinCAT to test yaskawa servo amplifier: SDO and processdata communication is fine.

Then, I run SOEM master on my notebook computer to connect two yaskawa servo amplifiers.
My notebook computer: Ubuntu 16.04.3 LTS, Linux version 4.10.0-28-generic.

I download SOEM Version 1.3.3, and look the simple_test example.
[simple_test.c]
* part1. simpletest() to init slave device and run cyclic loop
I check simpletest() is ok: initialize servo amplifier, change slave to OP-state, and exchange processdata.

* part2. Create thread1: run ecatcheck()
I add `ec_SDOread(slave:1, index:0x6064, subindex:0, CA:FALSE, size:4, &target, EC_TIMEOUTRXM)` in ecatcheck().

ec_SDOread() always failed, sometime succeed.
I don't know what reason make SDOread failed？
Could anyone give me some suggestion？
Thanks!

---
trace code:
```
ecx_SDOread(...)
{
	/* send CoE request to slave */
	wkc = ecx_mbxsend(context, slave, (ec_mbxbuft *)&MbxOut, EC_TIMEOUTTXM);
	printf("wkc2=%d\n", wkc);
	...   
	
	if(wkc>0)
	{
		/* read slave response */
		wkc = ecx_mbxreceive(context, slave, (ec_mbxbuft *)&MbxIn, timeout);
		printf("wkc3=%d\n", wkc);
	}
}
```

SDOread Failed:
wkc2 = 0 or wkc2, wkc3 = 1, 0

SDOread succeed:
wkc2, wkc3 = 1, 1

ArthurKetels, 2018-10-26:
It is possible the timout value is set too low for this slave. Try to replace EC_TIMEOUTRXM by 100000. 

You are not supposed to do CoE transfers when a slave is in OP mode. It is possible according to the EtherCAT standard, but there is no guaranteed response time. When the slave is too busy doing work in OP mode there might be none left for mailbox communication. 

Object 6064 should be mappable in the process data. This is the best way to monitor the state.

Then there could be something wrong with SOEM. To check that please send a wireshark trace.

lo11212001, 2018-10-29:
Hi, ArthurKetels:
I find that I modify code in ecx_main_send_processdata().
I restore those code, and then SDO_read() will be fine.
Thanks for your reply.

closed 2018-10-29 by lo11212001