OpenEtherCATsociety/SOEM #221: adddatagram issue

lo11212001, 2018-10-29:
Hi, everyone:

I find `ecx_main_send_processdata()` that contain a `datagram(EC_CMD_FRMW, ECT_REG_DCSYSTIME)`.

I trace code:
After setup LRD datagram, it will call `ecx_adddatagram(EC_CMD_FRMW, ECT_REG_DCSYSTIME)`.

```
ecx_main_send_processdata(...)
{
	...
	/* if inputs available generate LRD */
	if(context->grouplist[group].Ibytes)
	{
		...
		ecx_setupdatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_LRD, idx, w1, w2, sublength, data);
		if(first)
		{
			context->DCl = sublength;
			/* FPRMW in second datagram */
			context->DCtO = ecx_adddatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FRMW, idx, FALSE,
								   context->slavelist[context->grouplist[group].DCnext].configadr,
								   ECT_REG_DCSYSTIME, sizeof(int64), context->DCtime);
			first = FALSE;
		}
		...
	}
}
```

I want to add the other datagram(EC_CMD_FPWR) after the `datagram(EC_CMD_FRMW, ECT_REG_DCSYSTIME)`.

```
	if(first)
	{
		context->DCl = sublength;
		/* FPRMW in second datagram */
		context->DCtO = ecx_adddatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FRMW, idx, FALSE,
							   context->slavelist[context->grouplist[group].DCnext].configadr,
							   ECT_REG_DCSYSTIME, sizeof(int64), context->DCtime);

			  /* add */
			  ecx_adddatagram(context->port, &(context->port->txbuf[idx]), EC_CMD_FPWR, idx, FALSE,
									   context->slavelist[context->grouplist[group].DCnext].configadr,
									   MY_REG_ADDRESS, sizeof(Rsize), &my_data);
		first = FALSE;
	}
```

After I add the other datagram(EC_CMD_FPWR), it will make communication abnormal.(wkc <expectedWKC)

How to add datagram to read/write register at the cyclic frame in the right way？
Thanks.

ArthurKetels, 2018-10-29:
ecx_main_send_processdata() is not supposed to be modified in this way. The current code can only handle two datagrams in one frame. You could however add your FPRW in a separate frame.

Of course feel free to propose a patch so it can handle an arbitrary number of datagrams in one frame ;-)

lo11212001, 2018-11-01:
Thanks for your reply.
I slove it.

closed 2018-11-01 by lo11212001