OpenEtherCATsociety/SOEM #223: question: Simple_test reports "Invalid input configuration"

gbmont, 2018-11-01:
Hello

I am using SOEM from master branch running simple_test

OS: Win 10 64 bit

Hardware: 
Beckhoff EK1100 EtherCAT coupler and 
Beckhoff EL7031 Stepper motor controller

I compiled the application with VC 2017 communinity editon.
Before I changed file **CMakeLists.txt** at the end from **linux** to **win32** directory.

if(BUILD_TESTS) 
  add_subdirectory(test/**win32**/slaveinfo)
  add_subdirectory(test/**win32**/eepromtool)
  add_subdirectory(test/**win32**/simple_test)
endif()

Furthermore I copied CMakeLists.txt from
/test/linux/simple_test/CMakeLists.txt to /test/win32/simple_test/CMakeLists.txt 

for all three applications. Otherwise, NMake stopped with a compilation error.

When I run application slave_info.exe I get following results:

_D:\SOEM\SOEM-master\build\test\win32\slaveinfo>slaveinfo \Device\NPF_{6DB05DCC-EC91-43F9-8E92-3C7767880B01}
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{6DB05DCC-EC91-43F9-8E92-3C7767880B01} succeeded.
2 slaves found and configured.
Calculated workcounter 0

Slave:1
 Name:EK1100
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.0.0
 Configured address: 1001
 Man: 00000002 ID: 044c2c52 Rev: 00120000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0

Slave:2
 Name:EL7031
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 150[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1002
 Man: 00000002 ID: 1b773052 Rev: 001a0000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   8 F:00010024 Type:1
 SM3 A:1180 L:   8 F:00010020 Type:1
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 27 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 120[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program_

That looks good for me.

When I run application simple_test.exe I get following result:
_D:\SOEM\SOEM-master\build\test\win32\simple_test>simple_test \Device\NPF_{6DB05DCC-EC91-43F9-8E92-3C7767880B01}
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on \Device\NPF_{6DB05DCC-EC91-43F9-8E92-3C7767880B01} succeeded.
2 slaves found and configured.
Found EL7031 at position 2
EL7031 slave 2 set, retval = 2
Slaves mapped, state to SAFE_OP.
segments : 1 : 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Not all slaves reached operational state.
Slave 2 State=0x12 StatusCode=0x001e : Invalid input configuration

Request init state for all slaves
End simple test, close socket
End program_

I had a look at the configuration in simple_test and according to the Beckhoff datasheet of the device,
the configuration data should be good.

EL7031: Year of production: 2013, Firmware version 9, hardware version 8.
I updated the EEPROM in TwinCAT.

When I run the simple_test.c located in linux folder under win32, then no configuration is written and
the two devices switch to operaional mode (As I saw the simple_test for Linux and win32 are not the same).

For me it seems, it is a problem during configuration.

Program line:

`while(EcatError) printf("%s", ec_elist2string()); `

in simple_test does not report a SDO-write error.

Does anybody have an idea?

Thx

Geri

gbmont, 2018-11-02:
Hi,

now, I came some steps further. All slaves switch to operational state.

I had a look at the configuration in TwinCAT and it seems to be a byte order problem in file simple_test.c, related to the win32 application folder.

[image]

I changed the code in folder SOEM/test/win32/simple_test/simple_test.c to
[image]

That looks fine.

**..but** when the application enters the loop for reading/writing of PDOs no data are available.

> D:\SOEM\SOEM-master\build\test\win32\simple_test>simple_test \Device\NPF_{6DB05DCC-EC91-43F9-8E92-3C7767880B01}
> SOEM (Simple Open EtherCAT Master)
> Simple test
> Starting simple test
> ec_init on \Device\NPF_{6DB05DCC-EC91-43F9-8E92-3C7767880B01} succeeded.
> 2 slaves found and configured.
> Found EL7031 at position 2
> EL7031 slave 2 set, retval = 4
> Slaves mapped, state to SAFE_OP.
> segments : 1 : 0 0 0 0
> Request operational state for all slaves
> Calculated workcounter 0
> Operational state reached for all slaves.
> Value expected WKC 0, value wkc: -1
> OK : all slaves resumed OPERATIONAL.
> OK : all slaves resumed OPERATIONAL.
> OK : all slaves resumed OPERATIONAL.
> 

After switching to operational mode, count of output and input bytes of slave 2 are read
 oloop = ec_slave[2].Obytes;
 iloop = ec_slave[2].Ibytes;

Both values are zero.
Also working counter
expectedWKC = 0
wkc: -1

call of
 `if (ec_slave[2].state == (EC_STATE_SAFE_OP + EC_STATE_ERROR))`

delivers fals, .no error

Does anybody have an idea?

Thank you 

Geri

mheden, 2018-11-02:
First, I think the Win32 "versions" is severely outdated and should not be used (and be removed from this repo). You should be able to build for Win32 without updating the CMake-files (but it will build the source under the `linux` folder).

It is often not a good way of doing a complete access write to the sync manager assignment so I would recommend adapting this method instead: https://github.com/hefloryd/SOEM/commit/22d8f5d (i.e. set the number of elements to 0 before changing the rest of the items and then set the number of elements to the correct value to make it "kick in").

ArthurKetels, 2018-11-02:
Can you post a wireshark trace of the test in question?

gbmont, 2018-11-02:
Hello

Thank you very much for your help!

@mheden:
I moved to the linux source and adapted some configuration parameters according to the link you addressed but I still get the same result.

@ArthurKetels 
Attached the Wireshark dump when I run the application simple_test without the modifications according to mheden.

[Wireshark_SOEM_Simple_Test.pcapng.gz](https://github.com/OpenEtherCATsociety/SOEM/files/2543686/Wireshark_SOEM_Simple_Test.pcapng.gz)

Thank you

Geri

gbmont, 2018-11-03:
Thank you Arthur for analyzing the data and your hints!!

I will contact Beckhoff and ask for Firmware updates.

Regarding the other way for configuration that would also be good jest for my evaluation.
I wrote a function, that will be called at the beginning of simple_test (see attachment) and set function pointer. slave->PO2SOconfig = EL7031setup; but it seems there are some other flags to be set before changing PDO mapping. At least If I write PDO-Mapping parameters the SOME lib returns  an errror "Attempt to write to a read only object"..
`
	retval += write_SD0_uint8(slave,  0x1600, 0, 0);
	retval += write_SD0_uint32(slave, 0x1600, 1, 0x70000001); // Enable Latch C
	retval += write_SD0_uint32(slave, 0x1600, 2, 0x70000002);
	retval += write_SD0_uint32(slave, 0x1600, 3, 0x70000003); // Set Counter
`
[EL7031config.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2544907/EL7031config.txt)

[image]

> D:\SOEMneu\SOEM-master\build\test\linux\simple_test>simple_test \Device\NPF_{6DB05DCC-EC91-43F9-8E92-3C7767880B01}
> SOEM (Simple Open EtherCAT Master)
> Simple test
> Starting simple test
> ec_init on \Device\NPF_{6DB05DCC-EC91-43F9-8E92-3C7767880B01} succeeded.
> 2 slaves found and configured.
> Found Moog drive
> 
>  Configure slave EL7031 stepper motor drive
> WARNING : slave 2 is in SAFE_OP, change to OPERATIONAL.
> INIT--------------------------------------
> Time:   97516.369 SDO slave:2 index:1600.00 error:06010002 Attempt to write to a read only object
> Time:   97516.370 SDO slave:2 index:1600.01 error:06010002 Attempt to write to a read only object
> Time:   97516.371 SDO slave:2 index:1600.02 error:06010002 Attempt to write to a read only object
> Time:   97516.372 SDO slave:2 index:1600.03 error:06010002 Attempt to write to a read only object
> Time:   97516.373 SDO slave:2 index:1600.04 error:06010002 Attempt to write to a read only object
> 

Thank you and best regards
Geri

gbmont, 2018-11-05:
@ArthurKetels
Last week, I tried to do the configuration manually but I had no access to the registers.

Today, Beckhoff sent me the firmware data (you will not find it on the homepage of Beckhoff) for my hardware version and I also updated EEPROM (EL7031-0000-0024) to my related firmware. Powerup, restart and now the t works fine!

Thank you Arthur!

Best regards
Geri

nakarlsson, 2018-11-05:
Can you please close the issue?

gbmont, 2018-11-05:
Thanks to all.

closed 2018-11-05 by gbmont