OpenEtherCATsociety/SOEM #224: Slave does not change state to OPERATIONAL mode.

symbioRahul, 2018-11-02:
Hello SOEM devs,

I am currently experiencing an issue with using SOEM for a robotic arm. I am unable to change the EtherCAT State Machine to ` OPERATIONAL` mode from `SAFE-OP`. My current theory revolves around the misuse of the distributed clock, but I am not completely sure. How do I go about properly setting up the master to communicate with the slave?

After seeing the lack of information on some issues on this repo, I'm going to try to dump as much relevant information as I can.

slave's ESI file: 
[slave_esi.xml.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2540586/slave_esi.xml.txt)

Here is my `./slaveinfo -map` dump:
```bash
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s31f6 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Robots
 Output size: 480bits
 Input size: 992bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00ecade1 ID: 0004d500 Rev: 00000001
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L: 124 F:00010064 Type:4
 SM3 A:1400 L:  60 F:00010020 Type:3
 FMMU0 Ls:00000000 Ll:  60 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000003c Ll: 124 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 27 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 inputs
     addr b   index: sub bitl data_type    name
  [0x003C.0] 0x7200:0x01 0x01 BOOLEAN      Enable
  [0x003C.1] 0x7200:0x02 0x01 BOOLEAN      Activate
  [0x003C.2] 0x7200:0x03 0x01 BOOLEAN      Home
  [0x003C.3] 0x7200:0x04 0x01 BOOLEAN      Reset Error
  [0x003C.4] 0x7200:0x05 0x01 BOOLEAN      Sim Mode
  [0x003C.5] 0x0000:0x00 0x1B
  [0x0040.0] 0x7310:0x01 0x01 BOOLEAN      SetPoint
  [0x0040.1] 0x7310:0x02 0x01 BOOLEAN      Pause
  [0x0040.2] 0x7310:0x03 0x01 BOOLEAN      Resume
  [0x0040.3] 0x7310:0x04 0x01 BOOLEAN      Clear Move
  [0x0040.4] 0x0000:0x00 0x1C
  [0x0044.0] 0x7305:0x00 0x20 UNSIGNED32   Move Command
  [0x0048.0] 0x7306:0x01 0x20 REAL32       SubIndex 001
  [0x004C.0] 0x7306:0x02 0x20 REAL32       SubIndex 002
  [0x0050.0] 0x7306:0x03 0x20 REAL32       SubIndex 003
  [0x0054.0] 0x7306:0x04 0x20 REAL32       SubIndex 004
  [0x0058.0] 0x7306:0x05 0x20 REAL32       SubIndex 005
  [0x005C.0] 0x7306:0x06 0x20 REAL32       SubIndex 006
  [0x0060.0] 0x7340:0x01 0x20 REAL32       SubIndex 001
  [0x0064.0] 0x7340:0x02 0x20 REAL32       SubIndex 002
  [0x0068.0] 0x7340:0x03 0x20 REAL32       SubIndex 003
  [0x006C.0] 0x7340:0x04 0x20 REAL32       SubIndex 004
  [0x0070.0] 0x7340:0x05 0x20 REAL32       SubIndex 005
  [0x0074.0] 0x7340:0x06 0x20 REAL32       SubIndex 006
  [0x0078.0] 0x7350:0x01 0x20 REAL32       SubIndex 001
  [0x007C.0] 0x7350:0x02 0x20 REAL32       SubIndex 002
  [0x0080.0] 0x7350:0x03 0x20 REAL32       SubIndex 003
  [0x0084.0] 0x7350:0x04 0x20 REAL32       SubIndex 004
  [0x0088.0] 0x7350:0x05 0x20 REAL32       SubIndex 005
  [0x008C.0] 0x7350:0x06 0x20 REAL32       SubIndex 006
  [0x0090.0] 0x7360:0x01 0x20 REAL32       Cart Speed
  [0x0094.0] 0x7360:0x02 0x20 REAL32       Cart Ang Speed
  [0x0098.0] 0x7360:0x03 0x20 REAL32       Cart Accel
  [0x009C.0] 0x7360:0x04 0x20 REAL32       Ang Speed
  [0x00A0.0] 0x7360:0x05 0x20 REAL32       Ang Accel
  [0x00A4.0] 0x7360:0x06 0x20 REAL32       Blending
  [0x00A8.0] 0x7360:0x07 0x08 UNSIGNED8    Config
  [0x00A9.0] 0x0000:0x00 0x18
  [0x00AC.0] 0x7500:0x04 0x01 BOOLEAN      Enable
  [0x00AC.1] 0x7500:0x03 0x01 BOOLEAN      Open
  [0x00AC.2] 0x0000:0x00 0x1E
  [0x00B0.0] 0x7500:0x01 0x20 REAL32       Velocity
  [0x00B4.0] 0x7500:0x02 0x20 REAL32       Force
  SM3 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6010:0x01 0x01 BOOLEAN      Busy
  [0x0000.1] 0x6010:0x02 0x01 BOOLEAN      Activated
  [0x0000.2] 0x6010:0x03 0x01 BOOLEAN      Homed
  [0x0000.3] 0x6010:0x04 0x01 BOOLEAN      SimActivated
  [0x0000.4] 0x6010:0x05 0x01 BOOLEAN      Error
  [0x0000.5] 0x0000:0x00 0x1B
  [0x0004.0] 0x6015:0x01 0x20 UNSIGNED32   FIFO Space
  [0x0008.0] 0x6015:0x02 0x08 UNSIGNED8    Actual Conf
  [0x0009.0] 0x6015:0x03 0x01 BOOLEAN      Paused
  [0x0009.1] 0x6015:0x04 0x01 BOOLEAN      EOB
  [0x0009.2] 0x6015:0x05 0x01 BOOLEAN      EOM
  [0x0009.3] 0x0000:0x00 0x15
  [0x000C.0] 0x6030:0x01 0x20 REAL32       SubIndex 001
  [0x0010.0] 0x6030:0x02 0x20 REAL32       SubIndex 002
  [0x0014.0] 0x6030:0x03 0x20 REAL32       SubIndex 003
  [0x0018.0] 0x6030:0x04 0x20 REAL32       SubIndex 004
  [0x001C.0] 0x6030:0x05 0x20 REAL32       SubIndex 005
  [0x0020.0] 0x6030:0x06 0x20 REAL32       SubIndex 006
  [0x0024.0] 0x6031:0x01 0x20 REAL32       SubIndex 001
  [0x0028.0] 0x6031:0x02 0x20 REAL32       SubIndex 002
  [0x002C.0] 0x6031:0x03 0x20 REAL32       SubIndex 003
  [0x0030.0] 0x6031:0x04 0x20 REAL32       SubIndex 004
  [0x0034.0] 0x6031:0x05 0x20 REAL32       SubIndex 005
  [0x0038.0] 0x6031:0x06 0x20 REAL32       SubIndex 006
End slaveinfo, close socket
End program
```

Also if you are interested, this is the `./eepromtool ifname 1 -i`
```bash
SOEM (Simple Open EtherCAT Master)
EEPROM tool
ec_init on enp0s31f6 succeeded.
1 slaves found.
Slave 1 data
 PDI Control      : 0E80
 PDI Config       : CC00
 Config Alias     : 0000
 Checksum         : 0022
   calculated     : 0022
 Vendor ID        : 00ECADE1
 Product Code     : 0004D500
 Revision Number  : 00000001
 Serial Number    : 00000000
 Mailbox Protocol : 0004
 Size             : 000F = 2048 bytes
 Version          : 0001
End, close socket
End program
```

Here is the wireshark of me using `./simpletest ifname`: 
[simpletest_slavedump.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2540582/simpletest_slavedump.pcapng.txt)

Also I wrote a small piece of code that does the following:
```c
#include <stdint.h>
#include <stdio.h>
#include <ethercat.h>

int main(int argc, char**argv) {
    if (!ec_init("enp0s31f6")) {
        printf("bad ec_init\n");
    }

    if (!ec_config_init(0)) {
        printf("bad ec_config_init\n");
    }

    printf("Robot name: %s\n", ec_slave[1].name);

    char iomap[4096];
    printf("config_map: %d\n", ec_config_map(&iomap));

    ec_slave[1].state = 8;
    // iomap[0x3C] = 0x02;
    ec_slave[1].outputs[0] = 0x02;
    printf("send: %d\n", ec_send_processdata());
    printf("recv: %d (wkc)\n", ec_receive_processdata(EC_TIMEOUTRET));

    printf("writestate: %d\n", ec_writestate(0));
    printf("statecheck: %d\n", ec_statecheck(0, EC_STATE_OPERATIONAL, EC_TIMEOUTSTATE));

    printf("current state: %d\n", ec_readstate());
}
```
The wireshark for the code is here: 
[samplec_code.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2540595/samplec_code.pcapng.txt)

As you can see in the end on the wireshark message, the `AL_Status` ends at `SAFEOP, ERROR` mode...

This is as far as I can get into getting this slave into `OPERATIONAL` mode. I saw an issue earlier about syncing clocks but I don't know much about how to do this...

Here's some system information you may find relevant:
`uname -a`:
```
Linux rahulbot 4.15.0-36-generic #39~16.04.1-Ubuntu SMP Tue Sep 25 08:59:23 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

`lscpu`:
```
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                8
On-line CPU(s) list:   0-7
Thread(s) per core:    2
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 158
Model name:            Intel(R) Core(TM) i7-7820HQ CPU @ 2.90GHz
Stepping:              9
CPU MHz:               1199.811
CPU max MHz:           3900.0000
CPU min MHz:           800.0000
BogoMIPS:              5808.00
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              8192K
NUMA node0 CPU(s):     0-7
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp flush_l1d
```

P.S. : Make sure you remove the `.txt` extensions from the xml/ shark file dumps to read them using text editor/ wireshark.

ArthurKetels, 2018-11-02:
Thanks for all the info provided. This makes giving an informed answer much easier.

First the simple_test program. The wireshark trace shows there are no errors in the communication. The master requests Operational but the Mecademic Meca500 robot never goes to to this state. Or at least not within 2 seconds. Aslo the robot does not signal a state error. So you could try to extend the simple_test timeout to 10s. The ESI file states 9000ms timeout for SafeOP->OP transition.

The Meca500 supports 2 operational modes, SM-synchronous and DC Sync0 synchronous. For first tests the SM-synchronous mode is the most simple. And I guess here is at least part of your problem. 

Simple_test has a very rudimentary process-data exchange. It sends a LRW every 50ms or so and then polls the slave for state transition. This is repeated until the state changes or the 2s timeout.

Most likely the robot wants to synchronize on the master LRW timing. But this timing must be within bounds. Unfortunately the manual of the Meca500 fails to provide these bounds so I have to guess. Mostly a 1ms cycle time will work.  When the slave sees enough LRW packets at a constant interval it will lock to that and release to Operational. You could change the timing and see if that solves it.

For a better understanding how to make a master that works under strict timing constraints see the red_test program.

Now to your samplec code. This is not the way how to make a master work with any slave. Again look at how simple_test or even better red_test does this.

symbioRahul, 2018-11-02:
Hi @ArthurKetels 
Thank you for the quick response. 

I have a few followup questions based on what you wrote earlier:
>The ESI file states 9000ms timeout for SafeOP->OP transition.
This is a very interesting find... 

Would this mean that I would be editing `ec_statecheck` parameters to allow for a 10s timeout ? 
If so how would this work in the `do while` loop since you'd want to be sending ec_send_processdata at every 1000us (which was previous at 2000us) and also polling `ec_statecheck` for 10s (previously EC_TIMEOUTSTATE, 2s) ? There could be something I'm not seeing...

I tried the following what I believe you suggested and changes the parameter for ec_receive_processdata to 1000us and received the following wireshark:

Hi @ArthurKetels 
Thank you for the quick response. 

I have a few followup questions based on what you wrote earlier:
>The ESI file states 9000ms timeout for SafeOP->OP transition.
This is a very interesting find... 

Would this mean that I would be editing `ec_statecheck` parameters to allow for a 10s timeout ? 
If so how would this work in the `do while` loop since you'd want to be sending ec_send_processdata at every 1000us (which was previous at 2000us) and also polling `ec_statecheck` for 10s (previously EC_TIMEOUTSTATE, 2s) ? There could be something I'm not seeing...

I tried the following what I believe you suggested and changes the parameter for ec_receive_processdata to 1000us and received the following wireshark:
[ethercatsimpletestmodified.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2544294/ethercatsimpletestmodified.pcapng.txt)

To be clear this is what i modified in the code:
```c
   64          printf("Request operational state for all slaves\n");
   65          expectedWKC = (ec_group[0].outputsWKC * 2) + ec_group[0].inputsWKC;
   66          printf("Calculated workcounter %d\n", expectedWKC);
   67          ec_slave[0].state = EC_STATE_OPERATIONAL;
   68          /* send one valid process data to make outputs in slaves happy*/
   69          ec_send_processdata();
   70          ec_receive_processdata(1000); // MODIFIED 2000 -> 1000
   71          /* request OP state for all slaves */
   72          ec_writestate(0);
   73          chk = 40;
   74          /* wait for all slaves to reach OP state */
   75          do
   76          {
   77             ec_send_processdata();
   78             ec_receive_processdata(1000); // MODIFIED 2000 -> 1000
   79             ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);
   80          }
```

ArthurKetels, 2018-11-03:
Your changes did not have the desired effect. When you would take te effort to look at your wireshark trace you see that the processdata interval is 50ms and not 1ms. This is easy to explain from your code `ec_statecheck(0, EC_STATE_OPERATIONAL, 50000);`. As long there is no state change it will keep polling for 50ms. You would want to change this timeout to 1000. Also you need at least 9000 loops instead of 40 to make the timeout of 9 sec.

Still it is better to follow the red_test example for proper timing. The state check loop in simple_test is not meant for timing critical processdata exchange.

symbioRahul, 2018-11-05:
Hi @ArthurKetels,

I did what you said an notice that there is a `SAFEOP, ERROR` state that occurs after a certain amount of time with both the `simple_test.c` (modified with the change you asked for) and `red_test.c` (which I modified to allow for 10,000,000us sleep while sending and receiving process data information in `ecatthread`). The error that occurs is a `Sync Manager Watchdog` error `(0x001b)`. Could this be caused by non-deterministic network jitter / lag ?

I also modified the `simple_test` loop such that `receive_process_timeout` and `ec_state_check` both have a 500us timeout (totaling ~1ms as verified by looking at the wireshark) and the same behavior occurs on both versions of `simple_test`.

`simple test` outputs:
```bash
SOEM (Simple Open EtherCAT Master)                  
Simple test               
Starting simple test      
ec_init on enp0s31f6 succeeded.                     
1 slaves found and configured.                      
Slaves mapped, state to SAFE_OP.                    
segments : 1 : 184 0 0 0  
Request operational state for all slaves            
Calculated workcounter 3  
Not all slaves reached operational state.           
Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog                                             

Request init state for all slaves                   
End simple test, close socket                       
End program     
```

`red_test` outputs:
SOEM (Simple Open EtherCAT Master)               
Redundancy test         
ifname2= 0x7ffd595ec8d1 
Starting Redundant test 
ecatcheck ptr = (nil)   
ec_init on enp0s31f6 succeeded.                  
1 slaves found and configured.                   
Slave:1 Name:Robots Output size:480bits Input size:992bits State: 4 delay:0.1                     
         Out:006242c0,  60 In:006242fc, 124      
Calculated workcounter 3                         
Request operational state for all slaves         
about to osal sleep     
Not all slaves reached operational state.        
Slave 1 State=0x14 StatusCode=0x001b : Sync manager watchdog                                      
Request safe operational state for all slaves    
End redundant test, close socket                 
End program  
```

wireshark for simple tests:
[simpletest2ms.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2550817/simpletest2ms.pcapng.txt)
[simpletest1ms.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2550816/simpletest1ms.pcapng.txt)

wireshark for red_test:
[redtestoutput.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2550818/redtestoutput.pcapng.txt)

symbioRahul, 2018-11-05:
accidential closing

ArthurKetels, 2018-11-06:
I think I have found a clue. 

In all wireshark files the LRW datagrams fail with WCK=0. All other datagrams work fine. The SM and FMMU are all configured correctly. The slave acknowledges this with a transition to safe-op. So we can be pretty sure the LRW should work and return at least 1 in safe-op and 3 in op. 

The only reason it can not is when the slave controller does not support LRW but need separate LRD and LWR datagrams. But then it should be mentioned in the EEPROM General data section. The print from simple_test shows that LRW are not disallowed by the slave.

I think this slave does not support LRW and need separate LRD LWR datagrams. You could either modify the EEPROM or work around it by setting the blockLRW=1 for this slave after config_init() but before config_map. This will force the use of LRD/LWR.

symbioRahul, 2018-11-06:
wow @ArthurKetels I think we're on to something here. I did what you said on both `simple_test` and `red_test`
_Both_ `LWR` and `LRD` is giving me a `WKC` of 0....
wire shark here: 
[simple_test_lwrlrd.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2555247/simple_test_lwrlrd.pcapng.txt)

I guess the question really is why..

P.S. one thing I am noticing is `LRD` is done at index `0x003c` and `LRW` is done to `0x0000`. `0x003c` is supposedly the inputs and `0x000` is supposedly the outputs. Is this (and ECAT in general) mapped with the slave in mind (i.e. "slave read at this address [0x003c], slave write at this address [0x0000]"). Because otherwise, it would seem as if this the input and output mappings are swapped... however, I'm guessing this is unlikely since all of the other `slaveinfo -map` dumps I've seen on this repo are similar to the Mecademic. 
Another bit I noticed in the `slaveinfo -map` dump for the Meca is that MBX protocol says `0x04` (which according to Section I of the Ethercat documentation should be FoE) and not `0x03` (CoE MBX type)... Is this correct?

symbioRahul, 2018-11-20:
I could not fix this issue yet. There are plans to contact Mecademic for more information on what is happening during boot.

nakarlsson, 2019-05-22:
Is this issue still active?

symbioRahul, 2019-05-23:
Hi @nakarlsson 
Unfortunately, this is still an unsolved and open issue for us. Currently, making any progress on Meca's EtherCAT drivers is blocked for many reasons. Feel free to close this if you want and we could consider reopening when we start active development on it.

closed 2019-05-23 by nakarlsson

simonedwards-cats3, 2019-10-10:
Hello SOEM community,

I'm new to EtherCAT and SOEM and have been trying to get my TI ICEv2 demonstration boards to work with SOEM.  I have two demo boards and one still has the existing code as programmed by TI.  From what I can tell, this demo board uses SSC v5.11 and WORKS with SOEM v1.3.3 simple_test.

I have re-programmed the other board to try out my own slave code using SSC v5.12.  This doesn't work out of the box with SOEM v1.3.3 but DOES work with Acontis ECMaster.  After reading other threads and posts about TI ESC + SSC v5.12 I was suspicious about the set up of the FMMU and SM.

_BTW 'works' means that I can run simple_test and achieve OP state as well as see process data being exchanged_

Long story short - adding the above patch to SOEM allows the TI ESC +SSC v5.12 combination to work.  In my view, there's something about v5.12 that SOEM doesn't like or work with.  If anyone from SOEM can add comment to this, I (and I'm sure others) would be very grateful.

One last thing, I have forced blockLRW bits in the slave and grouplist as well - this seems mandatory with TI ESC.

ArthurKetels, 2019-10-10:
It is not something in SOEM but in the slave. The EtherCAT standard clearly defines the proper settings of the EEPROM. There is a bug in the SSC v5.12 that generates wrong EEPROM files. ECmaster does not read the slave EEPROM but relies on the ESI file. The ESI file is correct and the slave goes to OP. Also the blockLRW should be set from the EEPROM. TI has a bug there.

So you have two options, work around the problem with a hack. Or, better, try to convince the people that introduced the bugs to solve them.

nakarlsson, 2019-10-10:
1.  have you patch’ed SSC5.12 with the available patches from ETG? At least one patch address the SM issue in SSC.

2. we have a second API that uses an overlapping IOmap, the take care of the LRW TI errata. Check the documentation.

simonedwards-cats3, 2019-10-10:
@nakarlsson 
I believe I have patched the code.  I followed these instructions from TI:
http://software-dl.ti.com/processor-industrial-sw/esd/docs/indsw/EtherCAT_Slave/01_00_07/PRU_ICSS_EtherCAT.html#generating-ethercat-sources-using-beckhoff-ssc-tool-option-2

As for other patches, I don't know where to get these from.  I've searched the members area of ETG for patches but cannot find any.

nakarlsson, 2019-10-10:
You’ll find info here : 
https://www.ethercat.org/memberarea/forum.asp?../memberarea/forum/free-download-slave-stack-code-v5-12_topic1590.html

simonedwards-cats3, 2019-10-10:
@nakarlsson 
Amazingly rapid and first class support!  Thank you.
I've removed the 'hack' to SOEM and applied the patch from the link you've given above (members only!) - the one posted 27 Jun 2018 at 7:49am.
It now works as expected.
Thanks again!