OpenEtherCATsociety/SOEM #225: Attempt to write to a read only object

schwarzp93, 2018-11-02:
Hi Guys,

I saw, that many issues referenced to this link: https://github.com/hefloryd/SOEM/commit/22d8f5d, so I tried to adapt it, to access my Festo CPX-FB37 bus node.

Unfortunately I get the following error:
SDO slave:1 index:6009.01 error:06010002 Attempt to write to a read only object

I'm not sure, what to do. Does anybody know, what to do?

Thanks in advance. Kind regards,
Philipp

mheden, 2018-11-02:
What is the question? Why you can't write to a read-only object? Or does the manual specify that you should be able to write to 0x6009:01?

schwarzp93, 2018-11-07:
sorry, if the question might have been a bit silly, I'm not really confident about SOEM and EtherCAT.

I think, I should be able to write to this.

The situation is the following. I'm using a Festo CPX-Fb37 bus node. It is connected to 3*8 digital IOs and 10 analog IOs. So slaveinfo -map gives the following output:
Slave:1
 Name:CPX-FB37
 Output size: 184bits
 Input size: 184bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000001d ID: 0029bf58 Rev: 00000005
 SM0 A:1200 L:1400 F:00010026 Type:1
 SM1 A:2200 L:1400 F:00010022 Type:2
 SM2 A:1000 L:  23 F:00010064 Type:3
 SM3 A:1100 L:  23 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  23 Lsb:0 Leb:7 Ps:1000 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000017 Ll:  23 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:0 3:0
 MBX length wr: 1400 rd: 1400 MBX protocols : 0e
 CoE details: 27 FoE details: 01 EoE details: 01 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7001:0x01 0x01 BOOLEAN      Output 0
  [0x0000.1] 0x7001:0x02 0x01 BOOLEAN      Output 1
  [0x0000.2] 0x7001:0x03 0x01 BOOLEAN      Output 2
  [0x0000.3] 0x7001:0x04 0x01 BOOLEAN      Output 3
  [0x0000.4] 0x7001:0x05 0x01 BOOLEAN      Output 4
  [0x0000.5] 0x7001:0x06 0x01 BOOLEAN      Output 5
  [0x0000.6] 0x7001:0x07 0x01 BOOLEAN      Output 6
  [0x0000.7] 0x7001:0x08 0x01 BOOLEAN      Output 7
  [0x0001.0] 0x7002:0x01 0x01 BOOLEAN      Output 0
  [0x0001.1] 0x7002:0x02 0x01 BOOLEAN      Output 1
  [0x0001.2] 0x7002:0x03 0x01 BOOLEAN      Output 2
  [0x0001.3] 0x7002:0x04 0x01 BOOLEAN      Output 3
  [0x0001.4] 0x7002:0x05 0x01 BOOLEAN      Output 4
  [0x0001.5] 0x7002:0x06 0x01 BOOLEAN      Output 5
  [0x0001.6] 0x7002:0x07 0x01 BOOLEAN      Output 6
  [0x0001.7] 0x7002:0x08 0x01 BOOLEAN      Output 7
  [0x0002.0] 0x7003:0x01 0x01 BOOLEAN      Output 0
  [0x0002.1] 0x7003:0x02 0x01 BOOLEAN      Output 1
  [0x0002.2] 0x7003:0x03 0x01 BOOLEAN      Output 2
  [0x0002.3] 0x7003:0x04 0x01 BOOLEAN      Output 3
  [0x0002.4] 0x7003:0x05 0x01 BOOLEAN      Output 4
  [0x0002.5] 0x7003:0x06 0x01 BOOLEAN      Output 5
  [0x0002.6] 0x7003:0x07 0x01 BOOLEAN      Output 6
  [0x0002.7] 0x7003:0x08 0x01 BOOLEAN      Output 7
  [0x0003.0] 0x7004:0x01 0x10 UNSIGNED16   Output 0
  [0x0005.0] 0x7005:0x01 0x10 UNSIGNED16   Output 0
  [0x0007.0] 0x7006:0x01 0x10 UNSIGNED16   Output 0
  [0x0009.0] 0x7007:0x01 0x10 UNSIGNED16   Output 0
  [0x000B.0] 0x7008:0x01 0x10 UNSIGNED16   Output 0
  [0x000D.0] 0x7009:0x01 0x10 UNSIGNED16   Output 0
  [0x000F.0] 0x700A:0x01 0x10 UNSIGNED16   Output 0
  [0x0011.0] 0x700B:0x01 0x10 UNSIGNED16   Output 0
  [0x0013.0] 0x700C:0x01 0x10 UNSIGNED16   Output 0
  [0x0015.0] 0x700D:0x01 0x10 UNSIGNED16   Output 0
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0017.0] 0x6001:0x01 0x01 BOOLEAN      Input 0
  [0x0017.1] 0x6001:0x02 0x01 BOOLEAN      Input 1
  [0x0017.2] 0x6001:0x03 0x01 BOOLEAN      Input 2
  [0x0017.3] 0x6001:0x04 0x01 BOOLEAN      Input 3
  [0x0017.4] 0x6001:0x05 0x01 BOOLEAN      Input 4
  [0x0017.5] 0x6001:0x06 0x01 BOOLEAN      Input 5
  [0x0017.6] 0x6001:0x07 0x01 BOOLEAN      Input 6
  [0x0017.7] 0x6001:0x08 0x01 BOOLEAN      Input 7
  [0x0018.0] 0x6002:0x01 0x01 BOOLEAN      Input 0
  [0x0018.1] 0x6002:0x02 0x01 BOOLEAN      Input 1
  [0x0018.2] 0x6002:0x03 0x01 BOOLEAN      Input 2
  [0x0018.3] 0x6002:0x04 0x01 BOOLEAN      Input 3
  [0x0018.4] 0x6002:0x05 0x01 BOOLEAN      Input 4
  [0x0018.5] 0x6002:0x06 0x01 BOOLEAN      Input 5
  [0x0018.6] 0x6002:0x07 0x01 BOOLEAN      Input 6
  [0x0018.7] 0x6002:0x08 0x01 BOOLEAN      Input 7
  [0x0019.0] 0x6003:0x01 0x01 BOOLEAN      Input 0
  [0x0019.1] 0x6003:0x02 0x01 BOOLEAN      Input 1
  [0x0019.2] 0x6003:0x03 0x01 BOOLEAN      Input 2
  [0x0019.3] 0x6003:0x04 0x01 BOOLEAN      Input 3
  [0x0019.4] 0x6003:0x05 0x01 BOOLEAN      Input 4
  [0x0019.5] 0x6003:0x06 0x01 BOOLEAN      Input 5
  [0x0019.6] 0x6003:0x07 0x01 BOOLEAN      Input 6
  [0x0019.7] 0x6003:0x08 0x01 BOOLEAN      Input 7
  [0x001A.0] 0x6004:0x01 0x10 UNSIGNED16   Input 0
  [0x001C.0] 0x6005:0x01 0x10 UNSIGNED16   Input 0
  [0x001E.0] 0x6006:0x01 0x10 UNSIGNED16   Input 0
  [0x0020.0] 0x6007:0x01 0x10 UNSIGNED16   Input 0
  [0x0022.0] 0x6008:0x01 0x10 UNSIGNED16   Input 0
  [0x0024.0] 0x6009:0x01 0x10 UNSIGNED16   Input 0
  [0x0026.0] 0x600A:0x01 0x10 UNSIGNED16   Input 0
  [0x0028.0] 0x600B:0x01 0x10 UNSIGNED16   Input 0
  [0x002A.0] 0x600C:0x01 0x10 UNSIGNED16   Input 0
  [0x002C.0] 0x600D:0x01 0x10 UNSIGNED16   Input 0

You can see the 3*8 digital an 10 analog IOs there. That's why I thought, I could write values here...

nakarlsson, 2018-11-07:
Are you trying to read/write processdata?

https://github.com/hefloryd/SOEM/commit/22d8f5d, is about howto do custom configuration of slaves, do you need to do that? 

To access processdata you should look in samples on handling of the IOmap,ex_slave[x]->inputs/->outputs and send/recv processdata.

schwarzp93, 2018-11-08:
Hey, thanks for your answer!

Yes, I was trying to write processdata... As I said, I'm really struggling with SOEM, so sorry, for the silly question.

Thanks, I will try what you proposed. Do you have any tutorials or examples besides the "official" one?

nakarlsson, 2018-11-08:
The basic can be found here, just dig into code, docs and old issues.

schwarzp93, 2018-11-14:
Hey again!

I made it work :)
The only problem that remains, is that I don't fully understand, how to define a set of output-values and decide, which value reaches which output.

As you can see in the slaveinfo-map output I posted above, it lists 3*8 digital outputs (type BOOLEAN) and 10 analog outpus (type UNSIGNED16). This is analog to the configuration I can see if I try to access the device via a TwinCAT on Windows (3 Modules digital, each 8 Outputs. 10 Modules analog, each one output).
That's why I thought, I should declare an set of output values as followed:

typedef struct PACKED
{
	boolean output1_0;
	boolean output1_1;
	boolean output1_2;
	boolean output1_3;
	boolean output1_4;
	boolean output1_5;
	boolean output1_6;
	boolean output1_7;

	boolean output2_0;
	boolean output2_1;
	boolean output2_2;
	boolean output2_3;
	boolean output2_4;
	boolean output2_5;
	boolean output2_6;
	boolean output2_7;

	boolean output3_0;
	boolean output3_1;
	boolean output3_2;
	boolean output3_3;
	boolean output3_4;
	boolean output3_5;
	boolean output3_6;
	boolean output3_7;
	
	uint16 output4;
	uint16 output5;
	uint16 output6;
	uint16 output7;
	uint16 output8;
	uint16 output9;
	uint16 output10;
	uint16 output11;
	uint16 output12;
	uint16 output13;
}output_philipp;

Unfortunately, the order of the values doesn't correspond to the order of outputs I know from the windows application. It seems pretty random.
I hope you guys get what I#m trying to explain... :)

Is there some rule to define a set of output values?

Thanks a lot for your help here!

Regards,
Philipp

nakarlsson, 2018-11-14:
In the IOmap a boolean is 1 bit 

Ex this two values only take on 2 bit in a byte.
[0x0000.0] 0x7001:0x01 0x01 BOOLEAN Output 0
[0x0000.1] 0x7001:0x02 0x01 BOOLEAN Output 1

For you that would require to use bitfields in the struct 

typedef struct PACKED
{
uint8 output1_0 : 1
uint8 output1_1 : 1
uint8 output1_2 : 1
...

schwarzp93, 2018-11-14:
That worked well, thanks!

But I still can't access the analog outputs... How would you define them in the output-definition?

schwarzp93, 2018-11-14:
sorry, mad a silly mistake :)
This also works!

Again, thanks a lot for your help here!

nakarlsson, 2018-11-16:
You seems to have made progress so I'll close this issue, write a new issue if needed that address new questions that might arise.

closed 2018-11-16 by nakarlsson