OpenEtherCATsociety/SOEM #229: Add EoE write/read/setIP/getIP component

nakarlsson, 2018-11-13:


nakarlsson, 2018-11-13:
@ArthurKetels , I'm working on basic SOES/SOEM EoE support. It is pretty straight forward except the case mentioned in #217. In current implementation it is safe to just drop and try to fetch the CoE frame again. But adding EoE to the equation would need some dispathing. Any thoughts or ideas?

ArthurKetels, 2018-11-13:
Great news. 

I think we need to implement a mailbox dispatcher anyway. Just like SOES does it. 

Because I would like to send a pull request for a VoE scope function (high speed parameter streaming) in SOES. And this needs it's counterpart in SOEM. 

Also we have AoE and FoE. Those can't be used in a running application right now. Then we have the real time contention issue with simultaneous process data and CoE.

To me the best solution would be to implement a FMMU on SM1 status  in each slave (provided the slave has a spare FMMU) and read the flags in the process data task. 

The actual reading and writing of slave mailboxes would be implemented in the process data task too. So we need an asynchronous FIFO buffer in both directions to the main task doing the CoE EoE etc.

The CoE functions are then to be rewritten to connect to the FIFO buffers instead of sending packets directly. The EoE functions can be implemented very easy this way.

When the EtherCAT state is pre-op the process data task can be changed to dispatch the FIFO buffers more quickly (no process data to send). This way the configuration phase can still be handled quickly.

I need some more thought how to implement the FIFO buffers. Do we have only two (in out) or do we split per XoE? To me it does not make sense to split per slave, so a header needs to be attached to the FIFO elements to identify source, destination, type and size.

When reading the above I realise this is a BIG change in the internals.

nakarlsson, 2018-11-13:
Could you give me an preview of the VoE send in SOES, I've a draft of SOES EoE including a function ESC_eoeprocess_tx. soes branch "feature/eoe"

bascially it does

1 ESC_eoeprocess_tx claims a Mbx
2. ESC_eoeprocess_tx is called from the main loop. (the alt is to make sure to guard the internals)

   /* Check mailboxes */
   if (ESC_mbxprocess())
   {
      ESC_coeprocess();
      ESC_foeprocess();
      ESC_eoeprocess();
      ESC_xoeprocess();
   }
   ESC_eoeprocess_tx();

nakarlsson, 2019-01-31:
basic support added

closed 2019-01-31 by nakarlsson