OpenEtherCATsociety/SOEM #232: include ec_mbxerror2string in the header

nicolapedrocchi, 2018-11-23:
Hi, 
just as very small enhancement of the project, is it possible to include the declaration of ec_mbxerror2string in ethercatprint.h ? 
Just to improve the user-diagnostic when the raised error is of type EC_ERR_TYPE_MBX_ERROR

Thanks
n.

nakarlsson, 2018-11-24:
what is the use-case? How do you use? It should go into the header or turned static.

nicolapedrocchi, 2018-11-24:
Hi, thanks for the prompt answer
I am using SOEM build libraries in a ROS application.
I push all the errors in a diagnostic message, to ease the user comprehension of what is happening. 

Actually, it's never happen that a EC_ERR_TYPE_MBX_ERROR is caught, but it is a pity not having the description of the error available at user level when is already implemented in the cpp file.
 
I guess that just having the declaration in the header is ok.

Many thanks
Nicola

commit c892921d (2019-06-12): Add ec_error to string print function, fixes #227 and fixes #232
closed 2019-06-12 by ArthurKetels