OpenEtherCATsociety/SOEM #235: Workcounter issue with simple test and Beckhoff IO modules

primarchhorus, 2018-12-03:
Hi There

I have been attempting to get some software running on a system that comprises of a Beckhoff C6930-0050 Industrial PC and 2 EtherCat modules a EP2008-0001 and a EP1008-0001. Previously the software i have that uses SOEM as the ethercat communication library was successfully running on a Beckhoff CX Embedded PC setup with IO modules connected via the CX back plane setup. Porting this software directly to the new setup showed that the devices were being recognized but no data was being written to or from the IO devices. To look into it further i ran simple_test and it showed that both devices were successfully configured and it indicated that all slaves had reached operational state however it would repeatedly print out "OK : all slaves resumed OPERATIONAL.". After looking around in previous issues that seemed similar i added a work counter print out to simple_test and this was the result.

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp4s0 succeeded.
2 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Operational state reached for all slaves.
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
MESSAGE : slave 1 reconfigured
wkc: -1 expectedWKC: 0
MESSAGE : slave 2 reconfigured
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
WARNING : slave 1 is in SAFE_OP, change to OPERATIONAL.
WARNING : slave 2 is in SAFE_OP, change to OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.
wkc: -1 expectedWKC: 0
wkc: -1 expectedWKC: 0
OK : all slaves resumed OPERATIONAL.

And after further digging around i saw that someone with a similair issue had had success by setting blockLRW on the slave to 1. So i added this bit of code into simple test after the ec_config_init and before the ec_config_map.

         for (int i = 1; i <= ec_slavecount; i++)
         {
                ec_slave[i].blockLRW = 1;
         }
This seemed to slightly improve the issue as the new output of simple_test was now.

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp4s0 succeeded.
2 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 2 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
wkc: 2 expectedWKC: 3
OK : all slaves resumed OPERATIONAL.
wkc: 2 expectedWKC: 3
OK : all slaves resumed OPERATIONAL.
wkc: 2 expectedWKC: 3
OK : all slaves resumed OPERATIONAL.
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3
wkc: 2 expectedWKC: 3

And when i add the blockLRW to my own software i am able to set outputs on the EP2008-0001 module but still get no data from the input module EP1008-0001 and i would expect that the workcounter should be 3 so there is still something up with this module. So as had been suggested in previous tickets i have attached a wireshark trace of the NIC while simple_test is running and hopefully this is enough information to help me figure out what the issue with this module is.

[simple_test_wireshark.pcap.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2641583/simple_test_wireshark.pcap.zip)

Thanks.

nakarlsson, 2018-12-04:
What happens if you don't init DC? Just to see if the frame length matter. From the log you can see that you drop/loose alot of the frames, sent frames without response frames with wkc !=0 and modifed src MAC. Is the wireshark taken from the master PC?
Is it related to https://github.com/OpenEtherCATsociety/SOEM/issues/178 , given you run Linux.

ArthurKetels, 2018-12-04:
Indeed I think it is related to the kernel bug described in #178. All packets >60 bytes do not return from the slave to the master because they have too many padding bytes added. Up or downgrade your kernel (or use vanilla kernel).

primarchhorus, 2018-12-04:
Hi @nakarlsson taking out ec_configdc() did the trick, it all works with simple_test and my own software, @ArthurKetels I will look at #178  just to be sure i know which kernel versions are troublesome. Thank you for the help and the quick reply.

primarchhorus, 2018-12-04:
Just another FYI, i added another module to the system after i had the first 2 working a EP1816-3008 and the issue returned with the third module in the chain, so i updated my kernel from 4.4 to 4.15 and everything seems good now, i will close this issue.

closed 2018-12-04 by primarchhorus