OpenEtherCATsociety/SOEM #236: Distributed Clocks Api

prabhakarlad, 2018-12-06:
I have implemented a acyclic example on lan9252 evb [1],  and enabled DC on all the slaves and had a couple of questions on usage:
 a] Is there an api  to say use my first slave as clock master or is it by default the first slave ?
 b] How to write time delay on individual slave devices ?
 c] In the ec_sync() the dt calculated for the entire system ?
 d] Is there a way to calculate individual slave dt ?
 e] Is ec_DCtime distributed to all the slaves on every ec_send_processdata() call ?

[1] https://github.com/prabhakarlad/soem-lan9252-hbiplus/blob/master/simple_test.c

nakarlsson, 2018-12-07:
a, not explicitly but ecx_configdc will use the first slave supporting DC
b, you can access the DC registers of the slave via FPRD/FPWR or use the current API to enable SYNC0,1. Beware to mess up calculated data from ecx_configdc like offsets and propagation delays
c, it uses the DC ref slave system time
d, you would need to access that slave explicitly via register read/write
e. yes, if you started dc via ecx_configdc

nakarlsson, 2018-12-18:
I'll close this, if any specific issue remains write a new issue.

closed 2018-12-18 by nakarlsson