OpenEtherCATsociety/SOEM #241: SDO size

hwstrijker, 2019-01-03:
I have a question, is there any way to get the size and read/write access of the sdo from a slave? A kind of SDObytes object like the Ibytes and Obytes for the pdo. So I can loop all sdo and get the description and access of every sdo entity and print it for the user. 

I expect something like: 
```
int sdo_size = ec_slave[1].SDObytes ;
for(i=0;i<sdo_size;i++)
	printf(“entity 0x%04X, index 0x%02X, access %c, name %s, has value %0X \n”, …);
```

So I get a autoput in the same style as in the objectlist of SOES, like:
entity 0x1000, index 0x00, access RO, name Device Type ,has value 0x00001389
entity 0x1018, index 0x00, access RO, name Max SubIndex ,has value 0x04
entity 0x1018, index 0x01, access RO, name Vendor ID, has value 0x00001337
…

I know the same information can be found in the slave config.

nakarlsson, 2019-01-03:
Check slaveinfo with -map -sdo switches?

hwstrijker, 2019-01-04:
Yes that is where I was looking for, thank you

nakarlsson, 2019-01-04:
Ok, I’ll close this

closed 2019-01-04 by nakarlsson