OpenEtherCATsociety/SOEM #242: SOEM Master fail to change the state from SAFEOP to OP in simple_test project under Windows OS.

WangKaimeng, 2019-01-08:
In recent days,  I tried to compile the soem simple_test project on windows, the state of master is always stay at SAFEOP, then i tried the soem SDK, the situation is same. Beside, the error code is 0x0000 and no error. I doubt if the slave or object dictionary affect it, so i try TWINCAT master and it works well. There are two questions about these.
1. Whether SOEM master can communication with ET1100 Slave;
2. Whther SOEM master is effected by XML file and slave code; 
[image]

nakarlsson, 2019-01-08:
Does the slave pass the EtherCAT Conformance Test?

WangKaimeng, 2019-01-08:
I don't check the Ethercat conformance, but i tried it with twincat and it works.

WangKaimeng, 2019-01-08:
> Does the slave pass the EtherCAT Conformance Test?

I don't check the Ethercat conformance, but i tried it with twincat and it works

nakarlsson, 2019-01-08:
Yes but SOEM and TwinCAT don't work exactly the same, when the slave pass CTT you can assume it should work with SOEM also out-of-the-box.

WangKaimeng, 2019-01-08:
> Yes but SOEM and TwinCAT don't work exactly the same, when the slave pass CTT you can assume it should work with SOEM also out-of-the-box.

well, it may not be Master error. Could you tell me how to check this issue. Should I change the slave code or object dictionary?

nakarlsson, 2019-01-08:
Given you run SSC 5.12, you might wanna check that you don't suffer from the root cause of these issues, https://github.com/OpenEtherCATsociety/SOEM/issues/214

WangKaimeng, 2019-01-08:
> SSC 5.12

thanks a lot!
 i do generate the slave code with SSC 5.12

WangKaimeng, 2019-01-09:
hello nakarlsson!
Should I  update slave EEPROM before running the simple_test, and what XML file should  I choose? I updated EEPROM using xml file which TI example provided before.

WangKaimeng, 2019-01-09:
> Given you run SSC 5.12, you might wanna check that you don't suffer from the root cause of these issues, #214

hello nakarlsson!
Should I update slave EEPROM before running the simple_test, and what XML file should I choose? I updated EEPROM using xml file which TI example provided before.

nakarlsson, 2019-01-09:
From a SOEM perspective just design a conformant slave, best way to test that is to run CTT.

WangKaimeng, 2019-01-09:
> From a SOEM perspective just design a conformant slave, best way to test that is to run CTT.
hello nakarlsson!
To be honest, I am not ETG member, so I am so sorry that I can't get CTT tools to conduct compliant test.
When I changed the SSC Vesion from 5.12 to 5.11 acoording to #214 ,the status code changed to 0x001E (Invalid input mapping) .
Then I see the master code , and  I discovered that ec_slave[0] is reserved for master, but at function simple_test,  ec_slave[0] is always used to read the slave state, as i understand, ec_slave should begin with ec_slave[1]. Acording to my idea, i change the code, and the state can change to OP, however, it always pop up message: salve reconfigured. 
Thank you in advance!

WangKaimeng, 2019-01-23:
> From a SOEM perspective just design a conformant slave, best way to test that is to run CTT.

I may find the same issue in #142, 
[image]
with your advice, I program the code, but I don't know how to determine period_us and calc_and_copy values at ec_dcsync0(slave, TRUE, period_us * 1000, calc_and_copy);
Could you help me? Thank you in advance!

nakarlsson, 2019-04-29:
Consult the slave reference manual for values concerning slave specific DC values

closed 2019-04-29 by nakarlsson