OpenEtherCATsociety/SOEM #245: Strange behavior when trying to use SOEM with Elmo Gold Twitters

wieset, 2019-01-21:
Hi all,

I am trying to use SOEM with some Elmo Gold Twitter drives. For now, I am using SDO communication only. When trying to command a velocity to a drive, sometimes it works as expected, but most of the time it doesn't move. Sometimes, it even rotates with maximum velocity in negative direction.

I have adapted the simple_test program for my needs. I bring the drive to Operation Enabled state and set the Mode of Operation to Profiled Velocity. After that, I am setting the Target Velocity. In the "good" case, a non-zero Velocity Demand value is generated and the motor rotates. However, in the "bad" case, the drive controller does not react accordingly and the Velocity Demand value stays at zero.

Here is the output log for the "good" case:

```
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp1s0 succeeded.
20 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 400 0 0 0
Request operational state for all slaves
Calculated workcounter 60
Operational state reached for all slaves.

Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0218 (536)	[Statusword]
Write to slave 4 at 0x6040:0 => wkc: 1; data: 0x0080	{Fault Reset}
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0250 (592)	[Statusword]
Write to slave 4 at 0x6040:0 => wkc: 1; data: 0x0006	{Enable Voltage}
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0231 (561)	[Statusword]
Write to slave 4 at 0x6040:0 => wkc: 1; data: 0x0007	{Switch On}
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0233 (563)	[Statusword]
Write to slave 4 at 0x6040:0 => wkc: 1; data: 0x000f	{Enable Operation}
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0237 (567)	[Statusword]

Read from slave 4 at 0x6502:0 => wkc: 1; data: 0x000003ed (1005)	[Supported drive modes]
Write to slave 4 at 0x6060:0 => wkc: 1; data: 0x03	{Mode of Operation}
Read from slave 4 at 0x6061:0 => wkc: 1; data: 0x03 (3)	[Mode of Operation Display]

Read from slave 4 at 0x606d:0 => wkc: 1; data: 0x0064 (100)	[Velocity Window]
Read from slave 4 at 0x606d:0 => wkc: 1; data: 0x0064 (100)	[Velocity Threshold]
Read from slave 4 at 0x607f:0 => wkc: 1; data: 0x00010aab (68267)	[Max profile velocity]
Read from slave 4 at 0x6080:0 => wkc: 1; data: 0xffffffff (-1)	[Max motor speed]
Read from slave 4 at 0x6083:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Profile acceleration]
Read from slave 4 at 0x6084:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Profile deceleration]
Read from slave 4 at 0x6085:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Quick stop deceleration]
Read from slave 4 at 0x60c5:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Max acceleration]
Read from slave 4 at 0x60c6:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Max deceleration]

Read from slave 4 at 0x6040:0 => wkc: 1; data: 0x000f (15)	[Control word]
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0237 (567)	[Statusword]
Write to slave 4 at 0x60ff:0 => wkc: 1; data: 0x00001000	{Target Velocity}
Read from slave 4 at 0x60ff:0 => wkc: 1; data: 0x00001000 (4096)	[Target Velocity]
Read from slave 4 at 0x6083:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Profile acceleration]
Read from slave 4 at 0x60c5:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Max acceleration]
Read from slave 4 at 0x606b:0 => wkc: 1; data: 0x00001000 (4096)	[Velocity Demand Value]
Read from slave 4 at 0x606c:0 => wkc: 1; data: 0x00000f41 (3905)	[Velocity Actual Value]
Read from slave 4 at 0x6069:0 => wkc: 1; data: 0x00000f68 (3944)	[Velocity Sensor Actual Value]

Read from slave 4 at 0x6040:0 => wkc: 1; data: 0x000f (15)	[Control word]
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0237 (567)	[Statusword]
Write to slave 4 at 0x60ff:0 => wkc: 1; data: 0x00001000	{Target Velocity}
Read from slave 4 at 0x60ff:0 => wkc: 1; data: 0x00001000 (4096)	[Target Velocity]
Read from slave 4 at 0x6083:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Profile acceleration]
Read from slave 4 at 0x60c5:0 => wkc: 1; data: 0x7fffffff (2147483647)	[Max acceleration]
Read from slave 4 at 0x606b:0 => wkc: 1; data: 0x00001000 (4096)	[Velocity Demand Value]
Read from slave 4 at 0x606c:0 => wkc: 1; data: 0x000010cd (4301)	[Velocity Actual Value]
Read from slave 4 at 0x6069:0 => wkc: 1; data: 0x00001024 (4132)	[Velocity Sensor Actual Value]
```

And for the "bad" case:

```
SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp1s0 succeeded.
20 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 400 0 0 0
Request operational state for all slaves
Calculated workcounter 60
Operational state reached for all slaves.

Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0250 (592)	[Statusword]
Write to slave 4 at 0x6040:0 => wkc: 1; data: 0x0080	{Fault Reset}
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0250 (592)	[Statusword]
Write to slave 4 at 0x6040:0 => wkc: 1; data: 0x0006	{Enable Voltage}
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0231 (561)	[Statusword]
Write to slave 4 at 0x6040:0 => wkc: 1; data: 0x0007	{Switch On}
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0233 (563)	[Statusword]
Write to slave 4 at 0x6040:0 => wkc: 1; data: 0x000f	{Enable Operation}
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x0637 (1591)	[Statusword]

Read from slave 4 at 0x6502:0 => wkc: 1; data: 0x000003ed (1005)	[Supported drive modes]
Write to slave 4 at 0x6060:0 => wkc: 1; data: 0x03	{Mode of Operation}
Read from slave 4 at 0x6061:0 => wkc: 1; data: 0x03 (3)	[Mode of Operation Display]

Read from slave 4 at 0x606d:0 => wkc: 1; data: 0x0064 (100)	[Velocity Window]
Read from slave 4 at 0x606d:0 => wkc: 1; data: 0x0064 (100)	[Velocity Threshold]
Read from slave 4 at 0x607f:0 => wkc: 1; data: 0x00010aab (68267)	[Max profile velocity]
Read from slave 4 at 0x6080:0 => wkc: 1; data: 0xffffffff (-1)	[Max motor speed]
Read from slave 4 at 0x6083:0 => wkc: 1; data: 0x00000001 (1)	[Profile acceleration]
Read from slave 4 at 0x6084:0 => wkc: 1; data: 0x00000001 (1)	[Profile deceleration]
Read from slave 4 at 0x6085:0 => wkc: 1; data: 0x00000001 (1)	[Quick stop deceleration]
Read from slave 4 at 0x60c5:0 => wkc: 1; data: 0x00000001 (1)	[Max acceleration]
Read from slave 4 at 0x60c6:0 => wkc: 1; data: 0x00000001 (1)	[Max deceleration]

Read from slave 4 at 0x6040:0 => wkc: 1; data: 0x000f (15)	[Control word]
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x1637 (5687)	[Statusword]
Write to slave 4 at 0x60ff:0 => wkc: 1; data: 0x00001000	{Target Velocity}
Read from slave 4 at 0x60ff:0 => wkc: 1; data: 0x00001000 (4096)	[Target Velocity]
Read from slave 4 at 0x6083:0 => wkc: 1; data: 0x00000001 (1)	[Profile acceleration]
Read from slave 4 at 0x60c5:0 => wkc: 1; data: 0x00000001 (1)	[Max acceleration]
Read from slave 4 at 0x606b:0 => wkc: 1; data: 0x00000000 (0)	[Velocity Demand Value]
Read from slave 4 at 0x606c:0 => wkc: 1; data: 0x00000000 (0)	[Velocity Actual Value]
Read from slave 4 at 0x6069:0 => wkc: 1; data: 0x00000000 (0)	[Velocity Sensor Actual Value]

Read from slave 4 at 0x6040:0 => wkc: 1; data: 0x000f (15)	[Control word]
Read from slave 4 at 0x6041:0 => wkc: 1; data: 0x1637 (5687)	[Statusword]
Write to slave 4 at 0x60ff:0 => wkc: 1; data: 0x00001000	{Target Velocity}
Read from slave 4 at 0x60ff:0 => wkc: 1; data: 0x00001000 (4096)	[Target Velocity]
Read from slave 4 at 0x6083:0 => wkc: 1; data: 0x00000001 (1)	[Profile acceleration]
Read from slave 4 at 0x60c5:0 => wkc: 1; data: 0x00000001 (1)	[Max acceleration]
Read from slave 4 at 0x606b:0 => wkc: 1; data: 0x00000000 (0)	[Velocity Demand Value]
Read from slave 4 at 0x606c:0 => wkc: 1; data: 0x00000000 (0)	[Velocity Actual Value]
Read from slave 4 at 0x6069:0 => wkc: 1; data: 0x00000000 (0)	[Velocity Sensor Actual Value]
```

Interestingly, in the "good" case, the Acceleration values are at the default value of 0x7fffffff, as expected. However, in the "bad" case, these values are suddenly set to 0x00000001. These values are never written, so they should stay at their default. In addition, in the "bad" case, the Target Reached bit in the Statusword is set, which also makes no sense, since the Target Velocity is clearly not reached if the motor is not moving.

I am a bit at a loss to what's going on here. Is it an issue with my program, with SOEM or with the Elmos? I would appreciate any help.

ArthurKetels, 2019-01-22:
Slaves are not supposed to be controlled with SDO mailbox communication. You should use SDO communication. Look at the example code, and the documentation of SOEM. SDO is only used for setting and reading parameters non-realtime.

wieset, 2019-01-31:
I finally found the error. It had nothing to do with SDO or PDO communication. The max torque value was too low, so that the motor sometimes did not start moving. Fixed now. I don't really see how this might have caused the strange acceleration values above, though, but does not matter for now.

closed 2019-01-31 by wieset

DocTU, 2019-02-25:
Hi，I also have some problems in dealing with elmo and soem，could you please share your code ？

muenchenstrasse, 2020-03-20:
> I finally found the error. It had nothing to do with SDO or PDO communication. The max torque value was too low, so that the motor sometimes did not start moving. Fixed now. I don't really see how this might have caused the strange acceleration values above, though, but does not matter for now.

What parameters did you set? Apparently I am also having the same issue with my ELMOs.

cemtolgamunyas, 2024-07-02:
Did you change the update rate in controller manager to change the starting torque of the motor?

tejas2292, 2024-08-28:
@wieset thank you so much for mentioning the solution as well.

wieset, 2024-08-30:
> What parameters did you set? Apparently I am also having the same issue with my ELMOs.

I'm setting SDO object 0x6073 (max current), but you could also use object 0x6072 (max torque)

wieset, 2024-08-30:
> Did you change the update rate in controller manager to change the starting torque of the motor?

No I did not

tejas2292, 2024-08-30:
@wieset do you have working output log for profile position mode? or steps for profile position mode. Thank you in advance

wieset, 2024-08-30:
> @wieset do you have working output log for profile position mode? or steps for profile position mode. Thank you in advance

Unfortunately I don't. I haven't been working on this since a while.