OpenEtherCATsociety/SOEM #247: Encoder cannot switch to OP state

catfacedck, 2019-01-29:
Hello Everyone:

I am an EtherCAT newbie so please bear with me.
I am attempting to use a rotary encoder with SOEM. My test bed consists of a few Beckhoff devices together with the rotary encoders. The rotary encoders have passed the in-house conformance test tool. I do not observe any issues with the Beckhoff slaves using SOEM.

Here is a screen shot from simple test.
[image]

Here are the results from slaveinfo -map and slaveinfo -pdo along with the wireshark for simpletest. Also attached is a docx file of same. Slaves 4 and 5 are the same identical rotary encoders at the bottom of the docx file. I am unsure how to troubleshoot this problem. Any assistance would be greatly appreciated. Thank you.

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s25 succeeded.
5 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 4 State=12 StatusCode=  1e : Invalid input configuration
Slave 5 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:EK1100
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.1.1.0
 Configured address: 1001
 Man: 00000002 ID: 044c2c52 Rev: 00120000
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: -2000[mA]
 only LRD/LWR:0
PDO mapping according to SII :

Slave:2
 Name:EL5002
 Output size: 0bits
 Input size: 96bits
 State: 4
 Delay: 145[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1002
 Man: 00000002 ID: 138a3052 Rev: 00130000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1080 L: 128 F:00010022 Type:2
 SM2 A:1100 L:   0 F:00000004 Type:3
 SM3 A:1180 L:  12 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:  12 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
 FMMUfunc 0:2 1:3 2:0 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 03 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 130[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x6000:0x01 0x01 BOOLEAN      Data error
  [0x0000.1] 0x6000:0x02 0x01 BOOLEAN      Frame error
  [0x0000.2] 0x6000:0x03 0x01 BOOLEAN      Power failure
  [0x0000.3] 0x0000:0x00 0x0A
  [0x0001.5] 0x1C32:0x20 0x01
  [0x0001.6] 0x1800:0x07 0x01 BOOLEAN      TxPDO State
  [0x0001.7] 0x1800:0x09 0x01 BOOLEAN      TxPDO Toggle
  [0x0002.0] 0x6000:0x11 0x20 UNSIGNED32   Counter value
  [0x0006.0] 0x6010:0x01 0x01 BOOLEAN      Data error
  [0x0006.1] 0x6010:0x02 0x01 BOOLEAN      Frame error
  [0x0006.2] 0x6010:0x03 0x01 BOOLEAN      Power failure
  [0x0006.3] 0x0000:0x00 0x0A
  [0x0007.5] 0x1C32:0x20 0x01
  [0x0007.6] 0x1801:0x07 0x01 BOOLEAN      TxPDO State
  [0x0007.7] 0x1801:0x09 0x01 BOOLEAN      TxPDO Toggle
  [0x0008.0] 0x6010:0x11 0x20 UNSIGNED32   Counter value

Slave:3
 Name:EL6751
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 290[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1003
 Man: 00000002 ID: 1a5f3052 Rev: 00160000
 SM0 A:1000 L:1024 F:00010026 Type:1
 SM1 A:1400 L:1024 F:00010022 Type:2
 SM2 A:1800 L:   0 F:00000026 Type:3
 SM3 A:2400 L:   0 F:00000022 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 1024 rd: 1024 MBX protocols : 0d
 CoE details: 23 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 230[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x77352D80.0] 0x0000:0x00 0x0B
  [0x77352D81.3] 0xF101:0x0C 0x01 BOOLEAN      SYNC Toggle
  [0x77352D81.4] 0xF101:0x0D 0x01 BOOLEAN      Device Diag
  [0x77352D81.5] 0x0000:0x00 0x01
  [0x77352D81.6] 0xF101:0x0F 0x01 BOOLEAN      PDO Toggle
  [0x77352D81.7] 0xF101:0x10 0x01 BOOLEAN      PDO State
  [0x77352D82.0] 0xF101:0x11 0x10 UNSIGNED16   Cycle Counter
  [0x77352D84.0] 0xF101:0x12 0x08 UNSIGNED8    Slave Status Counter
  [0x77352D85.0] 0x0000:0x00 0x08
  [0x77352D86.0] 0xF101:0x14 0x10 UNSIGNED16   Cycle Time
  [0x77352D88.0] 0x0000:0x00 0x10
  [0x77352D8A.0] 0xF108:0x21 0x08 UNSIGNED8    RX error counter
  [0x77352D8B.0] 0xF108:0x22 0x08 UNSIGNED8    TX error counter
  [0x77352D8C.0] 0xF108:0x01 0x01 BOOLEAN      Bus-Off
  [0x77352D8C.1] 0xF108:0x02 0x01 BOOLEAN      warning limit reached
  [0x77352D8C.2] 0xF108:0x03 0x01 BOOLEAN      RX overflow
  [0x77352D8C.3] 0x0000:0x00 0x01
  [0x77352D8C.4] 0xF108:0x05 0x01 BOOLEAN      TX overflow
  [0x77352D8C.5] 0xF108:0x06 0x01 BOOLEAN      Ack error
  [0x77352D8C.6] 0x0000:0x00 0x02
  [0x77352D8D.0] 0x0000:0x00 0x08
  [0x77352D8E.0] 0x0000:0x00 0x10

Slave:4
 Name:? M:00ecadc0 I:adec0578
 Output size: 0bits
 Input size: 216bits
 State: 18
 Delay: 1025[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1004
 Man: 00ecadc0 ID: adec0578 Rev: 00010000
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1200 L: 512 F:00010022 Type:2
 FMMU0 Ls:0000000c Ll:  27 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 0e
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x000C.0] 0x6004:0x00 0x20 UNSIGNED32   Position value
  [0x0010.0] 0x6030:0x01 0x10 INTEGER16    Speed value channel 1
  [0x0012.0] 0x6503:0x00 0x10 UNSIGNED16   Alarms
  [0x0014.0] 0x6505:0x00 0x10 UNSIGNED16   Warnings
  [0x0016.0] 0x1001:0x00 0x08 UNSIGNED8    Error Register
  [0x0017.0] 0x6008:0x00 0x40 UNSIGNED64   High resolution position value
  [0x001F.0] 0x6030:0x01 0x10 INTEGER16    Speed value channel 1
  [0x0021.0] 0x6503:0x00 0x10 UNSIGNED16   Alarms
  [0x0023.0] 0x6505:0x00 0x10 UNSIGNED16   Warnings
  [0x0025.0] 0x6300:0x01 0x08 UNSIGNED8    CAM state channel 1
  [0x0026.0] 0x6400:0x01 0x08 UNSIGNED8    Area state channel 1

Slave:5
 Name:? M:00ecadc0 I:adec0578
 Output size: 0bits
 Input size: 216bits
 State: 18
 Delay: 1685[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.0.0.0
 Configured address: 1005
 Man: 00ecadc0 ID: adec0578 Rev: 00010000
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1200 L: 512 F:00010022 Type:2
 FMMU0 Ls:00000027 Ll:  27 Lsb:0 Leb:7 Ps:0000 Psb:0 Ty:01 Act:01
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 0e
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0027.0] 0x6004:0x00 0x20 UNSIGNED32   Position value
  [0x002B.0] 0x6030:0x01 0x10 INTEGER16    Speed value channel 1
  [0x002D.0] 0x6503:0x00 0x10 UNSIGNED16   Alarms
  [0x002F.0] 0x6505:0x00 0x10 UNSIGNED16   Warnings
  [0x0031.0] 0x1001:0x00 0x08 UNSIGNED8    Error Register
  [0x0032.0] 0x6008:0x00 0x40 UNSIGNED64   High resolution position value
  [0x003A.0] 0x6030:0x01 0x10 INTEGER16    Speed value channel 1
  [0x003C.0] 0x6503:0x00 0x10 UNSIGNED16   Alarms
  [0x003E.0] 0x6505:0x00 0x10 UNSIGNED16   Warnings
  [0x0040.0] 0x6300:0x01 0x08 UNSIGNED8    CAM state channel 1
  [0x0041.0] 0x6400:0x01 0x08 UNSIGNED8    Area state channel 1
End slaveinfo, close socket
End program

[slaveinfo.docx](https://github.com/OpenEtherCATsociety/SOEM/files/2808975/slaveinfo.docx)

[simpletest-wireshark.pcapng.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2808981/simpletest-wireshark.pcapng.txt)

catfacedck, 2019-01-31:
That you for your analysis. Much appreciated. The EtherCAT slave engine resides in FPGA and was supplied by another software vendor. The encoders and other products using this engine do function with some EtherCAT masters and yet fail with others. Would the XML configuration file be helpful?

ArthurKetels, 2019-01-31:
Yes of course, more info is more understanding.

catfacedck, 2019-01-31:
Attached is the slave XML file.

[EPCMAG-EtherCAT.xml.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2818149/EPCMAG-EtherCAT.xml.txt)

ArthurKetels, 2019-01-31:
Thanks for the data. And yes I understand other master can get these slaves to operational. 

The ESI (.xml) file gives the correct data on how to configure SM2 and SM3. 

But even so, this file has a weird syntax. It is correct but not like most others implement it. Have a look at the ESI files from Beckhoff slaves.

The excess length of index 0x1a00 in the object dictionary (online CoE) for example is exactly replicated in the  ESI file.

But to the point, SM2 and SM3 definition should be:
```<Fmmu>Outputs</Fmmu>
<Fmmu>Inputs</Fmmu>
<Fmmu>MBoxState</Fmmu>
<Sm DefaultSize="512" StartAddress="#x1000" ControlByte="#x26" Enable="1">MBoxOut</Sm>
<Sm DefaultSize="512" StartAddress="#x1200" ControlByte="#x22" Enable="1">MBoxIn</Sm>
<Sm DefaultSize="0" StartAddress="#x1400" ControlByte="#x04" Enable="0">Outputs</Sm>
<Sm DefaultSize="27" StartAddress="#x2000" ControlByte="#x20" Enable="1">Inputs</Sm>
```
SM2 start address 0x1400, flags 0x04, enable 0, type 3 (outputs)
SM3 start address 0x2000, flags 0x20, enable 1, type 4 (inputs)

Also the FMMU definitions are correct in the ESI,
FMMUfunc  0:1 1:2 2:3

If you can get the engineers to update the eeprom SII with this data I am sure SOEM will configure the slave correctly.

nakarlsson, 2019-03-17:
can we close this one?

catfacedck, 2019-03-17:
Yes please close thank you for your assistance.
Your insights helped to solve the issue.
--------------------------------------------

 Received: Sunday, 17 March, 2019, 6:56 PM
 

 
 it on GitHub, or mute
 the thread.

closed 2019-03-17 by nakarlsson