OpenEtherCATsociety/SOEM #250: EL6751 not failsafeing outputs on coms lost

Oysteinzx, 2019-02-04:
Hello,

I run a setup with these modules:

 * Slave:1 Name: EJ1100 
 * Slave:2 Name: EJ1889 
 * Slave:3 Name: EJ2809 
 * Slave:4 Name: EJ2502 
 * Slave:5 Name: EJ3108 
 * Slave:6 Name: EK1100 
 * Slave:7 Name: EL6002
 * Slave:8 Name: EL6751

When cable is removed between master and slave 1, all (output) modules fail safe their outputs, except the EL6751. The EL6751 controls a canbus motor drive, and this continues to run. The heartbeat from the master also indicate that it is OPERATIONAL (on canbus side). LEDs on the module indicate that it is SAFE-OP (on the ethercat side).

With exact same slave setup and a Twincat3 master, the canbus drive stops right away.

As far as I have been able to gather, the watchdog/fail safe mechanism is controlled by the ESC and setup in memory location 0x04yy. I have not been able to find any reference to it in the code, but does SOEM set up/modify this register at any point?

ArthurKetels, 2019-02-04:
Information about setting of the slave ESC watchdog is only available in the ESI xml file. It is not available in the slaves SII eeprom. Therefore SOEM has no way of knowing if and how to configure the ESC watchdog. Add to this the rare use of the ESC watchdog as most new slaves use firmware and check the SM events.

The only way of configuring the slave watchdog with SOEM is to use the preop to safeop hook for this slave and do the configuration manually.

nakarlsson, 2019-02-07:
Isn’t Watchdog info part of SII in the SM section? if the control byte is 0x64?

Oysteinzx, 2019-02-07:
Yes. The actual timings of the WD is set up in 0x0400 regs and the SM has to flag its use in config bit (0x01<<6) as far as i can understand. But setting it does not help. :)

But I am lead to belive that this has nothing to do with how SOEM sets up the SM and other settings, because if I connect the stack to a Twincat3 master, start it and when it is up and running, moving the stack connection to the SOEM master and starting it, the drive will stop if i disconnect the cable. So it has to be a setting that Twincat sets that SOEM does not touch.

Tracing the two systems startups with Wireshark, I have filtered on "FPWR" commands. Both systems seem to start up the node in almost the same way. Twincat zeros out all SM settings at startup and SOEM does not. SOEM reads a lot from EEPROM and Twincat does not. Twincat also sends an AMS packet before and after the SDO setup. And SDOs are set up in a different way then what is documented in the manual. So I suspect that Beckhoff has some dirty tricks under the hood, to discourage EL6751 use outside of Twincat. Beckhoff support has not been very helpful :/.

So I don't belive this to be an actual issue with SOEM. But I would like to keep this post open for a few more days, so I can post a solution, if I stumble upon it. If thats OK?

ArthurKetels, 2019-02-07:
It is fine to keep it open for a while. I know from other projects the EL6751 has some secret modes that are not mentioned in the manual. It allows for example to do raw CAN transmissions. This use is strictly TwinCAT only. 

If I thought CAN bridging would be a good idea then I would have reverse engineered this slave a long time ago. But bridging legacy bus systems on EtherCAT is rather pointless. Much better to select slaves that speak EtherCAT directly or if not available develop you own slave with the capability you need.

nakarlsson, 2019-04-29:
@Oysteinzx , active issue?

nakarlsson, 2019-05-22:
I'll close since not active

closed 2019-05-22 by nakarlsson

kunijiwa03, 2021-03-04:
Hello Oysteinzx, I am new to the use of el6751. I tried to configure the EL6751 with Startup SDOs, which failed with "State=0x12 StatusCode=0x001e : Invalid input configuration". I never found a good solution, and if you can share your configuration way or codes, it will help me a lot. 
Thank you!
Best wishes!

Oysteinzx, 2021-03-04:
Hello,

I have attached the latest file I used back then. As you can see its poorly documented and commented.
I do not know if its functional. I did not go down this route because of the issue described in this thread. 

Hope it can be to some use for you.

[el6751.c.zip](https://github.com/OpenEtherCATsociety/SOEM/files/6081706/el6751.c.zip)

Also a struct i used at some point that has some info:
```c
struct el6751_8000 {
	uint8_t comPara; 		//+ -1
	uint16_t nodeAdr; 		//++ -3
	uint32_t devType; 		//++++ -7
	uint32_t vid; 			//++++ -11
	uint32_t pcode;			//++++ -15
	uint32_t rev;			//++++ -19
	uint32_t serial;		//++++ -23
	uint16_t res1;			//++ -25
	uint16_t res2;			//++ -27
	uint8_t res3[6];		//++++++ -33
	uint16_t flags;			//++ -35         uninon with bitmap
	uint16_t guardTime;		//++ -37
	uint16_t lifeTime;		//++ -39
	uint16_t sdoTimeout;    //++ -41
	uint16_t bootTimeout;	//++ -43
	uint8_t aoeServices;	//+  - 44
	uint8_t faultAction			:1;
	uint8_t faultActionRestart	:1;
	uint8_t faultActionMaster	:1;
	uint8_t faultActionTxPDO	:1;
	uint8_t res4				:4;
							//+ -45
	uint8_t res5;			//+ -46 	must be 0x0A!
	uint8_t res6;			//+ -47
	uint32_t res7;			//++++ -51
	uint32_t res8;			//++++ -55
	uint32_t res9;			//++++ -59
	uint32_t res10;			//++++ -63
	uint32_t res11;			//++++ -67
	uint32_t res12;			//++++ -71
	uint32_t res13;			//++++ -75

};
```

Also a early version where the commit message indicate that the canbus was working. :)
```c
static uint8_t reg_0xF800[] = {
		0x18, 0x00,
		0x7F,
		0x03,
		0x80, 0x00,
		0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00,
		0x00,
		0x1e,
		0x64, 0x00,
		0x00,
		0x00,
		0x00, 0x00,
		0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00,
		0x00, 0x00, 0x00, 0x00
}; //EL67xx CoE Init Cmd 2 (F800:00)

[... 287 lines cut ...]
		g_message("Config CAN NODE %i, reg: %04X", canid, reg);
		ec_SDOwrite(slave, reg, 0x00, TRUE, sizeof(reg_0x8006), &reg_0x8006, EC_TIMEOUTSAFE);

		while(ec_iserror())
			g_message("CAN ID: %i :%s",canid,ec_elist2string());

		//setup slaves RxPDOs
		reg_0x8008[2] = canid;
		reg = 0x8008+(0x10*(canid-1));
		g_message("Config CAN NODE %i, reg: %04X", canid, reg);
		ec_SDOwrite(slave, reg, 0x00, TRUE, sizeof(reg_0x8008), &reg_0x8008, EC_TIMEOUTSAFE);

		while(ec_iserror())
			g_message("CAN ID: %i :%s",canid,ec_elist2string());

	}

	//map PDOs
	ec_SDOwrite(slave, 0x1C12, 0x00, TRUE, sizeof(reg_0x1C12), &reg_0x1C12, EC_TIMEOUTSAFE);
	//map PDOs
	ec_SDOwrite(slave, 0x1C13, 0x00, TRUE, sizeof(reg_0x1C13), &reg_0x1C13, EC_TIMEOUTSAFE);

	return(1);
}
```

kunijiwa03, 2021-03-05:
Thank you so muchu!