OpenEtherCATsociety/SOEM #251: Sensor not entering OP state

dgehri, 2019-02-05:
Hi,

I know that this sounds similar to many other posts here. But I've tried hard to find a solution and could use some help. 

I'm trying to bring a force sensor into OP mode, but of course it fails with "StatusCode=  1e : Invalid input configuration". Unlike similar issues in this forum, the data dumped by _slaveinfo_ looks good to me:

`slaveinfo.exe \Device\NPF_{024D720E-5DFD-4E16-8619-C6076D023768}
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on \Device\NPF_{024D720E-5DFD-4E16-8619-C6076D023768} succeeded.
1 slaves found and configured.
Calculated workcounter 3
Not all slaves reached safe operational state.
Slave 1 State=12 StatusCode=  1e : Invalid input configuration

Slave:1
 Name:ATI EtherCAT F/T Sensor
 Output size: 64bits
 Input size: 224bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000732 ID: 26483052 Rev: 00010111
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   8 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000008 Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 23 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
`

I've also attached the XML file for this device. I'm trying to use the "simple_test" example code, more or less as is, to enter OP mode. As you probably can tell, I'm not an expert in this domain, and any hint to what would be needed to fix this would be appreciated!

-- Daniel

- [ATIEtherCATFT.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2831348/ATIEtherCATFT.txt)
- [ATIEtherCATFT.xml](https://github.com/OpenEtherCATsociety/SOEM/files/2831353/ATIEtherCATFT.zip)

dgehri, 2019-02-05:
I'm also adding the Wireshark capture during `simple_test.exe \Device\NPF_{024D720E-5DFD-4E16-8619-C6076D023768}`.

[simple_test_wireshark.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2831448/simple_test_wireshark.zip)

dgehri, 2019-02-05:
Update: after working through issue #224, I know manage to bring the slave to state OP (see below). However, the workcounter in `RTthread` is always _2_, and hence below `expectedWKC` of _3_.

```
ATI F/T Sensor Reader
Starting simple test
ec_init on \Device\NPF_{024D720E-5DFD-4E16-8619-C6076D023768} succeeded.
1 slaves found and configured.
Found ATI EtherCAT F/T Sensor at position 1
Slaves mapped, state to SAFE_OP.
segments : 1 : 36 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
MESSAGE : slave 1 reconfigured
MESSAGE : slave 1 reconfigured
MESSAGE : slave 1 reconfigured
MESSAGE : slave 1 reconfigured
MESSAGE : slave 1 reconfigured
MESSAGE : slave 1 reconfigured
MESSAGE : slave 1 reconfigured
MESSAGE : slave 1 reconfigured
ERROR : slave 1 lost
MESSAGE : slave 1 recovered
MESSAGE : slave 1 reconfigured
```

ArthurKetels, 2019-02-07:
Thanks for all the info. The strange thing is that there seems to be nothing wrong with the configuration. Both the eeprom (SII), xml (ESI) and wireshark show the same data.

Still, going from pre-op to safe-op throws a slave error 0x001e invalid input configuration. 

My guess is there must be a timing issue in the slave. Could you test with twincat3 (can try out for free for 30 days) if you can get the slave to operational? And if yes, show me the wireshark trace.

dgehri, 2019-02-07:
Thank you for getting back to me! 

I'm going to try TwinCAT 3, yes. But in the meantime, I have the following info: 

I have a ECAT master from Elmo Motion Control ("GMAS"), and I tried to use it to set up the ECAT configuration with the sensor. The strange thing is that, if I let it detect the sensor, it too will fail to bring it to OP state. 

However, when, on the master, I disable the use of the DC (see image below, Enable is unchecked), it can bring it into OPERATIONAL state and I can see process data coming in. Unfortunately, I don't think I can log this with Wireshark.

I therefore suspect a clock issue, but don't know how to configure the equivalent timing on SOEM. I've attached the XML [configuration file](https://github.com/OpenEtherCATsociety/SOEM/files/2840026/Config.zip) used by ELMO for this slave, and in the <InitCmds> section, one can see the sequence of configuration commands. Again, I assume that SOEM actually performs most of them automatically based on the slave EEPROM data, but maybe I would have to perform some of them manually, but again, don't now why.

[image]

Best,
Daniel

ArthurKetels, 2019-02-07:
Hello Daniel,

I guess you have given the solution to your problem already. The use of DC is most likely triggering the slave to behave wrongly.

To test this, comment the following line in simple_test:
```ec_configdc();```
This function enables the distributed clock preparation in SOEM. Without it DC is not used.

I can see in the ESI that this slave has two modes, SM synchronous and SYNC0 synchronous. Perhaps the firmware in the slave has the wrong checks internally to decide in which mode it should go.

dgehri, 2019-02-07:
Hello Arthur. If only... I tried that but the problem is the same (see attached Wireshark log, linux/simple_test.c with ec_configdc() commented out).

[simple_test_disabled_ec_configdc.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2840252/simple_test_disabled_ec_configdc.zip)

dgehri, 2019-02-07:
Update. I tried to apply the PreOp -> SafeOp init as described in the sensor's XML configuration file (see above), and hence added the following in the P->S hook:

```
    printf("Configure PreOp -> SafeOp (state = %d)\n", ec_slave[slave].state);
    int wkc = 1;
    auto configadr = ec_slave[slave].configadr;

    // set sm 2 (output)
    smt.StartAddr = htoes(0x1800);
    smt.SMlength = htoes(0x0008);
    smt.SMflags = htoel(0x00010064);
    ec_FPWR(configadr, ECT_REG_SM2, sizeof(ec_smt), &smt, EC_TIMEOUTRET);

    // set sm 3 (input)
    smt.StartAddr = htoes(0x1c00);
    smt.SMlength = htoes(0x0020);
    smt.SMflags = htoel(0x00010000);
    ec_FPWR(configadr, ECT_REG_SM3, sizeof(ec_smt), &smt, EC_TIMEOUTRET);
```

It now goes into OP mode. I'll keep debugging, but just a quick question: I don't know anything about those flags -- are there any _defines_ I should use instead, on if so, which?

ArthurKetels, 2019-02-07:
Forgive my stupidity, but I just missed the very obvious thing in the beginning. I guess my cold interferes with my brain too much.

The SM configuration SOEM tries is :
``` 
 SM2 A:1800 L:   8 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
```
The length of input SM3 is written as 28. But in reality it should be 32. There are 8 objects of 32bits. The reason is that SOEM asks the PDO mapping from the slave with CompleteAccess and the slave answers with only 7 objects. So SOEM tries to do as the slave instructs. The error is in the slave returning too little objects from the SDO CA request. See line 270 of simple_test_wireshark.

There are two solutions:
1) Ask the manufacturer to fix the bug and update the firmware.
2) Disable Complete Access for this slave.

After ec_config_init() but before ex_config_map() you can do
ec_slave[slc].CoEdetails &= ~ECT_COEDET_SDOCA;
to clear the Complete Access bit in the CoE capabilities list.

dgehri, 2019-02-07:
Thank you very much! This works like a charm! 

One last question: the process data size calculated based on ec_slave[0].Obytes / ec_slave[0].Ibytes results in 8 output and 8 input bytes. But the sensor is supposed to have **28** input bytes (6 x DINT and 2 x UDINT) and 8 output bytes. Why is this?

dgehri, 2019-02-07:
One last question: why is simple_test incrementing  `++IOmap[0]` in the RT callback?

ArthurKetels, 2019-02-07:
That is just the "simple" in simple_test. Just to show how one could manipulate the IOmap directly to transfer process data to the slaves. A test setup with a simple output slave will blink some leds.

closed 2019-02-07 by dgehri