OpenEtherCATsociety/SOEM #253: Incorrect expected work counter

puneetsharma2793, 2019-02-05:
Hi, 
I started with red_test example, SOEM on linux and I have one Beckhoff slave EL9800, and got unexpected output regarding the work-count while running the application. If I am not wrong, workcount should be equivalent to no of active slaves on the chain. With just one beckhoff slave I am getting work-count as 3. Can anybody please clarify give some clues on this?
Many thanks.
Here is the output of the red_test prog
```
SOEM (Simple Open EtherCAT Master)
Redundancy test
Starting Redundant test
ec_init on enp0s31f6 succeeded.
1 slaves found and configured.
Slave:1 Name:FB1XXX Output size: 16bits Input size: 16bits State: 4 delay:0.1
         Out:0061a260,   2 In:0061a262,   2
Calculated workcounter **3**
Request operational state for all slaves
Operational state reached for all slaves.
Request safe operational state for all slaves
End redundant test, close socket
End program

```

nakarlsson, 2019-02-06:
For an input & output slave working counter 3 is expected.
This is basic knowledge, to retrieve more information on EtherCAT, beside trying SOEM you can go to www.ethercat.org.

puneetsharma2793, 2019-02-06:
Hi Nakarlsson, isn't it like work count should be increased by +1 each time a slave process the packet? I am unable to correlate between the theoretical and practical example.
An in case I have 2 input& output slaves, the expected Working counter should go to 6 ?

nakarlsson, 2019-02-06:
https://download.beckhoff.com/download/document/io/ethercat-development-products/ethercat_esc_datasheet_sec1_technology_2i2.pdf

Check: 2.4 Working Counter

closed 2019-02-06 by puneetsharma2793