OpenEtherCATsociety/SOEM #254: Ethercat frame reception latency too long on linux preemptRT

grandphil, 2019-02-07:
Hello,

I am testing the SOEM master on an intel based target on linux and I am suprised by the long dealy between frame emission and reception : ~240µs for a cyclic task of 1000µs. The test is done with a single ek1818 connected to an intel network card.

This timing is measured with tshark
......
10755	5.376589	Private_01:01:01	Broadcast	ECAT	30	'LRW': Len: 2, Adp 0x0, Ado 0x0, Wc 0 
10756	5.376899	03:01:01:01:01:01	Broadcast	ECAT	60	'LRW': Len: 2, Adp 0x0, Ado 0x0, Wc 3 
10757	5.377666	Private_01:01:01	Broadcast	ECAT	30	'LRW': Len: 2, Adp 0x0, Ado 0x0, Wc 0 
10758	5.377857	03:01:01:01:01:01	Broadcast	ECAT	60	'LRW': Len: 2, Adp 0x0, Ado 0x0, Wc 3 
10759	5.378669	Private_01:01:01	Broadcast	ECAT	30	'LRW': Len: 2, Adp 0x0, Ado 0x0, Wc 0 
10760	5.378910	03:01:01:01:01:01	Broadcast	ECAT	60	'LRW': Len: 2, Adp 0x0, Ado 0x0, Wc 3 
......

here are some informations about my target:
$ uname -a
Linux pp-target 4.9.0-4-rt-amd64 #1 SMP PREEMPT RT Debian 4.9.51-1 (2017-09-28) x86_64 GNU/Linux
$ lspci
...
00:19.0 Ethernet controller: Intel Corporation 82579LM Gigabit Network Connection (rev 04)
...

According to beckhoff documents the round trip of the ethercat frame on such a setup should be less than 10µs.

Is it a common performance ?

Best regards,

ArthurKetels, 2019-02-07:
Linux NIC drivers are not build for low latency but for maximum throughput. But with the help of ethtool some of the settings can be tweaked to improve latency. Most drivers have packet coalescing enabled by default. 

Read drvcomment.txt in the SOEM repo. 

When properly used it should reduce latency to around 30us.

grandphil, 2019-02-12:
This answer solved the latency problem. 
Thank you very much!

closed 2019-02-12 by grandphil