OpenEtherCATsociety/SOEM #255: Problems with DC channel of a Beckhoff EL2004.

claudioscordino, 2019-02-07:
Hi.
I'm using SOEM on Linux for setting the output voltage of a Beckhoff EL2004 device [1].

./slaveinfo -sdo reports

```
Slave:4
 Name:EL2004
 Output size: 4bits
 Input size: 0bits
 State: 4
 Delay: 435[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1004
 Man: 00000002 ID: 07d43052 Rev: 00110000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:3 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 100[mA]
 only LRD/LWR:0
```

while ./slaveinfo -maps reports

```
Slave:4
 Name:EL2004
 Output size: 4bits
 Input size: 0bits
 State: 4
 Delay: 430[ns]
 Has DC: 1
 DCParentport:1
 Activeports:1.1.0.0
 Configured address: 1004
 Man: 00000002 ID: 07d43052 Rev: 00110000
 SM0 A:0f00 L:   1 F:00090044 Type:3
 FMMU0 Ls:00000000 Ll:   1 Lsb:0 Leb:3 Ps:0f00 Psb:0 Ty:02 Act:01
 FMMUfunc 0:1 1:0 2:0 3:0
 MBX length wr: 0 rd: 0 MBX protocols : 00
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 100[mA]
 only LRD/LWR:0
PDO mapping according to SII :
  SM0 RXPDO 0x1600 Channel 1
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1601 Channel 2
     addr b   index: sub bitl data_type    name
  [0x0000.1] 0x7010:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1602 Channel 3
     addr b   index: sub bitl data_type    name
  [0x0000.2] 0x7020:0x01 0x01 BOOLEAN      Output
  SM0 RXPDO 0x1603 Channel 4
     addr b   index: sub bitl data_type    name
  [0x0000.3] 0x7030:0x01 0x01 BOOLEAN      Output
```

From different tools, I know that the channel is mapped onto PDO at 0x1600 and that the PDO Content is mapped at index 0x7000:01.

I was therefore expecting the following code to enable the channel:

```
uint8_t buffer8 = 0x1;
ec_slave[4].state = EC_STATE_OPERATIONAL;
ec_writestate(4);
ec_send_processdata();
ec_RxPDO(4, 0x7000, sizeof(buffer8), (void*) &buffer8);
```

but it generates a segmentation fault.
I've tried substituting the ec_RxPDO() call with 

```
ec_SDOwrite(4, 0x7000, 0x01, FALSE, sizeof(buffer8), (void*) &buffer8, EC_TIMEOUTRXM*4);
```

but it didn't work either.

Which is the right call for setting such PDO ? ex_RxPDO or ec_SDOwrite ?
Is a buffer of uint8_t correct ?

Many thanks.

[1] https://www.beckhoff.com/english.asp?ethercat/el2004.htm

nakarlsson, 2019-02-07:
simple_test is a better starting point to actually run process data. Where the ec_slave structure have pointer to the IOmap per slave.

ArthurKetels, 2019-02-07:
Yes, ec_RxPDO is a legacy EtherCAT function that has actually never been used. 

An asynchronous write via mailbox to slave PDO works, but only if the slaves supports CoE. An EL2004 is a simple slave without micro-controller and has no mailbox let alone CoE. Using slow (very very slow) mailbox transfers defeats the whole purpose of EtherCAT. To be very efficient and fast.

As Andreas mentioned, look at simple_test or red_test on how to set-up PDO transfers very efficiently. 

Another good read would be:
https://www.ethercat.org/download/documents/EtherCAT_Device_Protocol_Poster.pdf

claudioscordino, 2019-02-08:
> An asynchronous write via mailbox to slave PDO works, but only if the slaves supports CoE. An EL2004 is a simple slave without micro-controller and has no mailbox let alone CoE. Using slow (very very slow) mailbox transfers defeats the whole purpose of EtherCAT. To be very efficient and fast.

I see. Just to know, which would have been the right function for such a (slow) mailbox transfer ?

> As Andreas mentioned, look at simple_test or red_test on how to set-up PDO transfers very efficiently.

I guess you mean using 
```
data_ptr = ec_slave[4].outputs;
*data_ptr++ = (0x01 >> 0) & 0xFF;
*data_ptr++ = (0x00 >> 8) & 0xFF;
```
Indeed, this was my first attempt, as it is the suggested approach in https://openethercatsociety.github.io/doc/soem/tutorial_8txt.html.

However, I see the led of the slave blinking for just a while, then the led going back to off (and the output still to 0V).
Am I missing some further command for telling the slave to keep the set value ? 
Or is the slave maybe refusing the command for some reason ? (somewhere I read that CoE needs some kind of "set valid" message for telling the slave to accept commands).

> Another good read would be:
> https://www.ethercat.org/download/documents/EtherCAT_Device_Protocol_Poster.pdf

Thank you. This seems very instructive!

nakarlsson, 2019-02-08:
You should study the EtherCAT state machine. 

In summary, the outputs will only be enabled(possible to turn ON) in state Operational. To stay in Operational, you must cyclically send valid process data to the slave, otherwise the slave watchdog will trip and the slave go to SafeOp + Error disabling the outputs.

claudioscordino, 2019-02-08:
It seems that I missed a piece of quite fundamental information. Now it works, of course.
I'll look for a good document explaining the details of the state machine.
Thank you.

closed 2019-02-08 by claudioscordino