OpenEtherCATsociety/SOEM #257: Fsoe implementation

samAvr, 2019-02-26:
hello, 

i am totally new to the industrial communication protocols and i tried to get the data from ethercat slave and it works perfect.
my problem is that i am trying to get the data from a safety ethercat slave and to do that i need to do the crc calculation and build a new frame to send to the slave.
i searched online but i could not figure out how to do this calculation.
so is there any way to know to get the data from a safety slave?

thnx

ArthurKetels, 2019-02-28:
FSoE protocol is not supported by SOEM. It would require a separate master (FSoE master) on top of EtherCAT. It is not a simple protocol and unless someone knowledgeable is willing to implement it I do not think you will succeed.

For documentation write an email to the ETG or obtain IEC61784-3 "IEC specification of FSoE protocol".

mheden, 2019-03-04:
I think that there is no real safety requirements for FSoE on the master (not FSoE master, but EtherCAT master) side (apart from driving data, setting up slave-to-slave). There was a document on ethercat.org that described how to drive FSoE traffic with a standard master but I can't seem to find it now.

nakarlsson, 2019-03-04:
As Arthur wrote, he needs a FSoE Master to control the FSoE Slave, eg run the FSoE state machine ontop of the EtherCAT statemachine in OP. FSoE Master example, EL6910 from Beckhoff. It can be setup with TwinCAT, when the FSoE connections are configured it is possible to run the FSoE data via SOEM with slave to slave communication.

nakarlsson, 2019-03-17:
If you need help implementing a FSoE device you can contact sales[a]rt-labs.com, I think this issue can be closed.

samAvr, 2019-03-22:
sry for being late, i have a lot of exams at the university
what i need is not a FSOE Master cuz the data exchange is like the normal Ethercat, what i need is how to calculate the CRC for the FSOE devices.
when i calculate the CRC i can send it to the device and then read his answer, but how to do this calculation is my problem

nakarlsson, 2019-03-22:
Ok, then you need to contact ETG and ask for the FSoE specification. 
Please close this issue

closed 2019-03-31 by nakarlsson