OpenEtherCATsociety/SOEM #258: Accessing PDOs

claudioscordino, 2019-03-05:
Since `ec_RxPDO()` is deprecated, does SOEM provide any mean for accessing CoE PDOs based on _index:subindex_ ? If I'm not wrong, in fact, `ec_slave` just writes data in the IOmap regardless of PDO mapping.

ArthurKetels, 2019-03-05:
PDO (Process Data Object) have their representation in CoE. CoE is an asynchronous data transfer via mailbox. It is inefficient and slow but very flexible. It is possible to read and write PDO via mailbox. You then use the SDO (Service Data Object) with the corresponding index and subindex. The functions to use are ecx_SDOread() and ecx_SDOwrite(). 

The applications task is to define the structures to control the slave PDO's correctly. The application can use the mapping information in the slave or a pre defined set.

claudioscordino, 2019-03-07:
Thank you. Can you elaborate what you mean by _"It is possible to read and write PDO via mailbox"_ ?
Specifically, which functions are you referring to ?
Thanks again.

ArthurKetels, 2019-03-07:
An exmple from an earlier post here. Look at the output of "slaveinfo eth0 -map"

```
Slave:1
 Name:ATI EtherCAT F/T Sensor
 Output size: 64bits
 Input size: 224bits
 State: 18
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000732 ID: 26483052 Rev: 00010111
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   8 F:00010064 Type:3
 SM3 A:1c00 L:  28 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   8 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000008 Ll:  28 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 0c
 CoE details: 23 FoE details: 01 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7010:0x01 0x20 UNSIGNED32   Control 1
  [0x0004.0] 0x7010:0x02 0x20 UNSIGNED32   Control 2
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0008.0] 0x6000:0x01 0x20 INTEGER32    Fx/Gage0
  [0x000C.0] 0x6000:0x02 0x20 INTEGER32    Fy/Gage1
  [0x0010.0] 0x6000:0x03 0x20 INTEGER32    Fz/Gage2
  [0x0014.0] 0x6000:0x04 0x20 INTEGER32    Tx/Gage3
  [0x0018.0] 0x6000:0x05 0x20 INTEGER32    Ty/Gage3
  [0x001C.0] 0x6000:0x06 0x20 INTEGER32    Tz/Gage3
  [0x0020.0] 0x6010:0x00 0x20 UNSIGNED32   SubIndex 000
  [0x0024.0] 0x6020:0x00 0x20 UNSIGNED32   SubIndex 000
End slaveinfo, close socket
End program
```
SM2 shows the mapped output PDO's and SM3 the input PDO's. Via the ec_slave[1].Inputs and ec_slave[1].Outputs pointer you can access these structures directly. With mailbox SDO transfers those items can be accessed too, only less efficient.

For example the read Tx/Gage3 the application can do:
```
int32_t rdvar;
int rdl = 4;
int slave = 1;
int wkc;
wkc = ec_SDOread(slave, 0x6000 , 0x04, FALSE, &rdl, &rdvar, EC_TIMEOUTRXM);
```
wkc is the returned workcounter, 1 = succeed, 0 = refused, -1 = no asnwer.
rdvar is the returned value, and rdl holds the actual returned length in bytes (handy for strings).

To write Control2:
```
uint32_t wrvar = 0x1234abcd;
int wrl = 4;
int slave = 1;
int wkc;
wkc = ec_SDOwrite(slave, 0x7010 , 0x02, FALSE, wrl, &wrvar, EC_TIMEOUTRXM);
```
Keep in mind that the ethercat state machine will not update Control 2 unless the slave state is Operational.

Some slaves mark the PDO objects in the CoE dictionary as PDO only (rare), so then you can not use SDO transfers to read/write data in these objects. Only IOmap is allowed.

WangKaimeng, 2019-03-13:
> allowed

What's the meaning of IOmap? Does it contain the address of PDO just like ec_slave[1].Outputs? By the way, how to determine the offset address of every mapped PDO when I use ec_slave[1].Outputs pointer. 
Could you please give an example that reads Tx/Gage3 like above by using ec_slave[1].Inputs or IOmap?

ArthurKetels, 2019-03-13:
@WangKaimeng : Please read the tutorial.txt

WangKaimeng, 2019-03-14:
> tutorial.txt
Thank you so much!

claudioscordino, 2019-03-22:
Hi Arthur.
Thank you so much for providing the code snippet.
I'm dealing with a Beckhoff EL2004  and `ec_SDOwrite()` returns `0`.
It seems that `ecx_readeeprom2()` return `0` for the specific slave.

Here is my call to `ec_SDOwrite`:
```
uint8_t rdvar = 1;
int rdl = sizeof(rdvar);
`ec_SDOwrite(4, 0x7000 , 0x01, 0, rdl, &rdvar, 700000);`
```

Here is the output of `slaveinfo eth0 -map`:
```
Slave:4
Name:EL2004
Output size: 4bits
Input size: 0bits
State: 4
Delay: 430[ns]
Has DC: 1
DCParentport:1
Activeports:1.1.0.0
Configured address: 1004
Man: 00000002 ID: 07d43052 Rev: 00110000
SM0 A:0f00 L: 1 F:00090044 Type:3
FMMU0 Ls:00000000 Ll: 1 Lsb:0 Leb:3 Ps:0f00 Psb:0 Ty:02 Act:01
FMMUfunc 0:1 1:0 2:0 3:0
MBX length wr: 0 rd: 0 MBX protocols : 00
CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
Ebus current: 100[mA]
only LRD/LWR:0
PDO mapping according to SII :
SM0 RXPDO 0x1600 Channel 1
addr b index: sub bitl data_type name
[0x0000.0] 0x7000:0x01 0x01 BOOLEAN Output
SM0 RXPDO 0x1601 Channel 2
addr b index: sub bitl data_type name
[0x0000.1] 0x7010:0x01 0x01 BOOLEAN Output
SM0 RXPDO 0x1602 Channel 3
addr b index: sub bitl data_type name
[0x0000.2] 0x7020:0x01 0x01 BOOLEAN Output
SM0 RXPDO 0x1603 Channel 4
addr b index: sub bitl data_type name
[0x0000.3] 0x7030:0x01 0x01 BOOLEAN Output
```

Some possible reasons for this behaviour could be: 
1. Mailboxes need some kind of initialization that I'm missing (isn't any ?)
2. These specific PDOs cannot be accessed as SDOs (how can I check if it is a "PDO only" ?)
3. I'm mistaking the variable for dealing with BOOLEAN PDOs

Any suggestion ? Many thanks.

ArthurKetels, 2019-03-22:
The EL2004 is a dumb slave without any micro-controller or slave firmware. It only has an an ET1200 ESC, the digital outputs are directly connected to its pins. 

Therefore it has no CoE, not even mailboxes. You can see that from the slaveinfo output. There is one SM with type 3 (outputs), there are no SM with type 1 and 2 (mailbox in and out).

To enable PDO mapping the slave uses the alternative mapping source, the eeprom SII. So it looks like there is an object 7000:01, but it is only a fixed structure in the eeprom.

Readeeprom2 is one half of the eeprom read function readeerprom1. But I would not advise using those. SOEM has high level eeprom access functions that are much easier to use. There is also an eeprom dump function.

claudioscordino, 2019-03-25:
Thank you for the explanations.
In the end, which function do you suggest to use if we want to access such dump output through the 7000:01 index?
Thanks again.

ArthurKetels, 2019-03-25:
I am not sure if I understand your question. What is it that you want to do? 

If you want to use the master for process control / plc then just use the IOmap structure and attach the appropriate structures to it to directly access the slave in and outputs.

If you want to analyse the internal mapping of the slaves then look at the slaveinfo.c example.

There are multiple ways a slave can communicate with the master. If you want to be able to handle all slaves you have to support all of those methods. Most notably, SII, CoE, SoE and EoE. And for some Beckhoff slaves even AoE (not supporte yet by SOEM). So only focussing on this particular EL2004 is not the right way to go I guess.

closed 2019-03-27 by claudioscordino

Edphoenixmax, 2023-06-29:
> > tutorial.txt
> 
> Thank you so much
你好，请问你看懂了这部分内容吗，能不能给我解释一下呢