OpenEtherCATsociety/SOEM #259: SOEM benchmarking

claudioscordino, 2019-03-05:
Can you suggest me the simplest way for comparing SOEM performance on different OSs ?
Do you have any specific benchmarking suite/test ?
Is there any interesting metric other than latency ?
Many thanks.

ArthurKetels, 2019-03-05:
How do you want to define performance? :-)
Low latency is important, but jitter too. For some applications it is OK to have 10's of usecs of latency, but can not cope well with even a few usecs of jitter. 
There are three distinct services of the OS that are important:
a) the network performance to transmit and receive packets with lowest possible delay and jitter.
b) the OS clock accuracy, to be monotonous, and as jitter free as possible (drift is less important).
c) the OS task scheduler that needs to be on time and jitter free.

EtherCAT can help observing a) with great accuracy, because every slave has it's own clock and timestamps each incoming packet. Observing the slave clock in relation to the OS clock can give results for b).
And c) can be measured completely independent of SOEM.

If you want I can help you setting up some test programs.

claudioscordino, 2019-03-07:
Thank you.
Since b) and c) can be measured independently from EtherCAT (e.g. taking some numbers from the TSC), I just need to measure a).
If you can provide me some hints (or some code, if you have any) it would be greatly appreciated.
Thanks again.

ArthurKetels, 2019-03-07:
Watch out with comparing the OS clock with TSC. TSC is not guaranteed to be a good reference. Specially with modern CPUs. I found suprising results by comparing TSC with the EtherCAT reference slave clock. 

I have build slaves with Stratum 2 disciplined oscillators. COTS mainboard hardware does weird things if you compare to those standards. But it all depends on what you think is an "accurate" reference.

A quick hint to how you can monitor packet timestamps is to be found in SOEM/ethercatdc.c ecx_configdc().

```
   ecx_BWR(context->port, 0, ECT_REG_DCTIME0, sizeof(ht), &ht, EC_TIMEOUTRET);  /* latch DCrecvTimeA of all slaves */
   slaveh = context->slavelist[i].configadr;
   (void)ecx_FPRD(context->port, slaveh, ECT_REG_DCTIME0, sizeof(ht), &ht, EC_TIMEOUTRET);
   context->slavelist[i].DCrtA = etohl(ht);
   /* 64bit latched DCrecvTimeA of slave */
```
DCrtA gives you the 64 bit timestamp of when the BRW datagram has been received by slave i. Time is in local slave clock time in ns.

superlukeme, 2019-03-20:
HI, i'm trying to perform a measurement using the value of DCrtA taken after the code snippet in the previous post. The type of DCrtA is int32 (so not a 64 bit integer). Furthermore, i expected to find a continuous growing value. Instead in a few iteration i found a value smaller than the previous. Am i missing something?
Many thanks in advance.

nakarlsson, 2019-04-29:
Consult the ESC manual how to use DC registers

closed 2019-04-29 by nakarlsson