OpenEtherCATsociety/SOEM #261: Does the SOEM master really work in the following picture?

WangKaimeng, 2019-03-12:
[image]
[image]
When I ran the simple_test, the above situation was met. I didn't know whether the SOEM master really worked. Besides, as I understand, the master should run in a loop, however, the simple_test might finish after few minutes.

ArthurKetels, 2019-03-12:
No, this is not a properly working slave. Try slaveinfo -map and post the result. Please also attach a wireshark trace of the communication.

One slave responded, but there is no PDO mapping found.

WangKaimeng, 2019-03-13:
> No, this is not a properly working slave. Try slaveinfo -map and post the result. Please also attach a wireshark trace of the communication.
> One slave responded, but there is no PDO mapping found.
Dear ArthurKetels,

I realized I didn't add the PDO mapping code into the simple_test, when I added it, the master might be working!
[image]
By the way, I wondered how to change the simple_test to a real master (receive and transmit the process data in a infinite loop).

Thank you !

ArthurKetels, 2019-03-13:
Read the tutorial (always a good thing to start with). You can also try the red_test example.

miscz, 2019-03-15:
Can you maybe provide a link or an instruction on how to add the PDO mapping to the program from a given ESI file? I am getting a similar output like @WangKaimeng . I read already the tutorial and a lot of issues where similar problems (e.g.  [hefloryd/SOEM@22d8f5d](https://github.com/hefloryd/SOEM/commit/22d8f5d)) were described, but unfortunately I did not make it run.

Some information about the setup:
I have a "item24 C1 Controller" with following ESI file [item_EtherCAT_C-Series_V1_3b.xml.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2971131/item_EtherCAT_C-Series_V1_3b.xml.txt). A connection over TwinCAT works fine. Here is a small collection of all the data I gathered using SOEM:

[wireshark_snif.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2971273/wireshark_snif.txt)

```shell
./eepromtool enp0s31f6 1 -i

SOEM (Simple Open EtherCAT Master)
EEPROM tool
ec_init on enp0s31f6 succeeded.
1 slaves found.
Slave 1 data
 PDI Control      : 0008
 PDI Config       : 0600
 Config Alias     : 0000
 Checksum         : 00F5
   calculated     : 008D
 Vendor ID        : 000006A1
 Product Code     : 264860D3
 Revision Number  : 00020000
 Serial Number    : 00000000
 Mailbox Protocol : 0004
 Size             : 0007 = 1024 bytes
 Version          : 0001
End, close socket
End program
```

```shell
./slaveinfo enp0s31f6 -map

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp0s31f6 succeeded.
1 slaves found and configured.
Calculated workcounter 0

Slave:1
 Name:? M:000006a1 I:264860d3
 Output size: 0bits
 Input size: 0bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 000006a1 ID: 264860d3 Rev: 00020000
 SM0 A:1800 L:  16 F:00010026 Type:1
 SM1 A:1900 L:  16 F:00010022 Type:2
 FMMUfunc 0:0 1:0 2:0 3:0
 MBX length wr: 16 rd: 16 MBX protocols : 04
 CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket
End program
```

```shell
./simple_test enp0s31f6

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp0s31f6 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Operational state reached for all slaves.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
```

eepromtool ran without any error messages. Running slaveinfo and simple_test the controller gave an error message saying "invalid R-PDO and T-PDO length". Chasing down the error in the simpletest code, it seems that the error occurs in `ec_config_map(&IOmap);` with `char IOmap[4096];`. I tried already other sizes from 2^0 up to 2^13 with no success.

Could anybody please help me with some easy to understand instructions? I am trying to fix this problem since a couple days now.

And yes, I read the controller manual.

WangKaimeng, 2019-03-15:
> Can you maybe provide a link or an instruction on how to add the PDO mapping to the program from a given ESI file? I am getting a similar output like @WangKaimeng . I read already the tutorial and a lot of issues where similar problems (e.g.  hefloryd/SOEM@22d8f5d) were described, but unfortunately I did not make it run.
> Some information about the setup:
> I have a "item24 C1 Controller" with following ESI file item_EtherCAT_C-Series_V1_3b.xml.txt. A connection over TwinCAT works fine. Here is a small collection of all the data I gathered using SOEM:
> wireshark_snif.txt
> ./eepromtool enp0s31f6 1 -i
> 
> SOEM (Simple Open EtherCAT Master)
> EEPROM tool
> ec_init on enp0s31f6 succeeded.
> 1 slaves found.
> Slave 1 data
>  PDI Control      : 0008
>  PDI Config       : 0600
>  Config Alias     : 0000
>  Checksum         : 00F5
>    calculated     : 008D
>  Vendor ID        : 000006A1
>  Product Code     : 264860D3
>  Revision Number  : 00020000
>  Serial Number    : 00000000
>  Mailbox Protocol : 0004
>  Size             : 0007 = 1024 bytes
>  Version          : 0001
> End, close socket
> End program
> ./slaveinfo enp0s31f6 -map
> 
> SOEM (Simple Open EtherCAT Master)
> Slaveinfo
> Starting slaveinfo
> ec_init on enp0s31f6 succeeded.
> 1 slaves found and configured.
> Calculated workcounter 0
> 
> Slave:1
>  Name:? M:000006a1 I:264860d3
>  Output size: 0bits
>  Input size: 0bits
>  State: 4
>  Delay: 0[ns]
>  Has DC: 1
>  DCParentport:0
>  Activeports:1.0.0.0
>  Configured address: 1001
>  Man: 000006a1 ID: 264860d3 Rev: 00020000
>  SM0 A:1800 L:  16 F:00010026 Type:1
>  SM1 A:1900 L:  16 F:00010022 Type:2
>  FMMUfunc 0:0 1:0 2:0 3:0
>  MBX length wr: 16 rd: 16 MBX protocols : 04
>  CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: 0[mA]
>  only LRD/LWR:0
> PDO mapping according to CoE :
>   SM2 outputs
>      addr b   index: sub bitl data_type    name
>   SM3 inputs
>      addr b   index: sub bitl data_type    name
> End slaveinfo, close socket
> End program
> ./simple_test enp0s31f6
> 
> SOEM (Simple Open EtherCAT Master)
> Simple test
> Starting simple test
> ec_init on enp0s31f6 succeeded.
> 1 slaves found and configured.
> Slaves mapped, state to SAFE_OP.
> segments : 1 : 0 0 0 0
> Request operational state for all slaves
> Calculated workcounter 0
> Operational state reached for all slaves.
> OK : all slaves resumed OPERATIONAL.
> OK : all slaves resumed OPERATIONAL.
> OK : all slaves resumed OPERATIONAL.
> eepromtool ran without any error messages. Running slaveinfo and simple_test the controller gave an error message saying "invalid R-PDO and T-PDO length". Chasing down the error in the simpletest code, it seems that the error occurs in ec_config_map(&IOmap); with char IOmap[4096];. I tried already other sizes from 2^0 up to 2^13 with no success.
> Could anybody please help me with some easy to understand instructions? I am trying to fix this problem since a couple days now.
> And yes, I read the controller manual.

You should map the PDO firstly, as shown in the following code
int COP_XELsetup(uint16 slave)
{
    int retval;
    uint8 u8val;
    uint16 u16val;
	uint32 u32val;
    retval = 0;

    u8val = 0; 
	
    retval += ec_SDOwrite(slave, 0x1c12, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);
	
    retval += ec_SDOwrite(slave, 0x1600, 0x00, FALSE, sizeof(u8val), &u8val, EC_TIMEOUTRXM);

	u32val = 0x60600008;/* 0x6060:1/8bits operation mode  */
	retval += ec_SDOwrite(slave, 0x1600, 0x01, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	//u32val = 0x60600008;/* 0x6060:1/8bits operation mode  */
	//retval += ec_SDOwrite(slave, 0x1600, 0x02, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u32val = 0x60400010;/* 0x6040:0/16bits, control word */
	retval += ec_SDOwrite(slave, 0x1600, 0x03, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u32val = 0x607A0020;/* 0x607A:1/32bits target Position  */
	retval += ec_SDOwrite(slave, 0x1600, 0x04, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u32val = 0x60B10020;/* 0x60B1:1/32bits Velocity offset  */
	retval += ec_SDOwrite(slave, 0x1600, 0x05, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
	u32val = 0x60B20010;/* 0x60B2:1/16bits Torque offset  */
	retval += ec_SDOwrite(slave, 0x1600, 0x06, FALSE, sizeof(u32val), &u32val, EC_TIMEOUTRXM);
}
	After you map the PDO, the slave_info maybe following：
[image]
 
Then, you program will run.

miscz, 2019-03-15:
Thank you for your quick answer! Can you please post your ESI file and your simple_test program, where I can see how you solved this? Where did you put the COP_XELsetup(uint16 slave) function?

miscz, 2019-03-15:
Should I use the ec_SDOwrite() function when writing PDOs? There is no ec_PDOwrite() function right?

ArthurKetels, 2019-03-15:
@Micsz : When posting a wireshark capture please use the native format .pcap and not a text extract .txt. The text file misses a lot of data and is next to useless for EtherCAT debugging.

I can already tell that the slave in question does not follow the ETG slave rules and will not pass the conformance test.  The eeprom SII data is not complete. So first order would be to ask the vendor to update the slave firmware.

WangKaimeng, 2019-03-15:
SOEM is not similar for Twincat, it can't generate ENI file with ESI file. So you should map the PDO manually.  The path of this program: \SOEM-master\test\win32\simple_test. 

You can only config the PDO with SDO in the setup step.

miscz, 2019-03-15:
@ArthurKetels thank you for the hint: [wireshark.pcap.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2971421/wireshark.pcap.txt). I will try to get in touch with with the vendor.

@WangKaimeng Alright. I meant, where did you call the COP_XELsetup(uint16 slave) function in the simple_test program? Is this function complete? Don't you need a return value for this function, since you declared this as an integer?

Thank you for your help!!

miscz, 2019-03-18:
I talked with an employee of the vendor, who said I have to program the PDOs like listed in the ESI file. 

- Is the moog drive example [hefloryd/SOEM@22d8f5d](https://github.com/hefloryd/SOEM/commit/22d8f5d) a good starting pointing in order to program the PDO mapping?

Sorry I am still confused how I can make it working.

An other option I was thinking of, is to convert the ESI file into a SII binary like mentioned in this post - [ESI to SII binary](https://markosankovic.github.io/2016/06/28/esi-to-sii-binary.html).

- Can I provide this SII binary to SOEM or do I have to write it somehow on the slave?

ArthurKetels, 2019-03-18:
Yes the moog solution example is similar to your problem. If you read the ESI file init sequence it is easy transferable to the moog example. Make sure to follow the same steps in the same order as the ESI file. 

Generating or altering an SII file and writing that to the slave eeprom is possible but not advisable if you are not an expert. It is possible the slave firmware can not handle the updated information. And some slaves will not allow you to change the eeprom, or do not have  an actual eeprom but emulate it.

But aside to all this, the slave should pass the slave conformance test. A work around for a buggy slave implementation is not a real solution, and others will experience the same issues again and again.

nakarlsson, 2019-04-29:
I'll close, seems to be taken care of

closed 2019-04-29 by nakarlsson