OpenEtherCATsociety/SOEM #262: what if The wkc doesnt match in crc error case?

yuz835, 2019-03-15:
if master sends write to multiple slave ram with mailbox config
and The frame got corrupted in The last slave.

master will sends The write again， but apparently The wkc wont match as expected as The first several slaves' mailbox have already been successfuly written before.
same case applies to buffermode as well

what's The right action of master?

ArthurKetels, 2019-03-15:
I would not recommend using BRD/BWR on other parts of the slave than the registers. Any memory that is under control of a SM would behave different depending on the state. This means that a broadcast datagram is not guaranteed to have the same result every time you send it.

If you want to control multiple mailboxes then you have to control state too. So you have to use single addressing modes only.

An optimization could be to use BWR the first time and revert to single mode when the WCK is lower than expected. One can assume that the BRW will succeed in most cases so the slow path is rarely taken.

yuz835, 2019-03-15:
to make another example

master just writes one slave with single datagram cmd in single mode

if The master sends write to The first slave ram with SM controlled，and it's successfully written.

but The frame got corrput （crc err）in The second frame.
The master resends The frame，and The wkc wont match as expected.

ArthurKetels, 2019-03-15:
That is why state is also important. Next to reading or writing to a SM controlled memory area you can read the state of the SM. Then it is immediately clear what happened and how to proceed.

But this all is not new. That is why the resilient mailbox protocol is used with master and slave. This is implemented in SOEM and SOES (and all other standard compliant master and slave implementations). As long as you use the standard functions provided all those corner cases are covered and stable operation is guaranteed.

The  nice thing of SOEM is that all sources are available. So you can check for yourself how these problems are solved.

yuz835, 2019-03-16:
i got your ideas now
apparently master can read the state SM to resolve it
thanks for your explain
please close it

ArthurKetels, 2019-03-16:
You can close this item too if you want.
To read up on the SM state machine see:
https://download.beckhoff.com/download/Document/io/ethercat-development-products/ethercat_esc_datasheet_sec1_technology_2i3.pdf
https://download.beckhoff.com/download/Document/io/ethercat-development-products/ethercat_esc_datasheet_sec2_registers_2i9.pdf

closed 2019-03-16 by ArthurKetels