OpenEtherCATsociety/SOEM #263: Some questions about issue #177

WangKaimeng, 2019-03-16:
I try the code as #177 says, the inputs is right, but the output is not what I want.
my struct is as follow:
typedef struct PACKED
{
uint16 Ctrlword_6040;
uint32 TargetPos_607A;
uint32 VelocityOffset_60B1;
uint16 TorqueOffset_60B2;
} Copley_CyclicPos_outputs;
and Ctrlword_6040 equals 0x0100, TargetPos_607A equals 0x00000005, VelocityOffset_60B1 equals 0x00000007, TorqueOffset_60B2 equals 0x0006
Therefore, the ec_slave[1].outputs that I want is 0001 0000 0500 0000 0700 0600
but the actual outputs for 10 times are
O: 05 01 00 00 05 00 00 00 07 00 00 00
O: 09 01 00 00 05 00 00 00 07 00 00 00
O: 0d 01 00 00 05 00 00 00 07 00 00 00
O: 12 01 00 00 05 00 00 00 07 00 00 00
O: 17 01 00 00 05 00 00 00 07 00 00 00
O: 1b 01 00 00 05 00 00 00 07 00 00 00
O: 20 01 00 00 05 00 00 00 07 00 00 00
O: 24 01 00 00 05 00 00 00 07 00 00 00
O: 29 01 00 00 05 00 00 00 07 00 00 00
It's so strange that the Ctrlword_6040 is always changed and TorqueOffset_60B2 is equal to zero but not 0x0006.
Besides, this is my PDO map:
PDO mapping according to CoE :
SM2 outputs
addr b index: sub bitl data_type name
[0x0000.0] 0x6040:0x00 0x10
[0x0002.0] 0x607A:0x00 0x20
[0x0006.0] 0x60B1:0x00 0x20
[0x000A.0] 0x60B2:0x00 0x10
SM3 inputs
addr b index: sub bitl data_type name
[0x000C.0] 0x6061:0x00 0x08
[0x000D.0] 0x6061:0x00 0x08
[0x000E.0] 0x6041:0x00 0x10
[0x0010.0] 0x6064:0x00 0x20
[0x0014.0] 0x60F4:0x00 0x20
[0x0018.0] 0x606C:0x00 0x20
[0x001C.0] 0x6077:0x00 0x10
I hope to get some suggestions! Thanks!

WangKaimeng, 2019-03-16:
> What plattform are you running on?
> Best guess is Win32 and you use win32 simpletest.
> 1 . Ctrl Word is overwritten by the IOmap[0]++ example code
> 2. Inputs overwrite Outputs due to missing proper use of PACKED, Windows use PACKED_BEGIN/PACKED_END , simple PACKED is for Linux.

As you guess,I run it on win32 platform. 
Could you please provide an example about how to use PACKED_BEGIN/PACKED_END in the code? Just replace PACKED with PACKED_BEGIN/PACKED_END?

nakarlsson, 2019-03-16:
check ethercattype.h for example.

WangKaimeng, 2019-03-17:
> ethercattype.h

I try it and it works!
Thanks a lot! You can close this issue.

closed 2019-03-17 by nakarlsson