OpenEtherCATsociety/SOEM #264: how the state affect the slave?

meiguwiki, 2019-03-19:
the slave have INIT/PREOP/SAFEOP/OP states etc.
how these status affect the blocks in the slave blocks?
I know there is a page to explain 
https://infosys.beckhoff.com/english.php?content=../content/1033/ax2000-b110/html/bt_ecbasics_ecstatemachine.htm&id=14011

but I didnt find anything related to watchdog/DC etc, 
are these only enabled in OP state?

ArthurKetels, 2019-03-19:
The reason you didn't find anything about the function blocks of an ESC in relation to the operational states is because there is no relation. The operational state machine (OSM) is a protocol on top of the functional slave functions. 

DC control block is completely separate from OSM.
Syncmanagers, FMMU functionalities are completely separate from OSM, but the slave firmware will check and enable/disable depending on OSM.
ESC hardware watchdog is connected to SM but not OSM.
Error counters are completely separate from OSM.
Port control is completely separate from OSM.
Eeprom state machine is completely separate from OSM.

meiguwiki, 2019-03-19:
okay,
so the watchdog is connected to SM,
when we enable SM, and the state hasnt reached to OP state,
will the process watchdog timeout as master hasnt exchange data with slave thru SM?

meiguwiki, 2019-03-19:
this is what I observed in lan9252 at the startup with state INIT
when master enable SM and config SM, 
and try to write SM protected mailbox,
it's always failing with wkc 0

but when state INIT is changed to OP, 
master can successfully write mailbox with wkc counted

that's why I'm asking here as I thought these should be all controlled by firmware.
but it seems SM is also controlled in hardware with OSM

ArthurKetels, 2019-03-19:
As I understand the ESC watchdog it first needs to be triggered in order to start counting. Only when the count goes to zero a timeout event occurs. So if no data exchange has happened there is no trigger. Only after the first trigger the watchdog is "life".
One comment here is that I never used the ESC watchdog in my projects so the real behavior could be different from the documentation.
https://download.beckhoff.com/download/Document/io/ethercat-development-products/ethercat_esc_datasheet_sec1_technology_2i3.pdf

ArthurKetels, 2019-03-19:
A SM should work in INIT state. It is just that the PDI side does not respond. In mailbox mode a SM needs two sides to keep the memory region active. The master and the PDI. When the master writes to a mailbox the mailbox gets locked until the PDI reads it. Vice versa a read mailbox is locked for the master until the PDI writes to it. 

To make a statement if the SM works or not you have to read its state.

nakarlsson, 2019-04-29:
questions answered , close since inactive

closed 2019-04-29 by nakarlsson