OpenEtherCATsociety/SOEM #265: Default PDO mappings question

jopado1, 2019-03-19:
I am running SSC v5.12 on an Infineon XMC48 Relax kit.  When I run slaveinfo I expect to see some default PDO mappings per the Infineon documentation. However slaveinfo reports no default mappings.  What am I missing?  Here's the slaveinfo output with -map:

Starting slaveinfo.
ec_init on memac1 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Dev01
 Output size: 72bits
 Input size: 72bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000319 ID: 00000000 Rev: 00000000
 SM0 A:0010 L:32768 F:26000100 Type:1
 SM1 A:0014 L:32768 F:22000100 Type:2
 SM2 A:0018 L:2304 F:64000100 Type:3
 SM3 A:001c L:2304 F:20000100 Type:4
 FMMU0 Ls:00000000 Ll:2304 Lsb:0 Leb:7 Ps:0018 Psb:0 Ty:02 Act:01
 FMMU1 Ls:09000000 Ll:2304 Lsb:0 Leb:7 Ps:001c Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 23 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  SM3 inputs
     addr b   index: sub bitl data_type    name
End slaveinfo, close socket

Thank you.

ArthurKetels, 2019-03-19:
As far as I can tell from the program output you have your endianess reversed.

jopado1, 2019-03-20:
My SOEM is running on a big-endian machine.  I have EC_BIG_ENDIAN defined in ethercattype.h.  I'll look around for some byte order issues in the coe pieces...

ArthurKetels, 2019-03-20:
Can you run SOEM on some little endian box for tests?

jopado1, 2019-03-20:
Here is output from an Ubuntu 16.04 VM running on an Intel box using the same slave. This was the output I was expecting from my big-endian machine :(  Still going through the source... Any hints appreciated! Thank you.
joe@ubuntu:~/SOEM/build/test/linux/slaveinfo$ sudo ./slaveinfo eth1 -map
SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on eth1 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:Dev01
 Output size: 72bits
 Input size: 72bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000319 ID: 00000000 Rev: 00000000
 SM0 A:1000 L: 128 F:00010026 Type:1
 SM1 A:1400 L: 128 F:00010022 Type:2
 SM2 A:1800 L:   9 F:00010064 Type:3
 SM3 A:1c00 L:   9 F:00010020 Type:4
 FMMU0 Ls:00000000 Ll:   9 Lsb:0 Leb:7 Ps:1800 Psb:0 Ty:02 Act:01
 FMMU1 Ls:00000009 Ll:   9 Lsb:0 Leb:7 Ps:1c00 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 23 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x7000:0x01 0x10 UNSIGNED16   OUT_GEN_INT1
  [0x0002.0] 0x7000:0x02 0x10 UNSIGNED16   OUT_GEN_INT2
  [0x0004.0] 0x7000:0x03 0x10 UNSIGNED16   OUT_GEN_INT3
  [0x0006.0] 0x7000:0x04 0x10 UNSIGNED16   OUT_GEN_INT4
  [0x0008.0] 0x7000:0x05 0x01 BOOLEAN      OUT_GEN_Bit1
  [0x0008.1] 0x7000:0x06 0x01 BOOLEAN      OUT_GEN_Bit2
  [0x0008.2] 0x7000:0x07 0x01 BOOLEAN      OUT_GEN_Bit3
  [0x0008.3] 0x7000:0x08 0x01 BOOLEAN      OUT_GEN_Bit4
  [0x0008.4] 0x7000:0x09 0x01 BOOLEAN      OUT_GEN_Bit5
  [0x0008.5] 0x7000:0x0A 0x01 BOOLEAN      OUT_GEN_Bit6
  [0x0008.6] 0x7000:0x0B 0x01 BOOLEAN      OUT_GEN_Bit7
  [0x0008.7] 0x7000:0x0C 0x01 BOOLEAN      OUT_GEN_Bit8
  SM3 inputs
     addr b   index: sub bitl data_type    name
  [0x0009.0] 0x6000:0x01 0x10 UNSIGNED16   IN_GEN_INT1
  [0x000B.0] 0x6000:0x02 0x10 UNSIGNED16   IN_GEN_INT2
  [0x000D.0] 0x6000:0x03 0x10 UNSIGNED16   IN_GEN_INT3
  [0x000F.0] 0x6000:0x04 0x10 UNSIGNED16   IN_GEN_INT4
  [0x0011.0] 0x6000:0x05 0x01 BOOLEAN      IN_GEN_Bit1
  [0x0011.1] 0x6000:0x06 0x01 BOOLEAN      IN_GEN_Bit2
  [0x0011.2] 0x6000:0x07 0x01 BOOLEAN      IN_GEN_Bit3
  [0x0011.3] 0x6000:0x08 0x01 BOOLEAN      IN_GEN_Bit4
  [0x0011.4] 0x6000:0x09 0x01 BOOLEAN      IN_GEN_Bit5
  [0x0011.5] 0x6000:0x0A 0x01 BOOLEAN      IN_GEN_Bit6
  [0x0011.6] 0x6000:0x0B 0x01 BOOLEAN      IN_GEN_Bit7
  [0x0011.7] 0x6000:0x0C 0x01 BOOLEAN      IN_GEN_Bit8
End slaveinfo, close socket
End program

ArthurKetels, 2019-03-20:
Hmm, looking at slaveinfo.c I see big endian is broken. All library functions are endian aware, but slaveinfo misses quite a few spots.

jopado1, 2019-03-22:
I've got the endian issues patched up and working with some changes to slaveinfo.c and ethercatconfig.c. If I may ask an additional question....
The last 8 outputs shown for the XMC48 represent led's on the slave.  I can control led3 for example using SDOs with a snippet like this:

```
             int8_t rdvar, wrvar;
             int rdl = 1, wrl = 1;
             int slave = 1;
             int wkc;
             wkc = ec_SDOread(slave, 0x7000 , 0x07, FALSE, &rdl, &rdvar, EC_TIMEOUTRXM);
             printf("LED test: SDOread: wkc=%d; led3=%d\n", wkc, rdvar);
             
             wrvar= rdvar ? 0 : 1;
             wkc = ec_SDOwrite(slave, 0x7000 , 0x07, FALSE, wrl, &wrvar, EC_TIMEOUTRXM);
             printf("LED test: SDOwrite: wkc=%d; led3=%d\n", wkc, wrvar);
             
             wkc = ec_SDOread(slave, 0x7000 , 0x07, FALSE, &rdl, &rdvar, EC_TIMEOUTRXM);
             printf("LED test: SDOread: wkc=%d; led3=%d\n", wkc, rdvar);
```

However I want to use PDOs and cannot seem to get this working on either my big-endian or little endian machine.  I am using my little-endian machine until I get this resolved just in case there are additional issues with big. So using simple_test as a guide I have a snippet like the following but I cannot get the leds to light.
```
#define XMC4800_SLAVE 1
typedef struct PACKED
{
	uint16_t unused1;
	uint16_t unused2;
	uint16_t unused3;
	uint16_t unused4;
	uint8_t	led;
//	boolean  led1;
//	boolean  led2;
//	boolean  led3;
//	boolean  led4;
//	boolean  led5;
//	boolean  led6;
//	boolean  led7;
//	boolean  led8;
} in_xmc48_t;
...
         ec_statecheck(0, EC_STATE_SAFE_OP,  EC_TIMEOUTSTATE * 4);
         
         printf("Slave %d Name: %s\n", 0, ec_slave[0].name);
         printf("Slave %d Name: %s\n", XMC4800_SLAVE, ec_slave[XMC4800_SLAVE].name);
         
         in_xmc48 = (in_xmc48_t*)ec_slave[XMC4800_SLAVE].outputs;
         in_xmc48->unused1 = 0x1234;
         in_xmc48->unused2 = 0x5678;
         in_xmc48->unused3 = 0x9abc;
         in_xmc48->unused4 = 0xdef0;
         in_xmc48->led = 0xFF;
...
         printf("Slave %d (%s) State: %d\n", 0, ec_slave[0].name, ec_slave[0].state);
         printf("Slave %d (%s) State: %d\n", XMC4800_SLAVE, ec_slave[XMC4800_SLAVE].name, ec_slave[XMC4800_SLAVE].state);
         
         if (ec_slave[0].state == EC_STATE_OPERATIONAL )
         {
            printf("Operational state reached for all slaves.\n");
            inOP = TRUE;
                /* cyclic loop */
            for(i = 1; i <= 500; i++)
            {
               ec_send_processdata();
               wkc = ec_receive_processdata(EC_TIMEOUTRET);

                    if(wkc >= expectedWKC)
                    {
                        printf("Processdata cycle %4d, WKC %d , O:", i, wkc);

                        for(j = 0 ; j < oloop; j++)
                        {
                            printf(" %2.2x", *(ec_slave[XMC4800_SLAVE].outputs + j));
                        }

                        printf(" I:");
                        for(j = 0 ; j < iloop; j++)
                        {
                            printf(" %2.2x", *(ec_slave[XMC4800_SLAVE].inputs + j));
                        }
                        printf(" T:%lld\r",ec_DCtime);
                        needlf = TRUE;
                    }
                    osal_usleep(5000);
                }
```
And my console output looks normal except no led's are lit so I am not communicating with the slave they way I expect.  Here's the output:
Slaves mapped, state to SAFE_OP.
Slave 0 Name: 
Slave 1 Name: Dev01
segments : 1 : 18 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Slave 0 () State: 8
Slave 1 (Dev01) State: 2
Operational state reached for all slaves.
Processdata cycle  500, WKC 3 , O: 34 12 78 56 bc 9a f0 de 81 I: 00 00 00 00 00 00 00 00 00 T:3348450081087945041 State[0,1]:  8  2
Request init state for all slaves
Slave 0 () State: 1
Slave 1 (Dev01) State: 2
End simple test, close socket

Any ideas?  Excuse the naive question I'm a noob with the ethercat. Thank you.

ArthurKetels, 2019-03-22:
From your data it looks like the master functions as intended. You could post the wireshark dump from your test. That will make sure all data on the wire is correct.
 
But do you know for sure your slave is working? Can you debug what happens with the PDO in the slave?

jopado1, 2019-03-22:
This was a slave side issue - my bad. There were two functions in the slave for copying inputs bits to ESC memory, and outputs from ESC memory. Once I patched these up things are working as expected.  Thank you for you help on this! I would like to give you my mods for the big-endian changes for the next big-endian user. Please advise on how you prefer these.

nakarlsson, 2019-03-23:
Please create a pull request, single commit with clear description

nakarlsson, 2019-04-08:
Any update, can you provide a pull request?

jopado1, 2019-04-09:
I'm getting a permission error:
joe@joespc:~/SOEM$ git push -u origin be-host-fixes
Username for 'https://github.com': jopado1@yahoo.com
Password for 'https://jopado1@yahoo.com@github.com':
remote: Permission to OpenEtherCATsociety/SOEM.git denied to jopado1.
fatal: unable to access 'https://github.com/OpenEtherCATsociety/SOEM.git/': The requested URL returned error: 403

I'm not a daily git user so maybe I'm doing something wrong. Have added and commited to my local branch but can't push:
joe@joespc:~/SOEM$ git branch -a
* be-host-fixes
  master
  remotes/origin/HEAD -> origin/master
  remotes/origin/master

nakarlsson, 2019-04-09:
Create a branch or fork from master.  Commit your change there and push. After that you can easily create a PR with help from the GUI on GitHub, select a PR from your branch/fork to master.

nakarlsson, 2019-04-29:
I'll address the PR for the upcoming SOEM release

closed 2019-04-29 by nakarlsson