OpenEtherCATsociety/SOEM #266: Wago 750-354 .. simple_test output

Tugev, 2019-03-20:
Hi, I have a Wago 750-354 ethercat fieldbus coupler and when I execute the simple_test, I get the following output:

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp4s0f1 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 0 0 0 0
Request operational state for all slaves
Calculated workcounter 0
Not all slaves reached operational state.
Slave 1 State=0x11 StatusCode=0x9000 : Unknown

Request init state for all slaves
End simple test, close socket
End program

For slaveinfo output: 

SOEM (Simple Open EtherCAT Master)
Slaveinfo
Starting slaveinfo
ec_init on enp4s0f1 succeeded.
1 slaves found and configured.
Calculated workcounter 0
Not all slaves reached safe operational state.
Slave 1 State=11 StatusCode=9000 : Unknown

Slave:1
 Name:WAGO Series 750 Bus Coupler
 Output size: 0bits
 Input size: 0bits
 State: 17
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 00000021 ID: 07500354 Rev: 00000002
 SM0 A:1000 L: 512 F:00010026 Type:1
 SM1 A:1400 L: 512 F:00010022 Type:2
 SM2 A:1800 L:   0 F:00000064 Type:3
 SM3 A:2400 L:   0 F:00000000 Type:4
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 512 rd: 512 MBX protocols : 04
 CoE details: 33 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:0
End slaveinfo, close socket
End program

is this what suppose I get ?  I noticed this line in the simple test output "Not all slaves reached safe operational state" .. Thanks in advance for your help

ArthurKetels, 2019-03-20:
Can you post a wireshark dump from simple_test?

Tugev, 2019-03-20:
Hi Arthur, Thanks for your quick reply. 
I am new to thereat stuff , how can I do it (wireshark dump) ?

ArthurKetels, 2019-03-20:
Download and install wireshark on your test box (often available in the distribution repository). Start wireshark as root or sudo. Start capturing the same interface as the EtherCAT port (enp4s0f1). Start simple_test. Save the wireshark capture as .pcap file and post.

Tugev, 2019-03-21:
Hi, here is the wireshark file
[simple_test_wireshark.pcap.zip](https://github.com/OpenEtherCATsociety/SOEM/files/2991550/simple_test_wireshark.pcap.zip)

nakarlsson, 2019-03-21:
The slave can't start properly, it get stuck in state INIT with error/alstatuscode 0x9000

According to the Wago manual: 

0x9000 K bus error during first k bus boot.
 Ensure that all I/O modules are correctly connected.

Tugev, 2019-03-25:
Thanks a lot.

closed 2019-03-25 by Tugev