OpenEtherCATsociety/SOEM #267: Can not find the difference between C# SOEM and simple_test

maxlein, 2019-03-20:
Hi guys,

I am experimenting with SOEM on linux and I am trying to get some data of an IO Module.  
Only 1 slave on the bus.  

I also have a little C# app based on the SOEM Wrapper library for reference where I get data in! 
But I can't find any difference, both slaves are operational, same call goes out on wireshark but on linux I have only zeros..  

Don't know if some flag is wrong or where I should have a look.

I have attached 2 wireshark dumps with txt ending and some info from your tools.
Maybe your experienced eyes see the error in a second. ;)

[working.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2988826/working.txt)
[non_working.txt](https://github.com/OpenEtherCATsociety/SOEM/files/2988827/non_working.txt)

### Simple test output:  

SOEM (Simple Open EtherCAT Master)
Simple test
Starting simple test
ec_init on enp0s8 succeeded.
1 slaves found and configured.
Slaves mapped, state to SAFE_OP.
segments : 1 : 77 0 0 0
Request operational state for all slaves
Calculated workcounter 3
Operational state reached for all slaves.
ERROR : slave 1 lost
Processdata cycle    4, WKC 18 , O: 00 00 00 00 00 00 00 00 I: 00 00 00 00 00 00 00 00 T:585468007529054220
MESSAGE : slave 1 found
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.00 00 00 00 00 00 00 00 I: 00 00 00 00 00 00 00 00 T:585468007529054220

OK : all slaves resumed OPERATIONAL.
OK : all slaves resumed OPERATIONAL.

### SlaveInfo -map

Slaveinfo
Starting slaveinfo
ec_init on enp0s8 succeeded.
1 slaves found and configured.
Calculated workcounter 3

Slave:1
 Name:netX
 Output size: 208bits
 Input size: 408bits
 State: 4
 Delay: 0[ns]
 Has DC: 1
 DCParentport:0
 Activeports:1.0.0.0
 Configured address: 1001
 Man: 0000006c ID: 0000a72c Rev: 00020000
 SM0 A:1000 L: 128 F:00010036 Type:1
 SM1 A:1080 L: 128 F:00010032 Type:2
 SM2 A:1100 L:  26 F:00010074 Type:3
 SM3 A:1400 L:  51 F:00010030 Type:4
 FMMU0 Ls:00000000 Ll:  26 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
 FMMU1 Ls:0000001a Ll:  51 Lsb:0 Leb:7 Ps:1400 Psb:0 Ty:01 Act:01
 FMMUfunc 0:1 1:2 2:3 3:0
 MBX length wr: 128 rd: 128 MBX protocols : 04
 CoE details: 33 FoE details: 00 EoE details: 00 SoE details: 00
 Ebus current: 0[mA]
 only LRD/LWR:1
PDO mapping according to CoE :
  SM2 outputs
     addr b   index: sub bitl data_type    name
  [0x0000.0] 0x2000:0x01 0x08 Type 0x001E  1 Byte Out (0)
  [0x0001.0] 0x2000:0x02 0x08 Type 0x001E  1 Byte Out (1)
  [0x0002.0] 0x2000:0x03 0x08 Type 0x001E  1 Byte Out (2)
  [0x0003.0] 0x2000:0x04 0x08 Type 0x001E  1 Byte Out (3)
  [0x0004.0] 0x2000:0x05 0x08 Type 0x001E  1 Byte Out (4)
  [0x0005.0] 0x2000:0x06 0x08 Type 0x001E  1 Byte Out (5)
  [0x0006.0] 0x2000:0x07 0x08 Type 0x001E  1 Byte Out (6)
  [0x0007.0] 0x2000:0x08 0x08 Type 0x001E  1 Byte Out (7)
  [0x0008.0] 0x2000:0x09 0x08 Type 0x001E  1 Byte Out (8)
  [0x0009.0] 0x2000:0x0A 0x08 Type 0x001E  1 Byte Out (9)
  [0x000A.0] 0x2000:0x0B 0x08 Type 0x001E  1 Byte Out (10)
  [0x000B.0] 0x2000:0x0C 0x08 Type 0x001E  1 Byte Out (11)
  [0x000C.0] 0x2000:0x0D 0x08 Type 0x001E  1 Byte Out (12)
  [0x000D.0] 0x2000:0x0E 0x08 Type 0x001E  1 Byte Out (13)
  [0x000E.0] 0x2000:0x0F 0x08 Type 0x001E  1 Byte Out (14)
  [0x000F.0] 0x2000:0x10 0x08 Type 0x001E  1 Byte Out (15)
  [0x0010.0] 0x2000:0x11 0x08 Type 0x001E  1 Byte Out (16)
  [0x0011.0] 0x2000:0x12 0x08 Type 0x001E  1 Byte Out (17)
  [0x0012.0] 0x2000:0x13 0x08 Type 0x001E  1 Byte Out (18)
...

### EEPROM info:

EEPROM tool
ec_init on enp0s8 succeeded.
1 slaves found.
Slave 1 data
 PDI Control      : 0006
 PDI Config       : CC00
 Config Alias     : 0000
 Checksum         : 00B7
   calculated     : 00B7
 Vendor ID        : 0000006C
 Product Code     : 0000A72C
 Revision Number  : 00020000
 Serial Number    : 000113DD
 Mailbox Protocol : 0004
 Size             : 01FF = 65536 bytes
 Version          : 0001
End, close socket
End program

ArthurKetels, 2019-03-20:
Thanks for the wireshark dumps. They are very insightful.

The slave in question has a Hilsher NetX ESC. These do not handle packets in hardware but in some kind of microcode. A bit like the Texas Instruments PRU. The side effect is they are not quite as fast as pure hardware like the ET1100.

From the captures it seems that if two packets are send back-to-back with no time in between, and both packets trigger a SM, the NetX will not process the first frame. 

Your C# program sends all PDO data twice. The first packet is dropped, the second is processed and returns to the master. So it looks like it works. In reality it only works half.

The SOEM test program only sends PDO's  once. The LRD is back-to-back with the LWR. The LRD is dropped, the LWR succeeds.

Perhaps this is a firmware bug in the NetX? 

A work around is to add a small delay between each packet transmission (a few microseconds). But the real issue is in the slave.

Can you try with a slave that has no NetX?

maxlein, 2019-03-21:
Thanks for your help! Sounds very interesting, also saw these 2 packages from master but only 1 coming back but hoped this was due to some wireshark bug or so ;)

I have to say I forgot to mention that I tested SOEM on a virtual box environment!
Because after hours of annoyment we put it on a native linux machine and the slave returned correct data...  

So now it seems we have two "maybe" problems:
- Why do I get slave info and all data from virtualbox but no process data?
- And does the NetX is working correctly at all?

The NetX is mounted inside a Beckhoff IO Device which is configured and mounted itself on a plc.

ArthurKetels, 2019-03-21:
Good you mention the use of a VM. If  wireshark is also running on the same VM as SOEM it is possible the timestamps are off. At least we can not count on them to be correct. 

It is also possible the virtual network driver is the source of problems here. If the packets are reordered or concatenated EtherCAT will not function correctly. The only way to be sure is to sniff the packets outside the VM and preferably directly on the wire.

If you can get your hands on a managed network switch you can copy the EtherCAT traffic on a 3rd port. Connect a second PC on this port and capture the data with wireshark. Then you really know what is on the wire.

maxlein, 2019-03-21:
No, wireshark was running on Windows host. Will further look into it.

nakarlsson, 2019-04-29:
@maxlein , Active issue?

maxlein, 2019-04-29:
It looks like the bridged network was not working correctly with virtual box...
With vmware it worked and when using a usb ethernet adapter and mounting it directly in virtual box, its also working there.

nakarlsson, 2019-04-30:
ok, i'll close it than

closed 2019-04-30 by nakarlsson