OpenEtherCATsociety/SOEM #268: Recent changes?

stefanpoeter, 2019-03-20:
Haven't tracked the recent changes in this repository but slaveinfo suddenly gives me weird results. Slave 1 should be EK1100, Slave 2 EL2008, Slave 3 EL3011, Slave 4 EL4011 instead I get this:

> SOEM (Simple Open EtherCAT Master)
> Slaveinfo
> Starting slaveinfo
> ec_init on enp0s8 succeeded.
> Time:   23575.824 SDO slave:6 index:1c13.01 error:06090011 Subindex does not exist
> 7 slaves found and configured.
> Calculated workcounter 4
> Not all slaves reached safe operational state.
> Slave 1 State= 0 StatusCode=   0 : No error
> Slave 2 State= 0 StatusCode=   0 : No error
> Slave 3 State= 0 StatusCode=   0 : No error
> Slave 4 State= 0 StatusCode=   0 : No error
> 
> Slave:1
>  Name:? M:00000000 I:00000000
>  Output size: 0bits
>  Input size: 0bits
>  State: 0
>  Delay: 0[ns]
>  Has DC: 0
>  Activeports:0.0.0.0
>  Configured address: 1001
>  Man: 00000000 ID: 00000000 Rev: 00000000
>  FMMUfunc 0:0 1:0 2:0 3:0
>  MBX length wr: 0 rd: 0 MBX protocols : 00
>  CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: 0[mA]
>  only LRD/LWR:0
> PDO mapping according to SII :
> 
> Slave:2
>  Name:? M:00000000 I:00000000
>  Output size: 0bits
>  Input size: 0bits
>  State: 0
>  Delay: 0[ns]
>  Has DC: 0
>  Activeports:0.0.0.0
>  Configured address: 1002
>  Man: 00000000 ID: 00000000 Rev: 00000000
>  FMMUfunc 0:0 1:0 2:0 3:0
>  MBX length wr: 0 rd: 0 MBX protocols : 00
>  CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: 0[mA]
>  only LRD/LWR:0
> PDO mapping according to SII :
> 
> Slave:3
>  Name:? M:00000000 I:00000000
>  Output size: 0bits
>  Input size: 0bits
>  State: 0
>  Delay: 0[ns]
>  Has DC: 0
>  Activeports:0.0.0.0
>  Configured address: 1003
>  Man: 00000000 ID: 00000000 Rev: 00000000
>  FMMUfunc 0:0 1:0 2:0 3:0
>  MBX length wr: 0 rd: 0 MBX protocols : 00
>  CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: 0[mA]
>  only LRD/LWR:0
> PDO mapping according to SII :
> 
> Slave:4
>  Name:? M:00000000 I:00000000
>  Output size: 0bits
>  Input size: 0bits
>  State: 0
>  Delay: 0[ns]
>  Has DC: 0
>  Activeports:0.0.0.0
>  Configured address: 1004
>  Man: 00000000 ID: 00000000 Rev: 00000000
>  FMMUfunc 0:0 1:0 2:0 3:0
>  MBX length wr: 0 rd: 0 MBX protocols : 00
>  CoE details: 00 FoE details: 00 EoE details: 00 SoE details: 00
>  Ebus current: 0[mA]
>  only LRD/LWR:0
> PDO mapping according to SII :
> 
> Slave:5
>  Name:EL3001
>  Output size: 0bits
>  Input size: 32bits
>  State: 4
>  Delay: 0[ns]
>  Has DC: 1
>  DCParentport:0
>  Activeports:1.1.0.0
>  Configured address: 1005
>  Man: 00000002 ID: 0bb93052 Rev: 00140000
>  SM0 A:1000 L: 128 F:00010026 Type:1
>  SM1 A:1080 L: 128 F:00010022 Type:2
>  SM2 A:1100 L:   0 F:00000004 Type:3
>  SM3 A:1180 L:   4 F:00010020 Type:4
>  FMMU0 Ls:00000002 Ll:   4 Lsb:0 Leb:7 Ps:1180 Psb:0 Ty:01 Act:01
>  FMMUfunc 0:2 1:3 2:0 3:0
>  MBX length wr: 128 rd: 128 MBX protocols : 0c
>  CoE details: 07 FoE details: 01 EoE details: 00 SoE details: 00
>  Ebus current: 130[mA]
>  only LRD/LWR:0
> PDO mapping according to CoE :
>   SM2 outputs
>      addr b   index: sub bitl data_type    name
>   SM3 inputs
>      addr b   index: sub bitl data_type    name
>   [0x0002.0] 0x6000:0x01 0x01 BOOLEAN      Underrange
>   [0x0002.1] 0x6000:0x02 0x01 BOOLEAN      Overrange
>   [0x0002.2] 0x6000:0x03 0x02 BIT2         Limit 1
>   [0x0002.4] 0x6000:0x05 0x02 BIT2         Limit 2
>   [0x0002.6] 0x6000:0x07 0x01 BOOLEAN      Error
>   [0x0002.7] 0x0000:0x00 0x01
>   [0x0003.0] 0x0000:0x00 0x06
>   [0x0003.6] 0x6000:0x0F 0x01 BOOLEAN      TxPDO State
>   [0x0003.7] 0x6000:0x10 0x01 BOOLEAN      TxPDO Toggle
>   [0x0004.0] 0x6000:0x11 0x10 INTEGER16    Value
> 
> Slave:6
>  Name:EL4031
>  Output size: 16bits
>  Input size: 0bits
>  State: 4
>  Delay: 140[ns]
>  Has DC: 1
>  DCParentport:1
>  Activeports:1.1.0.0
>  Configured address: 1006
>  Man: 00000002 ID: 0fbf3052 Rev: 00130000
>  SM0 A:1000 L: 128 F:00010026 Type:1
>  SM1 A:1080 L: 128 F:00010022 Type:2
>  SM2 A:1100 L:   2 F:00010024 Type:3
>  SM3 A:1180 L:   0 F:00000000 Type:4
>  FMMU0 Ls:00000000 Ll:   2 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:02 Act:01
>  FMMUfunc 0:1 1:3 2:0 3:0
>  MBX length wr: 128 rd: 128 MBX protocols : 0c
>  CoE details: 03 FoE details: 01 EoE details: 00 SoE details: 00
>  Ebus current: 140[mA]
>  only LRD/LWR:0
> PDO mapping according to CoE :
>   SM2 outputs
>      addr b   index: sub bitl data_type    name
>   [0x0000.0] 0x7000:0x01 0x10 INTEGER16    Analog output
>   SM3 inputs
>      addr b   index: sub bitl data_type    name
> 
> Slave:7
>  Name:EL5001
>  Output size: 0bits
>  Input size: 40bits
>  State: 4
>  Delay: 285[ns]
>  Has DC: 1
>  DCParentport:1
>  Activeports:1.1.0.0
>  Configured address: 1007
>  Man: 00000002 ID: 13893052 Rev: 03fd0000
>  SM0 A:1800 L: 246 F:00010026 Type:1
>  SM1 A:18f6 L: 246 F:00010022 Type:2
>  SM2 A:1000 L:   0 F:00000004 Type:3
>  SM3 A:1100 L:   5 F:00010020 Type:4
>  FMMU0 Ls:00000006 Ll:   5 Lsb:0 Leb:7 Ps:1100 Psb:0 Ty:01 Act:01
>  FMMUfunc 0:2 1:3 2:0 3:0
>  MBX length wr: 246 rd: 246 MBX protocols : 0c
>  CoE details: 07 FoE details: 01 EoE details: 00 SoE details: 00
>  Ebus current: 120[mA]
>  only LRD/LWR:0
> PDO mapping according to CoE :
>   SM2 outputs
>      addr b   index: sub bitl data_type    name
>   SM3 inputs
>      addr b   index: sub bitl data_type    name
>   [0x0006.0] 0x3101:0x01 0x08 UNSIGNED8    Status
>   [0x0007.0] 0x3101:0x02 0x20 UNSIGNED32   Value
> End slaveinfo, close socket
> End program
>

ArthurKetels, 2019-03-20:
Can you post a wireshark dump?

stefanpoeter, 2019-03-20:
I don't know what happened, restartet the machine and everything works.

closed 2019-03-20 by stefanpoeter