OpenEtherCATsociety/SOEM #269: Motor vibration and Sync loss

dabhijeet, 2019-03-25:
Hi guys,

We are experimenting with SOEM on linux. After some changes in the redtest we are able to rotate the motor.

We are using Xenomai in our system. After testing the code we found that the sync is losing nearly after 1 day and even when we are rotating the motor using PID controller there is vibrations in movement.

We think that it maybe because the Network driver is not real time which is causing jitter. Maybe RTnet is required to get smooth motion with proper sync.

Please let us know what can be the cause of this problem.

Thanks in advance.

ArthurKetels, 2019-03-25:
Ehhh, the is not a lot of information you are giving us. How should I know?

I have run lots and lots of systems with linux that keep sync for as long as you want. There could be issues with your hardware both on the master and slave side. Good instrumentation is very important in finding these kind of failures. 

On the master side you can install latency triggers.

If you build the slave yourself then debugging there should be easy. Otherwise the slave often latches the sync fail reason and you can read it out to get a clue.

You could also swap master and/or slave system to see if it is something hardware or software specific.

What ind of slave controller are you running?

dabhijeet, 2019-03-26:
We are using RTA stepping motor drives (4 ms Cycle time). There should not be any problem on slave side since it is a standard industrial drive and we have rotated the motor using TwinCAT and it was working properly without any vibration. 

What kind of instrumentation (hardware) should we use to find the failures from master side?

Is Xenomai and RTnet necessary to get the proper sync without vibration?

If the Network driver is not real time will it cause jitter?

We are using Xenomai 3.0.7

ArthurKetels, 2019-03-26:
First you could log a wireshark trace. It has timestamps that could give a clue about system or driver jitter. You can also use the ehtercat time that gets returned by soem after each PDO transfer. Compare that with the real time tread clock to see if there is excessive jitter or spikes.

For a 4ms system even standard linux would be enough (approx 500us jitter). Preempt-rt patch would make it very good (50us). No special need to go to xenomai and rtnet.

I have seen hardware however that has spikes over 10ms because of bios activity. No amount of RT system can cure that.

Again, instrument (can be software only) to get to the problem.

dabhijeet, 2019-03-28:
Okay we will try wireshark for debugging.

In some websites we have seen that for best performance of EtherCAT, a portion of the CPU (i.e. if we have quad core then one core out of four) is dedicated for EtherCAT task only. Even sometimes if we have 2 or more RJ45 port then one port is dedicated for EtherCAT task. 

How can we allocate one core and one RJ45 port for EtherCAT in linux.

dabhijeet, 2019-04-08:
When we used Wireshark we got the following result..
[image]
This is the result of Normal Linux (CPU i5)

[image]
This is the result of Xenomai (CPU celeron)

[image]
This is the result of Xenomai (CPU i5)

[image]
This is the result of standard industrial motion controller (ARM)

ArthurKetels, 2019-04-09:
It is a pity you did not post the complete wireshark capture. An averaged plot does not tell so much about packet to packet timing. It are often the small details that make or brake the solution.

Anyway, what bothers me that the linux packets/s is way off from the ARM comparison. I would expect the average pkt/s to be exactly the same (2000) for all platforms.

It is possible your PDO real time task has timing issues. Can you post that part of your code?

dabhijeet, 2019-04-12:
`/*

#include "robot_ethercat_controller/ethercat_manager/ethercat_manager.h"

#include <unistd.h>
#include <stdio.h>
#include <time.h>

#include <boost/ref.hpp>
#include <boost/interprocess/sync/scoped_lock.hpp>
extern "C"{
#include <robot_ethercat_controller/ethercattype.h>
#include <robot_ethercat_controller/nicdrv.h>
#include <robot_ethercat_controller/ethercatbase.h>
#include <robot_ethercat_controller/ethercatmain.h>
#include <robot_ethercat_controller/ethercatdc.h>
#include <robot_ethercat_controller/ethercatcoe.h>
#include <robot_ethercat_controller/ethercatfoe.h>
#include <robot_ethercat_controller/ethercatconfig.h>
#include <robot_ethercat_controller/ethercatprint.h>
}

namespace 
{
static const unsigned cycletime = 4000000; // 4 ms
static const unsigned EC_TIMEOUTMON = 500;
static const int NSEC_PER_SECOND = 1e+9;

bool sync_flag = true; ///This hold the current status of EtherCat Sync

void timespecInc(struct timespec &tick, int nsec)
{
  tick.tv_nsec += nsec;
  while (tick.tv_nsec >= NSEC_PER_SECOND)
    {
      tick.tv_nsec -= NSEC_PER_SECOND;
      tick.tv_sec++;
    }
}

void handleErrors()
{
  // one ore more slaves are not responding 
  ec_group[0].docheckstate = FALSE;
  ec_readstate();

  // declaring argument of time() 
  time_t my_time = time(NULL); 
  
   
  for (int slave = 1; slave <= ec_slavecount; slave++)
  {
    if ((ec_slave[slave].group == 0) && (ec_slave[slave].state != EC_STATE_OPERATIONAL))
    {
      ec_group[0].docheckstate = TRUE;
      if (ec_slave[slave].state == (EC_STATE_SAFE_OP + EC_STATE_ERROR))
      {
      
        sync_flag = false;
      
        fprintf(stderr, "ERROR : slave %d is in SAFE_OP + ERROR, attempting ack.\n", slave);
        // ctime() used to give the present time 
        printf("%s", ctime(&my_time)); 

        ec_slave[slave].state = (EC_STATE_SAFE_OP + EC_STATE_ACK);
        ec_writestate(slave);
      }
      else if(ec_slave[slave].state == EC_STATE_SAFE_OP)
      {
      
[... 557 lines cut ...]
  }
  return val;
}
/////////////////////////////////////////////////////////////////////////////////////////////////

template uint8_t EtherCatManager::writeSDO<char> (int slave_no, uint16_t index, uint8_t subidx, char value) const;
template uint8_t EtherCatManager::writeSDO<int> (int slave_no, uint16_t index, uint8_t subidx, int value) const;
template uint8_t EtherCatManager::writeSDO<short> (int slave_no, uint16_t index, uint8_t subidx, short value) const;
template uint8_t EtherCatManager::writeSDO<long> (int slave_no, uint16_t index, uint8_t subidx, long value) const;
template uint8_t EtherCatManager::writeSDO<unsigned char> (int slave_no, uint16_t index, uint8_t subidx, unsigned char value) const;
template uint8_t EtherCatManager::writeSDO<unsigned int> (int slave_no, uint16_t index, uint8_t subidx, unsigned int value) const;
template uint8_t EtherCatManager::writeSDO<unsigned short> (int slave_no, uint16_t index, uint8_t subidx, unsigned short value) const;
template uint8_t EtherCatManager::writeSDO<unsigned long> (int slave_no, uint16_t index, uint8_t subidx, unsigned long value) const;

template char EtherCatManager::readSDO<char> (int slave_no, uint16_t index, uint8_t subidx) const;
template int EtherCatManager::readSDO<int> (int slave_no, uint16_t index, uint8_t subidx) const;
template short EtherCatManager::readSDO<short> (int slave_no, uint16_t index, uint8_t subidx) const;
template long EtherCatManager::readSDO<long> (int slave_no, uint16_t index, uint8_t subidx) const;
template unsigned char EtherCatManager::readSDO<unsigned char> (int slave_no, uint16_t index, uint8_t subidx) const;
template unsigned int EtherCatManager::readSDO<unsigned int> (int slave_no, uint16_t index, uint8_t subidx) const;
template unsigned short EtherCatManager::readSDO<unsigned short> (int slave_no, uint16_t index, uint8_t subidx) const;
template unsigned long EtherCatManager::readSDO<unsigned long> (int slave_no, uint16_t index, uint8_t subidx) const;

} //End of namespace
`

dabhijeet, 2019-04-12:
Hi, I thought it is better to upload the whole code. Here, instead of using the single code we made an EtherCAT manager class and in its constructor we start the PDO thread.

ArthurKetels, 2019-04-12:
Thanks for posting. Sorry to say, but there is quite a bit wrong in your code.
For reference is is better to look at the SOEM red_test.c example.

a) The PDO exchange task should be running at real-time priority. All other tasks should be running at lower priority. In your code everything is running at the same (non-real-time?) priority..
b) There should be no mutexes in your PDO exchange task. In your code you use : boost::mutex::scoped_lock lock(mutex);. You either have to do all PDO handling in real-time context or you have to copy the PDO data with lock-less principles (triple-buffer, RCU, etc.).
c) Confusingly you should not use CLOCK_REALTIME but CLOCK_MONOTONIC. Google the differences to understand why.
d) handleErrors() should be run in non-real-time context. You can not call this from the PDO exchange task. Also try to rate limit the handleError() taks to 100Hz or so. Some use case might require better response time though.
e) Not related to the code but please check the timing of the NIC driver you use. Many drivers have packet coalescing and interrupt mitigation as a default setting. See also drvcomment.txt.

Configuring slaves with SDO_write should be done with the special hook fucntion SOEM provides. Otherwise handleErrors() will not work in all conditions. Because a slave could reset and recovery is only possible by reconfiguring the slave. And SOEM should know how to do that.

Again, have a good look at the samples provided by SOEM. I believe most of your timing issues have more to do with your coding than internal hardware or OS issues.

dabhijeet, 2019-04-15:
Thank you so much Sir, for taking the time to write the response. Now, we got a better idea of the issue. We will change our code as per redtest.c and then observe the results.

nakarlsson, 2019-04-29:
Questions answered

closed 2019-04-29 by nakarlsson

dabhijeet, 2019-05-06:
Regarding the vibration problem, we found out the issue. 
If we make the executable with the help of ROS. we can't do sudo rosrun .....
so in most of the websites they have given to use    
                                    sudo setcap cap_net_raw+ep <binary>     
before running the ROS node.
but if we do that then motor will run but it won't run smoothly. maybe some permission is missing.
The best solution to this problem is to use 
                                    sudo -s
in the ROS terminal to elevate permissions to superuser
After using this, ROS will run the motor smoothly.