OpenEtherCATsociety/SOEM #270: tutorial about groups

NoeMurr, 2019-03-27:
Hi all,

I need to know more about groups: how to configure them and how to use them.
Is there some tutorial or reference about it?

ArthurKetels, 2019-03-27:
Yes, we need to add some documentation on how this feature can be used. The problem is that in order to make a sample program it is hard to know what slave configuration to expect.

The idea is to first do a config_init to discover all slaves on the port.
Then the application assigns each found slave to a group, the default group being 0. The assignment is simply doing ec_slave[i].group = groupnumber;
In the standard configuration of soem EC_MAXGORUP is set to 2 so only groups 0 and 1 exist. If you need  more adjust the define and recompile.
Then do config_map_group for each group you have defined.
Last run the same number of real time loops that do the process data exchange with send_processdata_group and receive_processdata_group.

The real time loops can be either the same or asynchronous. Each group generates its own and independent data exchange with the slaves. Group level mapping data can be found in the ec_group array.

NoeMurr, 2019-03-28:
All right, thanks. 
So when I init the library I don't need to call `ec_config_map();`, I have to cal `config_map_group()` with a map foreach group, right?

f.e.
```C
// init phase 
/* initialise SOEM, bind socket to ifname */
   if (ec_init(ifname))
   {
      printf("ec_init on %s succeeded.\n",ifname);
      /* find and auto-config slaves */

       if ( ec_config_init(FALSE) > 0 )
      {
         printf("%d slaves found and configured.\n",ec_slavecount);

         ec_config_map(&IOmap);
```

ArthurKetels, 2019-03-28:
One bug, &IOmpa[i] should be &IOmap. The function itself takes care of the offset needed.

After you do a config_map_group the slave group number is fixed. Or you need to do another config_map_group. And theoretically this can be done in OP mode, but no PDO transfer is possible during the mapping. And mapping can take a long time. So practically no.

A way to circumvent this is to make yet another group for "dynamic" slaves. For example you have group A and B for fixed slaves and group C for dynamic slaves.  The real time PDO loops can then either do (AC),(B)  or (A),(BC). Depending on the C group belonging to A or B.

NoeMurr, 2019-03-29:
Ok, I'm working with C++ and I want to make an object that represents a group..
Can I initialize each group independently from the others??

ArthurKetels, 2019-03-29:
config_map_group can be used independent, But the config_init is global.

commit b978b4cb (2019-06-12): Improve documentation fixes #226 and fixes #270
closed 2019-06-12 by ArthurKetels