OpenEtherCATsociety/SOEM #273: If #define EC_VER1 is commented, make option -Wextra prevents successful compilation

Apollo3zehn, 2019-03-29:
I commented #define EC_VER1 to disable the default context. Try to build SOEM fails with the following errors:

```
/home/appveyor/projects/onedas-ethercat/native/SOEM/soem/ethercatconfig.c: In function ‘ecx_config_from_table’:
/home/appveyor/projects/onedas-ethercat/native/SOEM/soem/ethercatconfig.c:252:48: error: unused parameter ‘context’ [-Werror=unused-parameter]
 static int ecx_config_from_table(ecx_contextt *context, uint16 slave)
                                                ^
/home/appveyor/projects/onedas-ethercat/native/SOEM/soem/ethercatconfig.c:252:64: error: unused parameter ‘slave’ [-Werror=unused-parameter]
 static int ecx_config_from_table(ecx_contextt *context, uint16 slave)
                                                                ^
cc1: all warnings being treated as errors
CMakeFiles/soem.dir/build.make:88: recipe for target 'CMakeFiles/soem.dir/soem/ethercatconfig.c.o' failed
make[2]: *** [CMakeFiles/soem.dir/soem/ethercatconfig.c.o] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/soem.dir/all' failed
make[1]: *** [CMakeFiles/soem.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

#123 introduced the -Wextra option to treat warnings as errors and commenting #define EC_VER1 makes the light version of `ecx_config_from_table` active, cause the errors.

nakarlsson, 2019-04-08:
Addressed in https://github.com/OpenEtherCATsociety/SOEM/pull/280

nakarlsson, 2019-04-29:
Fixed in #280

closed 2019-04-29 by nakarlsson