OpenEtherCATsociety/SOEM #274: connect coupler ek1818 to pc

radwa4, 2019-03-29:
after downloading soem and following  steps .i connect pc with coupler ek1818 beckhoff  by ethernat cable  only and run simple test by write in terminal (./simple_test) print in terminal name of simple test .i don't know how check pc run as master and connect by coupler ?and do you must connect power to coupler or only ethernet cable by pc?

nakarlsson, 2019-03-30:
You need power

radwa4, 2019-04-03:
i connect pc by ek1818 coupler by ethernet and connect coupler by power 24
volt and when run code slave info  this what happened what i should do to
see slave coupler??

‫في السبت، 30 مارس 2019 في 6:40 م تمت كتابة ما يلي بواسطة ‪nakarlsson‬‏ <‪
notifications@github.com‬‏>:‬

Tugev, 2019-04-05:
Hi, you should run the code as a root (sudo) and also include the coupler socket name (usually eth0) in your terminal. For example:

`sudo ./simple_test eth0`
If it is not eth0, then use `ifconfig` to get the right coupler socket name.

radwa4, 2019-04-14:
can you tell me how turn on led on ethercat ?and how connect circuit on
output and input pins?

‫في الجمعة، 5 أبريل 2019 في 7:59 م تمت كتابة ما يلي بواسطة ‪Tugev‬‏ <‪
notifications@github.com‬‏>:‬

Tugev, 2019-04-14:
Hi , actually I have no much experience in etherCats but the input module is for reading and the output one is for writing. 
There is a function in the main.c file called:
 `void set_output_bit (uint16 slave_no, uint8 module_index, uint8 value)`  use it to write the channels in the output module.

For connecting the circuit, you should refer to the manual.

nakarlsson, 2019-04-29:
There are examples online, please read them

closed 2019-04-29 by nakarlsson

radwa4, 2019-05-03:
can you help me with links of this online examples?