OpenEtherCATsociety/SOEM #277: Is there a CoE slave simulator/emulator available (question)

jopado1, 2019-04-04:
Hello - 
I am using SOEM to control a servo drive. However I do not have the servo drive hardware yet. Is there any kind of software simulator/emulator I could use in the meantime? All I want to test is the ability to get the slave to operational and then drive the CANopen state machine to the 'Operation Enabled' state (using PDOs). I do have an Infineon XMC48 Relax Kit but I don't want to spend a lot of time developing for the slave.  Is there something close to an out of the box solution for this? All suggestions welcome.  Thanks!

ArthurKetels, 2019-04-05:
No open source slave simulator as far as I know. But there is no reason it could not be made. 

There could be a legal snag for commercial use however. Each EtherCAT slave needs a license from Beckhoff. Normally this is paid with the purchase of the slave controller. In the case of an open source software solution this is not.

nakarlsson, 2019-04-29:
I don't think there a more info to add

closed 2019-04-29 by nakarlsson