OpenEtherCATsociety/SOEM #281: cannot read data by bpf under the RTEMS system

farawayfromhome, 2019-04-18:
Initialization of bpf did not return an error when initializing the SOEM. In the process of detecting the slave, the answer data of the slave can be caught by wireshark, but the return value of the ‘read’ function is 0, and errno is also 0, which causes the receiving answer data to fail, and the slave is not found.

Thanks

farawayfromhome, 2019-04-18:
[image]

farawayfromhome, 2019-04-18:
I use bpf0

[image]

nakarlsson, 2019-04-18:
I think this is a rtems question @lounick or anyone else that known rtems.

lounick, 2019-04-24:
Hi sorry for the late reply. It fell through. I won't be in office the next days so I won't have any access to EtherCAT equipment.

Which version of RTEMS are you using? And how did you build the libsoem?

AFAIK the RTEMS patches for EtherCAT still haven't been merged (or at least I stopped following it much given that I switched projects).

farawayfromhome, 2019-04-25:
@lounick I am very grateful for your reply. 
The version of RTEMS I am currently using is 4.11. 
I loaded libbsd and the command is.../rtems/4.11/tools/bin/arm-rtems4.11-gcc.

 What should I do next? What is the patch for RTEMS? How do i implement this part of the code?(Sorry I have so much questions, because this is very urgent.)
 I can currently send data through bpf, but I can't receive it.

farawayfromhome, 2019-04-25:
@lounick  In addition, through the debug, it is found that the response data of the slave can be received in the cgem_recv function, but the application still cannot receive it.

lounick, 2019-05-02:
Hi sorry for being late again. I had some family obligations that kept me away from office.

The patch for building SOEM with RTEMS 4.11 can be found [here](https://gist.github.com/lounick/472843aeb078b8977fbcd781f48e2008). It was submitted but was not included in RTEMS yet AFAIK.

For an example you can check [here](https://github.com/Intermodalics/soem-rtems-example).

I'm not that experienced with either RTEMS or SOEM myself but I'll try to help as much as I can.

nakarlsson, 2019-05-10:
@farawayfromhome , are you up and runing now?

nakarlsson, 2019-05-15:
i guess the problem is fixed,, https://github.com/OpenEtherCATsociety/SOEM/issues/291

closed 2019-05-15 by nakarlsson

farawayfromhome, 2019-06-06:
> Hi sorry for being late again. I had some family obligations that kept me away from office.
>
> The patch for building SOEM with RTEMS 4.11 can be found [...]

> Hi sorry for being late again. I had some family obligations that kept me away from office.
>
> The patch for building SOEM with RTEMS 4.11 can be found [...]

The path to the patch cannot open now.

lounick, 2019-06-06:
It should be a public gist. That's weird.

anyway here it is again:

```
diff --git a/rtems/config/4.11/net/soem.bset b/rtems/config/4.11/net/soem.bset
new file mode 100644
index 0000000..6934991
--- /dev/null
+++ b/rtems/config/4.11/net/soem.bset
@@ -0,0 +1,20 @@
+#
+# Build set for EtherCAT SOEM
+#
+
+%define release 1
+
+#
+# RTEMS Version
+#
+%define rtems_version 4.11
+
+#
+# The RTEMS URL paths.
+#
+%include rtems-urls.bset
+
+#
+# Build SOEM.
+#
+net/soem
diff --git a/rtems/config/net/soem.cfg b/rtems/config/net/soem.cfg
new file mode 100644
index 0000000..7378e66
--- /dev/null
+++ b/rtems/config/net/soem.cfg
@@ -0,0 +1,22 @@
+#
+# SOEM
+#
+
+%if %{release} == %{nil}
+ %define release 1
+%endif
+
+%include %{_configdir}/rtems-bsp.cfg
+
+#
+# SOEM Version
+#
+
+#
+# Patch for RTEMS support.
+#
+
+#
+# NTP Build configuration
+#
+%include %{_configdir}/soem.cfg
diff --git a/source-builder/config/soem.cfg b/source-builder/config/soem.cfg
new file mode 100644
index 0000000..1ad96ea
--- /dev/null
+++ b/source-builder/config/soem.cfg
@@ -0,0 +1,72 @@
+#
+# SOEM Version 1.
+#
+# This configuration file configures, makes and installs SOEM.
+#
+
+%if %{release} == %{nil}
+%define release 1
+%endif
+
+Name:      soem-%{_host}-%{release}
+Summary:   SOEM is a Simple Open EtherCAT Master implementation.
+Version:   Master
+Release:   %{release}
+URL: 	   https://openethercatsociety.github.io/
+BuildRoot: %{_tmppath}/%{name}-root-%(%{__id_u} -n)
+
+#
+# SOEM
+#
+%source set soem git://github.com/lounick/SOEM.git?branch=integrate-with-RTEMS
+
+#
+# Prepare the source code.
+#
+%prep
+  build_top=$(pwd)
+
+  source_dir_soem="soem"
+  %source setup soem -q -n soem
+
+  cd ${build_top}
+  echo ${build_top}
+
+%build
+  build_top=$(pwd)
+
+  %{build_directory}
+
+  mkdir -p ${build_dir}
+  cd ${build_dir}
+
+  %{host_build_flags}
+
+  cmake \
+    -DRTEMS_TOOLS_PATH=%{with_tools} \
+    -DCMAKE_INSTALL_PREFIX=%{_libdir} \
+    -DCMAKE_SYSTEM_NAME=rtems \
+    -DHOST=%{_host} \
+    -DRTEMS_BSP=%{rtems_bsp} \
+    -DINCLUDE_DIR=%{_includedir} \
+    -DLIB_DIR=%{_libdir} \
+    -DHOST_C_FLAGS=%{host_cflags} \
+    -DHOST_CXX_FLAGS=%{host_cxxflags} \
+    -DHOST_LIBS=%{host_libs} \
+    ../${source_dir_soem}
+
+  # SOEM does not build on sync.rtems.org with jobs
+  %{__make} %{?_smp_mflags} all
+  # %{__make} all
+
+  cd ${build_top}
+
+%install
+  build_top=$(pwd)
+
+  %{__rmdir} $SB_BUILD_ROOT
+
+  cd ${build_dir}
+  %{__make} DESTDIR=$SB_BUILD_ROOT install
+  #%{__make} install
+  cd ${build_top}
```