OpenEtherCATsociety/SOEM #285: SOEM ecx_readPDOmapCA（）ReadError

lyhawk2007, 2019-04-22:
Soem platform USES STM32F429, now process data can be read correctly, from the device can enter the OP state, but Coe configuration data transmission is not successful, rval = ecx_readPDOmapCA(context, slave, &Osize, &Isize);The return is not normal. The correct values of OSize and ISize should be 128, but the return values are both 0.Enter Ecx_SDOerror (context, slave, index, subindex, etohl(aSDOp->ldata[0]));Branch.

nakarlsson, 2019-04-22:
Post a wireshark and a console output of the start

lyhawk2007, 2019-04-23:
The data has been read. It's a byte alignment problem.But now the data I read is CoE Osize:256 Isize:16, which should not correspond to my slave station information, and the slave station ESI file is
			<Fmmu>MBoxState</Fmmu>
				<Sm MinSize="34" MaxSize="256" DefaultSize="128" StartAddress="#x1000" ControlByte="#x26" Enable="1">MBoxOut</Sm>
				<Sm MinSize="34" MaxSize="256" DefaultSize="128" StartAddress="#x1500" ControlByte="#x22" Enable="1">MBoxIn</Sm>
				<Sm MinSize="256" MaxSize="256" StartAddress="#x1a00" ControlByte="#x64" DefaultSize="4" Enable="1">Outputs</Sm>
				<Sm MinSize="256" MaxSize="256" StartAddress="#x1b00" ControlByte="#x20" DefaultSize="4" Enable="1">Inputs</Sm>

lyhawk2007, 2019-04-23:
[ESI.txt](https://github.com/OpenEtherCATsociety/SOEM/files/3105429/ESI.txt)

lyhawk2007, 2019-04-23:
Slave device information
Slave 1 State=0x0012 StatusCode=0x0017

closed 2019-04-23 by lyhawk2007