OpenEtherCATsociety/SOEM #287: If only slave without PDO is connected, then *context->DCtime is not updated.

Apollo3zehn, 2019-04-29:
In my very simple test setup I have only one EK1100 with no PDOs connected. In that case, since [length](https://github.com/OpenEtherCATsociety/SOEM/blob/8832ef0f2fa09fb0b06acaca239b51263ca1878c/soem/ethercatmain.c#L1723-L1728) is zero, no frame is sent out. Thus, `ecx_receive_processdata` cannot receive anything and so `*context->DCtime` keeps unchanged, although it should contain the current time.

I would say it makes sense to not send any frame to update the PDOs, but I think that DCtime should be updated, whenever at least one DC capable slave is present. What do you think?

ArthurKetels, 2019-05-02:
Why bother having an EtherCAT system running when it has no IO?

Apollo3zehn, 2019-05-03:
In general (with more slaves), I use the DCtime as the source of timestamps for measurement data since it is a monotonic clock. I also constantly monitor the time difference between DC time and OS time, so I can slowly adapt the DC time, when it drifts. When I disconnect all slaves (for testing), and only leave EK1100 connected, the time difference becomes huge as the DC time is not updated (it keeps equal to 0).

I understand it is not a very important use case, but I would expect to have the same DC behavior with and without additional slaves.